fwupd/fwupd #289: make-images fails strangely when no fonts can be found

DimStar77, 2017-10-27:
While building packages for openSUSE, we observed this build error

```
[  209s] Traceback (most recent call last):
[  209s]   File "/home/abuild/rpmbuild/BUILD/fwupd-0.9.7/po/make-images", line 194, in <module>
[  209s]     r.render()
[  209s]   File "/home/abuild/rpmbuild/BUILD/fwupd-0.9.7/po/make-images", line 183, in render
[  209s]     render_one(lang, string, width, height, filename)
[  209s]   File "/home/abuild/rpmbuild/BUILD/fwupd-0.9.7/po/make-images", line 113, in render_one
[  209s]     log = data['log']
[  209s] KeyError: 'log'
```

Turns out, some 'sans' fonts are needed in the buildroot (probably not much of an issue on regular users running the build, but in minimal buildroot specifically setup, it's easy for no fonts to be there)

Would be great if the error message were a bit more verbose as to what is missing

superm1, 2017-10-27:
At least in the CI jobs we do run in a very minimal environment, however Debian, Ubuntu, Arch and Fedora all install the fonts.  See https://github.com/hughsie/fwupd/blob/master/contrib/ci/dependencies.txt#L10 for more information.

I tried to reproduce this by `# rm /usr/share/fonts/truetype/dejavu -rf` in a build root before running the build so I could make a better error but it still didn't actually fail for me.

So I tried to axe the conf files associated with fonts in /etc/fonts.d, but that just made a bigger (different) mess.
```
Running custom install script '/build/build/po/make-images.sh'
Traceback (most recent call last):
  File "/build/build/po/make-images", line 193, in <module>                                                                                                                                                    
    r.render()
  File "/build/build/po/make-images", line 182, in render                                                                                                                                                      
    render_one(lang, string, width, height, filename)                                                                                                                                                          
  File "/build/build/po/make-images", line 97, in render_one                                                                                                                                                   
    length = len(bytes(string, "utf8"))                                                                                                                                                                        
UnicodeEncodeError: 'utf-8' codec can't encode character '\udce2' in position 26: surrogates not allowed
gzip: /usr/share/locale/*/LC_IMAGES/*.bmp: No such file or directory                                                                                                                                           
Failed to run install script '/build/build/po/make-images.sh'
```

Since you can trivially reproduce this exact error, would you mind proposing a patch to test for the fonts that cause the failure for you?  Preferably a patch that tests via the existing dependency test for `make-images` (https://github.com/hughsie/fwupd/blob/master/po/test-deps).

superm1, 2017-10-27:
I finally got it to reproduce, I think this should be adequate (the build will fail with a clear message).
https://github.com/hughsie/fwupd/pull/291

If you would like to adjust the `test-deps`, patches welcome for that too.  It would make it fail sooner ("before build" rather than during "installation")

commit f8451a90 (2017-10-27): Make the error message clearer when sans fonts are missing (Closes #289)
closed 2017-10-27 by superm1