fwupd/fwupd #294: Del XPS (9360) on Manjaro : can't update

1LinuxGuy, 2017-10-30:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

`fwupdmgr --version
client version:	1.0.0
daemon version:	1.0.0
compile-time dependency versions
	appstream-glib:	0.6.13
	gusb:	0.2.11

```

```shell
wupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             /dev/mei
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.7.0
  VersionBootloader:    11.7.0
  Icon:                 computer
  Created:              2017-10-30

XPS 13 9360 TPM 2.0
  DeviceId:             DELL-73c2051d-8688-56fb-a93f-d56a9b455e52lu
  Guid:                 73c2051d-8688-56fb-a93f-d56a9b455e52
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              1.3.1.0
  Icon:                 computer
  Created:              2017-10-30

XPS 13 9360 TPM 1.2
  DeviceId:             DELL-f9bdd338-b410-5e73-902d-7b6e4694bb56lu
  Guid:                 f9bdd338-b410-5e73-902d-7b6e4694bb56
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2017-10-30

XPS 13 9360 System Firmware
  DeviceId:             UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.2.1
  VersionLowest:        0.2.2.1
  Icon:                 computer
  Created:              2017-10-30

HD Graphics 620
  DeviceId:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2017-10-30

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001
Boot0000* Windows Boot Manager	HD(2,GPT,39e0b55a-4a55-401a-93b4-cfeaa5a837a7,0xfa000,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...e................
Boot0001* manjaro	HD(1,GPT,e2a2c6ea-ad8b-4ad1-9384-32845aedf7cf,0x1001,0x96001)/File(\EFI\manjaro\grubx64.efi)
Boot0002* antergos_grub	HD(1,GPT,37831561-3e40-4e13-89fa-be6e27347215,0x800,0x64000)/File(\EFI\antergos_grub\grubx64.efi)
Boot0004* ubuntu	HD(1,GPT,a3bf858c-cead-4393-9df0-774aff50c28a,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)

```

```shell
$ efivar -l | grep fw
No info was returned.

```

```shell
$ tree /boot
command not found
```

Please answer the following questions:

- Operating system and version: Manjaro Gnome 
- How did you install fwupd: pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk?
- Is secure boot enabled (only for the UEFI plugin)? No.

In terminal, here's what I get when I try to update the firmware:

fwupdmgr update
No upgrades for device, current is 1.3.1.0: 1.3.1.0=same
Downloading 0.2.3.1 for XPS 13 9360 System Firmware...
Updating 0.2.3.1 on XPS 13 9360 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [   -                                   ]UEFI firmware update failed: {error #0} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #1} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #2} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #3} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #4} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #5} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #6} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #7} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #8} libfwup.c:1194 get_fd_and_media_path(): failed to make /boot/efi/EFI/arch/fw: Aucun fichier ou dossier de ce type	{error #9} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #10} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #11} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #12} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #13} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type	{error #14} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type	{error #15} libfwup.c:1194 get_fd_and_media_path(): failed to make /boot/efi/EFI/arch/fw: Aucun fichier ou dossier de ce type

superm1, 2017-10-30:
(I Just updated your comment for missing triple back ticks)

superm1, 2017-10-30:
So your error is translated to 

> no file or folder of this type

Do you not have a `/sys/firmware/efi/efivars `directory?
Do you have your ESP located at `/boot/efi/EFI/`?

Please install the packaged that provides the `tree `command and share the tree output of `/boot`.

ephase, 2017-10-31:
Hi @superm1, i'v got the same issue with my dell 9360 : 

Archlinux, package installed with pacman
this is an NVMe disk

### fwupdmgr get-devices
~~~
Intel AMT (unprovisioned)
  DeviceId:             /dev/mei
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.7.0
  VersionBootloader:    11.7.0
  Icon:                 computer
  Created:              2017-10-31

XPS 13 9360 System Firmware
  DeviceId:             UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.2.1
  VersionLowest:        0.2.2.1
  Icon:                 computer
  Created:              2017-10-31

HD Graphics 620
  DeviceId:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2017-10-31

~~~
### fwupdmgr --version
~~~
client version: 1.0.0
daemon version: 1.0.0
compile-time dependency versions
        appstream-glib: 0.6.13
        gusb:   0.2.11
~~~

### efibootmgr -v
~~~
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000
Boot0000* Arch  PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-1D-E4-AD)/HD(1,GPT,63f9c347-ceaa-4954-a288-33ec24fd8593,0x800,0x3f800)/File(\EFI\archlinux\grubx64.efi)
~~~

###  tree /boot/efi
~~~
/boot/efi/
└── EFI
    ├── archlinux
    │   └── grubx64.efi
    └── Dell
        └── logs
            ├── diags_current.xml
            └── diags_previous.xml
~~~
I've the `/sys/firmware/efi/efivars` directory

superm1, 2017-10-31:
@ephase you didn't follow the directions to install the bootloader to the ESP after you installed fwupdate.

@ArchangeGabriel since this can't be done automatically (per other discussions), can you please make it more verbose at least on your wiki that explains how to do this?  I swear almost every issue coming into this tracker around UEFI FW update issues is:
1. Forgot to install bootloader
2. ESP at unexpected location

We have to get a hold of this.

ephase, 2017-10-31:
@superm1 Because it's not mentioned in my distribution documentationm but it's done : 
~~~
/boot/efi/EFI/
├── arch
│   ├── fw
│   └── fwupx64.efi
├── archlinux
│   └── grubx64.efi
└── Dell
    └── logs
        ├── diags_current.xml
        └── diags_previous.xml
~~~

Then 
~~~
UEFI firmware update failed: {error #0} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fw
update-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type        {error #1} lib.c:152 efi_get_variable(): ops->get
_variable failed: Aucun fichier ou dossier de ce type   {error #2} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8
c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type     {error #3} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou doss$
er de ce type   {error #4} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c7
39416): Aucun fichier ou dossier de ce type     {error #5} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type  {error #6} efivarfs.c:230
 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce ty
pe      {error #7} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type  {error #8} libfwup.c:1194 get_fd_and_media_path(): failed to make
 /boot/efi/EFI/arch/fw: Aucun fichier ou dossier de ce type     {error #9} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a80
3-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type     {error #10} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier
 ou dossier de ce type  {error #11} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf
5-4d9d1c739416): Aucun fichier ou dossier de ce type    {error #12} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type {error #13} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type     {error #14} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type {error #15} libfwup.c:1194 get_fd_and_media_path(): failed to make /boot/efi/EFI/arch/fw: Aucun fichier ou dossier de ce type    {error #16} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type    {error #17} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type  {error #18} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type    {error #19} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type {error #20} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): Aucun fichier ou dossier de ce type     {error #21} lib.c:152 efi_get_variable(): ops->get_variable failed: Aucun fichier ou dossier de ce type {error #22} libfwup.c:1207 get_fd_and_media_path(
): mkostemps(/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-HLCIbM.cap) failed: Aucun fichier ou dossier de ce type
~~~
I've no folder `fw` into `/sys/firmware/efi/efivars`

superm1, 2017-10-31:
Alright now you're hitting the duplicate path issue.  That's this one: https://github.com/rhboot/fwupdate/issues/80

Update to fwupdate 9-3.

I'm closing this issue, it's Arch packaging for fwupdate and documentation problems, not a fwupd issue.

closed 2017-10-31 by superm1

ArchangeGabriel, 2017-10-31:
@superm1 I live under the same impression, and I agree this is a quite a pain.

Unfortunately I have very limited free time available at the moment, but maybe starting a page about fwupd in Arch wiki would be a good idea.

@ephase There should definitively have been a message upon installing/upgrading fwupdate. If you don’t read install/upgrade messages when using Arch, you’re driving yourself into troubles.

superm1, 2017-10-31:
@ArchangeGabriel sure.  Well when you can get to it, I think it will certainly help curb more of these issues. arch wiki, fwupd wiki, wherever you want to put it let's make sure it's linked back and forth between the two.  

I think it would probably help a lot to have clear documentation that people will see and be able to walk through step by step.

ephase, 2017-10-31:
@ArchangeGabriel Yes i saw it when I upgrade to 9.3  apologies ... Then I was able ti update flawlessly my dell firmware.

ArchangeGabriel, 2018-02-27:
@superm1 Well you’re already aware normally, but just in case and for anyone else coming here, there is now https://wiki.archlinux.org/index.php/Fwupd.