fwupd/fwupd #296: logitech udpate fails: failed predicate [BOT01.0[0-1]_* regex BOT01.02_B0015]

arekm, 2017-11-03:
Trying to update logitech unifying receiver that came with logitech mx ergo trackball and that fails:

````
[root@xps ~]# fwupdmgr refresh
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc

[root@xps ~]# fwupdmgr -v update
Value of bootloader incorrect: failed predicate [BOT01.0[0-1]_* regex BOT01.02_B0015]
````

```shell
$ fwupdgmr --version
client version: 1.0.1
daemon version: 1.0.1
compile-time dependency versions
        appstream-glib: 0.6.13
        gusb:   0.2.10
```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.03_B0025
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2017-11-03
```

Please answer the following questions:

- Operating system and version:
pld-linux.org, Th, current

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
built from git (674ed347d27f4fb1fb7e6a45ff0e0d95102bfdfc)

- Have you tried rebooting?
No.

- Are you using an NVMe disk?
No.

hughsie, 2017-11-03:
I think Logitech used the wrong regex override on the LVFS.  I've fixed it to match what's in the metainfo. Can you do `fwupdmgr refresh` and try again please?

arekm, 2017-11-03:
Updating receiver worked now, thanks!

````
# fwupdmgr get-devices
Unifying Receiver
  DeviceId:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2017-11-03
  Modified:             2017-11-03
````

fwupd doesn't see devices connected to receiver, so I assume it cannot update firmware in my trackball. Well, half job done :)

btw. Funny thing - this device (mx trackball) was just recently introduced to the market, yet, it is sold with old firmware in receiver.

hughsie, 2017-11-03:
> fwupd doesn't see devices connected to receiver

It would if the device connected was supported; we used to list all devices but some devices do crazy things when you start interrogating them for firmware details so we stopped it. I don't think there are any trackball firmwares uploaded yet https://fwupd.org/lvfs/devicelist although I know Logitech are testing more internally.

> is sold with old firmware in receiver.

Yes, I also think this is strange, but at least for the receivers I guess they were made a long time ago. Easily fixed. Thanks for reporting the issue so quickly.

closed 2017-11-03 by hughsie