fwupd/fwupd #298: RQR12.07_B0029 breaks Zone Touch Mouse T400

octoploid, 2017-11-05:
RQR12.07_B0029 breaks the "METRO START SCREEN" button on the Zone Touch Mouse T400.
This mouse has two middle mouse buttons, the lower one works fine.
The upper one is called "METRO START SCREEN" button and is completely dead 
after the update. No xev signal at all.

Downgrading by hand to RQR12.05_B0028 "fixes" the issue.

hughsie, 2017-11-05:
@mviredaz ?

hughsie, 2017-11-08:
I've pinged Logitech to have a look at this issue.

mviredaz, 2017-11-10:
A few of Logitech's mice used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature.  You can read more about this [here](https://community.logitech.com/s/question/0D531000058b3B7CAI/logitech-response-to-research-findings).  Although Logitech does not recommend it, these features may be re-activated by downgrading the receiver to an older firmware.

hughsie, 2017-11-10:
I guess that means we should have a "ignore this device" functionality in fwupd for this kind of scenario. We do have `BlackListDevices=` in the config file, but it's hardly easy to discover. @mviredaz would it also be a good idea to edit the description on the LVFS for the update? I don't know if it's possible to only match and show the warning for affected devices, as it's the peripheral device that's affected, not the unifying receiver.

hughsie, 2018-01-17:
I think the response from Logitech is as good as we can get. I'm sure 99.99% of users would prefer a secure device that's missing a button than one that's easily hacked. Sorry we can't do better here.

closed 2018-01-17 by hughsie