fwupd/fwupd #301: Support Meson 0.29.0 shipped in common distributions to build fwupd

paulmenzel, 2017-11-06:
Ubuntu 16.04.3 LTS ships Meson 0.29.0. With the current build dependency on Meson greater or equal than 0.37.0 it’s inconvenient to build and test it.

```
$ meson /dev/shm
The Meson build system
Version: 0.29.0
Source dir: /home/joey/src/fwupd
Build dir: /dev/shm
Build type: native build
Build machine cpu family: x86_64
Build machine cpu: x86_64

Meson encountered an error in file meson.build, line 1, column 0:
Meson version is 0.29.0 but project requires >=0.37.0.
```

hughsie, 2017-11-06:
We depend on this version of meson as it has the required fixes needed to build fwupd. 0.29.0 is so old that it won't actually compile most projects that chose meson, and 0.37.0 is actually me being quite conservative... Is there a compelling reason you have to support 16.04? I think meson can be installed "locally" if 16.04 is a hard requirement.

paulmenzel, 2017-11-06:
Ubuntu 16.04 LTS is the distribution shipped by Dell on devices like the Dell XPS 13 9360. As their devices are one of the few supporting UEFI Capsule, it’d be great to easily built the latest fwupd version (and probably also to make it easy for the Ubuntu folks to update to later fwupd versions).

hughsie, 2017-11-06:
Does `pip3 install --user meson` give you a usable meson? If so, could we add this to README please?

superm1, 2017-11-06:
Making master (1.x.x) work on 16.04 is an exercise in yak shaving beyond just meson.  I've tried a few times, but then also newer glib2.0, gdk-pixbuf,  appstream-glib, gcab, libgusb need to come at the same time.  If you want to do a packaged version with upstream packaging then you need a newer version of debhelper which requires a newer dpkg which requires a newer dh-strip-nondeterminism.  

Let alone doing all that will then put you into a situation that there may have been ABI changes for rdepends of those packages.  With 18.04 around the corner, it's just not going to happen to move all those pieces.

paulmenzel, 2017-11-06:
I noticed that too, and had to give up trying to build the latest version under Ubuntu 16.04.3 LTS.

Waiting for Ubuntu 18.04 is unacceptable in my opinion, as 16.04.3 will probably be in use for almost a year in my opinion. I do not think Dell will migrate right away in April 2018.

What is the latest version you can get to build under Ubuntu 16.04.3 LTS?

superm1, 2017-11-06:
You're welcome to that opinion, but I my opinion is that  _installing newer versions of fwupd_ will not convince anyone in ubuntu-land to move mountains of dependencies.  The main reason for fwupd in 16.04 is UEFI capsule and it already works properly.
Because of the [Ubuntu SRU policy](https://wiki.ubuntu.com/StableReleaseUpdates) you'll need to make a pretty good case for each dependency along with thorough testing too.

I've tried at various point of the 0.9.x releases without success.  I think 0.8.x will be more likely, but you probably need a newer appstream-glib.

@lyoncore guys did some work to put fwupd in a snap.  That might be a better avenue for supporting newer fwupd on really old Ubuntu.

superm1, 2017-11-06:
I'm closing this bug, we won't change the meson dependency as mentioned by @hughsie 

FWIW meson 0.40.1 is backported to 16.04 in xenial-backports:
https://launchpad.net/ubuntu/+source/meson/0.40.1-1~ubuntu16.04.1

closed 2017-11-06 by superm1

dragon788, 2018-01-26:
Would it be possible to build fwupdmgr in a snap "container" or does that prevent access to the hardware that it needs? I believe doing it in a container would let you pull in the dependencies you need without affecting the host.

superm1, 2018-01-26:
There was a snap built for Ubuntu by lyoncore, but it hasn't been updated in a long time.  You can start with that if you want to try again.
https://github.com/Lyoncore/fwupd-snapcraft

From an upstream perspective any confinement rules applied by a snap wouldn't be supported.