fwupd/fwupd #303: [0.7.0] Unable to update from 1.3.5 to 2.3.1 on Dell XPS 13 9360

paulmenzel, 2017-11-06:
Following up on a [Linux kernel regression](https://bugzilla.kernel.org/show_bug.cgi?id=196907) on the Dell XPS 13 9360, I want to update the firmware, but that doesn’t work with fwupdmgr from fwupd 0.7.0-0ubuntu4.3 shipped with Ubuntu 16.04.3 LTS used by Dell. 

```shell
$ fwupdmgr get-devices
ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Provider:             Udev
  Flags:                internal|locked
  DeviceVendor:         Intel Corporation
  Created:              2017-11-06
  Trusted:              none

UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  DisplayName:          XPS 13 9360
  Provider:             UEFI
  Flags:                internal|allow-offline|require-ac
  Version:              0.1.3.5
  VersionLowest:        0.1.3.5
  Created:              2017-11-06
  Trusted:              none

usb:00:05
  Guid:                 87c78d19-a3ed-5778-9b69-8eb701529940
  DisplayName:          Integrated_Webcam_HD
  Provider:             USB
  Flags:                none
  Version:              99.24
  Created:              2017-11-06
  Trusted:              none

usb:00:04
  Guid:                 0f15c153-cc04-589b-8886-aba87f98918d
  DisplayName:          Touchscreen
  Provider:             USB
  Flags:                none
  Version:              17.17
  Created:              2017-11-06
  Trusted:              none

$ sudo fwupdmgr refresh
$ sudo fwupdmgr get-updates
No devices can be updated: Nothing to do
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000
Boot0000* ubuntu        HD(1,GPT,142dc41d-2c62-47f2-94d2-75165b1bc9cf,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater \fwupx64.efi   HD(1,GPT,142dc41d-2c62-47f2-94d2-75165b1bc9cf,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0006* Windows Boot Manager  HD(3,GPT,1cb9c60d-fbc9-4f52-94d2-fe1b0a342fa4,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...d................
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0
```

```shell
$ tree /boot
/boot
├── abi-4.10.0-38-generic
├── config-4.10.0-38-generic
├── config-4.12.0+
├── config-4.12.0-rc5+
├── config-4.12.0-rc7+
├── config-4.14.0-rc7+
├── efi [error opening dir]
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├── de.mo
│   │   ├── en_AU.mo
│   │   ├── en_CA.mo
│   │   ├── en_GB.mo
│   │   └── en@quot.mo
│   ├── unicode.pf2
│   └── x86_64-efi
[…]
├── initrd.img-4.10.0-37-generic
├── initrd.img-4.10.0-38-generic
├── initrd.img-4.12.0+
├── initrd.img-4.12.0-rc5+
├── initrd.img-4.12.0-rc7+
├── initrd.img-4.14.0-rc7+
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.10.0-38-generic
├── System.map-4.12.0+
├── System.map-4.12.0-rc5+
├── System.map-4.12.0-rc7+
├── System.map-4.14.0-rc7+
├── vmlinuz-4.10.0-38-generic
├── vmlinuz-4.10.0-38-generic.efi.signed
├── vmlinuz-4.12.0+
├── vmlinuz-4.12.0-rc5+
├── vmlinuz-4.12.0-rc7+
└── vmlinuz-4.14.0-rc7+

6 directories, 296 files
```

Please answer the following questions:

- Operating system and version: Ubuntu 16.04.3 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting? No.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

Unfortunately, I am unable to build the latest fwupd on Ubuntu 16.04 as fwupd’s build requirements are too new and not supported by that distribution.

Rant: I really do not understand, why Dell doesn’t test this better. It’s a real bad experience, and in the current state does not safe any time over using an USB device, and even causes big frustration due to the time waste. That is really sad. Dell and Ubuntu should really get their act together, and first test fwupd with their latest firmware releases, and if newer fwupd is required, backport that into the Ubuntu version they ship.

hughsie, 2017-11-06:
Is it the missing update in `fwupdmgr` that's troubling? Although 0.7.0 is super old, if you get the daemon log with --verbose (showing refresh and get-updates) maybe we can spot something obvious. I also wonder if https://fwupd.org/downloads/6fd391badc79a4d9a2c9916073f6941f885a22fd-firmware_XPS_9360_2_3_1.cab is install-able directly -- although if that works it obviously makes this mug impossible to track down...

superm1, 2017-11-06:
Do you have AC plugged in?

paulmenzel, 2017-11-06:
No. Why should I to list updates?

paulmenzel, 2017-11-06:
```
$ sudo /usr/lib/x86_64-linux-gnu/fwupd/fwupd --verbose
14:31:43	Verbose debugging enabled (on console 1)
14:31:43	adding plugin /usr/lib/x86_64-linux-gnu/fwupd-plugins-1/libfu_plugin_test.so
14:31:43	performing init() on test
14:31:43	adding plugin /usr/lib/x86_64-linux-gnu/fwupd-plugins-1/libfu_plugin_steelseries.so
14:31:44	Loading fallback values from /etc/fwupd.conf
14:31:44	adding udev device: /sys/devices/pci0000:00/0000:00:02.0
14:31:44	fixing GUID 0x8086:0x5916->3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
14:31:44	emit added from Udev: ro__sys_devices_pci0000_00_0000_00_02_0
14:31:44	emit added from UEFI: UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0
14:31:44	emit added from USB: usb:00:05
14:31:44	emit added from USB: usb:00:04
14:31:44	no product string descriptor
#################################################################                      @0413ms AsStore:load 413ms
#################################################################                      @0413ms AsStore:load{per-system} 413ms
 #                                                                                     @0017ms AsStore:app-info{/var/lib/app-info/yaml/security.ubuntu.com_ubuntu_dists_xenial-security_main_dep11_Components-amd64.yml.gz} 7ms
 #                                                                                     @0017ms AsStore:store-from-file 7ms
   #######                                                                             @0069ms AsStore:app-info{/var/lib/app-info/yaml/de.archive.ubuntu.com_ubuntu_dists_xenial-updates_main_dep11_Components-amd64.yml.gz} 50ms
   #######                                                                             @0069ms AsStore:store-from-file 50ms
           ##                                                                          @0086ms AsStore:app-info{/var/lib/app-info/yaml/de.archive.ubuntu.com_ubuntu_dists_xenial-updates_universe_dep11_Components-amd64.yml.gz} 16ms
           ##                                                                          @0086ms AsStore:store-from-file 16ms
             #########################################                                 @0343ms AsStore:app-info{/var/lib/app-info/yaml/de.archive.ubuntu.com_ubuntu_dists_xenial_universe_dep11_Components-amd64.yml.gz} 257ms
             #########################################                                 @0343ms AsStore:store-from-file 257ms
                                                      #                                @0348ms AsStore:app-info{/var/lib/app-info/yaml/de.archive.ubuntu.com_ubuntu_dists_xenial-backports_main_dep11_Components-amd64.yml.gz} 5ms
                                                      #                                @0348ms AsStore:store-from-file 5ms
                                                       #########                       @0408ms AsStore:app-info{/var/lib/app-info/yaml/de.archive.ubuntu.com_ubuntu_dists_xenial_main_dep11_Components-amd64.yml.gz} 59ms
                                                       #########                       @0408ms AsStore:store-from-file 59ms
                                                                  ###################  @0539ms FuMain:coldplug 121ms
                                                                   ##################  @0539ms FuMain:coldplug{USB} 118ms
                                                                   ##################  @0538ms FuProviderUsb:added{0bda:568b} 117ms
14:31:44	FuMain: acquired name: org.freedesktop.fwupd
14:31:44	devices now in store:
14:31:44	1	com.steelseries.rival-legacy.firmware	Rival SteelSeries
14:31:44	2	UEFI-dummy-dev0	UEFI Updates
14:31:44	3	com.via.VL811.firmware	VL811 Firmware
14:31:44	4	com.via.VL811+.firmware	VL811+ Firmware
14:31:44	5	com.via.VL812.firmware	VL812 Firmware
14:31:44	6	com.via.VL812_B2.firmware	VL812 B2 Firmware
14:31:44	7	com.8bitdo.fc30.firmware	FC30
14:31:44	8	com.8bitdo.fc30arcade.firmware	FC30 Joystick
14:31:44	9	com.8bitdo.fc30pro.firmware	FC30 Pro
14:31:44	10	com.8bitdo.n30arcade.firmware	N30 Arcade
14:31:44	11	com.8bitdo.nes30.firmware	NES30
14:31:44	12	com.8bitdo.nes30pro.firmware	NES30 Pro
14:31:44	13	com.8bitdo.sfc30.firmware	SFC30
14:31:44	14	com.8bitdo.snes30.firmware	SNES30
14:31:44	15	com.dell.uefi10c40550.firmware	OptiPlex 7050 System Update
14:31:44	16	com.dell.uefi124c207d.firmware	XPS 15 9550/Precision 5510 System Update
14:31:44	17	com.dell.uefi1367a91a.firmware	OptiPlex 5050 System Update
14:31:44	18	com.dell.uefi1610b70e.firmware	Inspiron 14 7000 Gaming/Inspiron 15 7000 Gaming System Update
14:31:44	19	com.dell.uefi169d9146.firmware	Vostro 14-3468 System Update
14:31:44	20	com.dell.uefi1d4362ca.firmware	Inspiron 14-3467 System Update
14:31:44	21	com.dell.uefi212026ee.firmware	Latitude E7X70 System Update
14:31:44	22	com.dell.uefi21f94926.firmware	Edge Gateway 3000/3001/3002/3003 System Update
14:31:44	23	com.dell.uefi222e3362.firmware	Precision Tower 3420 System Update
14:31:44	24	com.dell.uefi22d63f4.firmware	TPM 2.0 Update
14:31:44	25	com.dell.uefi293af847.firmware	OptiPlex 3050 System Update
14:31:44	26	com.dell.uefi33773727.firmware	XPS 13 9350 System Update
14:31:44	27	com.dell.uefi34578c72.firmware	XPS 15 9560/Precision 5520 System Update
14:31:44	28	com.dell.uefi3c20b9e1.firmware	OptiPlex 7450 AIO System Update
14:31:44	29	com.dell.uefi416d1c90.firmware	OptiPlex 5250 AIO System Update
14:31:44	30	com.dell.uefi43ca3264.firmware	OptiPlex 7440 AIO System Update
[... cut ...]
14:32:08	59	com.hughski.ColorHugALS.firmware	ColorHugALS
14:32:08	60	com.logitech.MPK01.firmware	Logitech K780 Multi-Device Wireless Keyboard
14:32:08	61	com.logitech.Unifying.RQR12.firmware	Logitech Unifying Receiver
14:32:08	62	com.logitech.Unifying.RQR24.firmware	Logitech Unifying Receiver
14:32:11	Called GetUpdates()
14:32:11	ignoring update for UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0 as not on AC power
14:32:14	Called GetUpdates()
14:32:14	ignoring update for UEFI-5ffdbc0d-f340-441c-a803-8439c8c0ae10-dev0 as not on AC power
```

paulmenzel, 2017-11-06:
With AC plugged in:

```
$ sudo fwupdmgr get-updates
Firmwareaktualisierungen für XPS 13 9360 verfügbar:
Kennung:             com.dell.uefi5ffdbc0d.firmware
GUID:                5ffdbc0d-f340-441c-a803-8439c8c0ae10
Version:             0.2.3.1
Prüfsumme:          18d9501fc82e8c39acb57693e2c6ad60a2712f9a
Checksum Type:       sha1
Ort:                 https://fwupd.org/downloads/6fd391badc79a4d9a2c9916073f6941f885a22fd-firmware_XPS_9360_2_3_1.cab
```

paulmenzel, 2017-11-06:
```
$ sudo fwupdmgr update
Downloading 0.2.3.1 for XPS 13 9360...
Updating 0.2.3.1 on XPS 13 9360...
 * Firmware wird entpackt
Erneuter Versuch als Offline-Aktualisierung...
 * Untätig
 * Firmware wird entpackt
 * Aktualisierung wird geplant
```

paulmenzel, 2017-11-06:
Updates should be shown despite the AC being plugged in or not.

paulmenzel, 2017-11-06:
Thank you for the quick help though.

hughsie, 2017-11-06:
> Updates should be shown despite the AC being plugged in or not.

This is fixed in 0.8.0, released nearly a year ago. I think 16.04 really needs to update from 0.7.x...

superm1, 2017-11-06:
As mentioned by @hughsie this is the behavior associated with 0.7.x releases.  It's been adjusted in 0.8.x, so from an upstream perspective this won't be fixed in fwupd 0.7.x.

closed 2017-11-06 by superm1