fwupd/fwupd #307: updating unified receiver error

akostadinov, 2017-11-07:
```
$ sudo fwupdmgr update
Downloading RQR24.05_B0029 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
Downloading…           [****************************************]
Updating RQR24.05_B0029 on Unifying Receiver...
Decompressing…         
request timed out
```

```shell
$ fwupdmgr --version
client version:	0.9.6
daemon version:	0.9.6
compile-time dependency versions
	appstream-glib:	0.7.0
	gusb:	0.2.11
```

```shell
$ fwupdmgr get-devices
...
Unifying Receiver
  DeviceID:             /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
  Plugin:               unifying
  Flags:                allow-online|supported
  DeviceVendor:         Logitech
  DeviceVendorId:       USB:0x046D
  Version:              RQR24.01_B0023
  VersionBootloader:    BOT03.01_B0008
  Created:              2017-11-07
...
```

Please answer the following questions:

- Operating system and version: Fedora 26
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf install
- Have you tried rebooting?  no
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2017-11-07:
Can you get the `fwupd --verbose` log when the update times out please.

akostadinov, 2017-11-07:
```
$ sudo fwupdmgr update --verbose
Downloading RQR24.05_B0029 for Unifying Receiver...
(fwupdmgr:5717): Fu-DEBUG: skpping download as file already exists
Updating RQR24.05_B0029 on Unifying Receiver...
(fwupdmgr:5717): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         (fwupdmgr:5717): Fwupd-DEBUG: Emitting ::status-changed() [idle]

(fwupdmgr:5717): Fwupd-DEBUG: Emitting ::device-removed((null))
(fwupdmgr:5717): Fwupd-DEBUG: Emitting ::changed()
(fwupdmgr:5717): Fwupd-DEBUG: Emitting ::device-added((null))
(fwupdmgr:5717): Fwupd-DEBUG: Emitting ::changed()
request timed out
```

hughsie, 2017-11-07:
No, I actually mean `/usr/libexec/fwupd/fwupd --verbose` as root.

akostadinov, 2017-11-08:
```
$ sudo /usr/libexec/fwupd/fwupd --verbose 
00:06:23:0270 Fu  Verbose debugging enabled (on console 1)
00:06:23:0271 Fu  loading config values from /etc/fwupd.conf
00:06:23:0272 Fu  using config path of /etc/fwupd
00:06:23:0272 Fu  loading from /etc/fwupd/remotes.d/fwupd.conf
00:06:23:0272 Fu  loading from /etc/fwupd/remotes.d/lvfs-testing.conf
00:06:23:0272 Fu  loading from /etc/fwupd/remotes.d/lvfs.conf
00:06:23:0272 Fu  loading from /etc/fwupd/remotes.d/vendor.conf
00:06:23:0272 Fu  ordering lvfs-testing=lvfs+1
00:06:23:0272 Fu  ordering lvfs=fwupd+1
00:06:23:0272 Fu  ordering lvfs-testing=lvfs+1
00:06:23:0272 Fu  remote lvfs-testing not enabled, so skipping
00:06:23:0272 As  run AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz}
00:06:23:0277 As  run AsStore:store-from-root
00:06:23:0277 As  using icon path /var/lib/fwupd/remotes.d/lvfs/icons
00:06:23:0278 As  run AsStore:match-addons
00:06:23:0278 As  Emitting ::changed() [from-root]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0278 As  Emitting ::changed() [add-app]
00:06:23:0279 As  run AsStore:store-from-file{/usr/share/fwupd/remotes.d/fwupd/metadata.xml}
00:06:23:0279 As  run AsStore:store-from-root
00:06:23:0279 As  using icon path /usr/share/fwupd/remotes.d/fwupd/icons
00:06:23:0279 As  run AsStore:match-addons
00:06:23:0279 As  Emitting ::changed() [from-root]
00:06:23:0279 As  Emitting ::changed() [add-app]
00:06:23:0279 As  Emitting ::changed() [add-app]
00:06:23:0279 Fu  remote vendor not enabled, so skipping
00:06:23:0279 Fu  devices now in store:
00:06:23:0279 Fu  1	com.dell.uefi5b7ab884.firmware	Precision E7X10 System Update
00:06:23:0279 Fu  2	com.8bitdo.fc30.firmware	FC30
00:06:23:0279 Fu  3	com.dell.uefi9cb573d5.firmware	Latitude 3480 System Update
00:06:23:0279 Fu  4	com.dell.uefi6180aaaa.firmware	Latitude E5X60 System Update
00:06:23:0279 Fu  5	com.dell.uefia5d83d78.firmware	Precision 5720 AIO System Update
00:06:23:0279 Fu  6	com.8bitdo.fc30arcade.firmware	FC30 Joystick
00:06:23:0279 Fu  7	com.dell.uefid636245a.firmware	XPS 13 9365 System Update
00:06:23:0279 Fu  8	com.dell.uefi21f94926.firmware	Edge Gateway 3000/3001/3002/3003 System Update
00:06:23:0279 Fu  9	com.8bitdo.snes30.firmware	SNES30
00:06:23:0279 Fu  10	com.dell.uefiaee2604a.firmware	Embedded Box PC 3000 System Update
00:06:23:0279 Fu  11	com.dell.uefi4fed6c9d.firmware	OptiPlex 7050 System Update
00:06:23:0279 Fu  12	com.dell.uefi169d9146.firmware	Vostro 14-3468 System Update
[... 475 lines cut ...]
00:09:18:0889 Unifying [HID] device->host:02 00 00 ff 0f 00 00 00 
00:09:18:0889 Unifying HID++1.0 report 0x02 has unknown length, ignoring
00:09:18:0900 Unifying [HID] device->host:11 01 00 17 04 05 aa 00 00 00 00 00 00 00 00 00 00 00 00 00 
00:09:18:0900 Unifying [HID] device->host:20 01 02 00 00 ff 0f 00 00 00 30 00 00 00 00 
00:09:23:0856 Unifying [HID] device->host:02 00 00 ff 0f 00 00 00 
00:09:23:0856 Unifying [HID] host->device:10 01 00 17 00 00 aa 
00:09:23:0861 Unifying [HID] device->host:02 01 00 06 00 00 00 00 
00:09:23:0861 Unifying HID++1.0 report 0x02 has unknown length, ignoring
00:09:23:0869 Unifying [HID] device->host:11 01 00 17 04 05 aa 00 00 00 00 00 00 00 00 00 00 00 00 00 
00:09:23:0869 Unifying [HID] device->host:20 01 02 00 00 ff 0f 00 00 00 30 00 00 00 00 
00:09:28:0861 Unifying [HID] device->host:02 01 00 05 00 00 00 00 
00:09:28:0861 Unifying [HID] host->device:10 01 00 17 00 00 aa 
00:09:28:0865 Unifying [HID] device->host:02 01 00 fd 1f 00 00 00 
00:09:28:0865 Unifying HID++1.0 report 0x02 has unknown length, ignoring
00:09:28:0873 Unifying [HID] device->host:02 01 00 fd 1f 00 00 00 
00:09:28:0873 Unifying HID++1.0 report 0x02 has unknown length, ignoring
00:09:28:0881 Unifying [HID] device->host:11 01 00 17 04 05 aa 00 00 00 00 00 00 00 00 00 00 00 00 00 
00:09:28:0882 Unifying [HID] device->host:20 01 02 00 00 01 00 00 00 00 3d 00 00 00 00 
00:09:33:0867 Unifying [HID] device->host:02 01 00 ff 0f 00 00 00 
00:09:33:0867 Unifying [HID] host->device:10 01 00 17 00 00 aa 
00:09:33:0876 Unifying [HID] device->host:02 01 00 fe ef ff 00 00 
00:09:33:0876 Unifying HID++1.0 report 0x02 has unknown length, ignoring
00:09:33:0883 Unifying [HID] device->host:11 01 00 17 04 05 aa 00 00 00 00 00 00 00 00 00 00 00 00 00 
00:09:33:0883 Unifying [HID] device->host:20 01 02 00 00 01 00 00 00 00 3d 00 00 00 00 
```

hughsie, 2018-09-12:
@akostadinov -- Can you try with fwupd 1.1.2 please -- we rewrote and improved the unifying plugin and it now seems to work a lot better for more people. I've done RQR24 and RQR12 updates and each has had 100% success now.

akostadinov, 2018-09-12:
I will check in a couple of days when I get back home. Thank you.

superm1, 2018-10-15:
I'm closing older bugs with `needsinfo` tags such as this one.  Can you please retry with a current release (1.1.3) and if the issues continue to exist, please re-open.

Even if the current release is not yet packaged for your distro, you can install it via snap like this:
```
# snap install --classic fwupd
```

closed 2018-10-15 by superm1

akostadinov, 2018-10-16:
Sorry for forgetting about this issue. Now trying it I'm getting:
```
$ sudo fwupdmgr update --verbose
No upgrades for Unifying Receiver, current is RQR24.05_B0029: RQR24.05_B0029=same
```
So I can't really test. I got my mouse replaced at some point due to mechanical issue. Maybe new receiver was already updated. Not sure I can get my hands on the old one. But if I do, I'll try it out.