fwupd/fwupd #316: fwupd uses 100% of a cpu core when using a eGPU over thunderbolt

ArtOfSettling, 2017-11-14:
This happens every time I reboot my machine, I can simply kill the process, but it's a bit annoying on every startup.

```shell
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.7.2
	gusb:	0.2.11
```

```shell
$ fwupdmgr get-devices
Blade Stealth System Firmware
  DeviceID:             UEFI-2f41f3c2-bb7c-4895-b9cd-621622c4de2c-dev0
  Guid:                 2f41f3c2-bb7c-4895-b9cd-621622c4de2c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.101
  VersionLowest:        0.0.101
  Created:              2017-11-14

Touchscreen
  DeviceID:             usb:00:09
  Guid:                 d1693353-e280-5dce-b6e9-31bd3e5e847d
  Guid:                 9ac200a9-eded-50b8-a009-d34b0bbdab90
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x04F3
  Version:              87.4
  Created:              2017-11-14

Razer Blade Stealth
  DeviceID:             usb:00:08
  Guid:                 67712ef8-7c73-5ace-a961-2692fe8d21e8
  Guid:                 a5923fe2-8da1-5bd7-80dc-e59fb227daaa
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x1532
  Version:              2.0
  Created:              2017-11-14

Magic Keyboard
  DeviceID:             usb:00:07
  Guid:                 4151381f-eeb0-5fc1-b4d1-673eae998f49
  Guid:                 4bc99c89-15df-59dc-8a8f-e6965dd4a3e2
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x05AC
  Version:              8.5
  Created:              2017-11-14

Gaming Mouse G502
  DeviceID:             usb:00:02
  Guid:                 c384f7cc-f925-599d-bc08-ff08c8a3b253
  Guid:                 bd11411b-b73c-5ea7-9846-c4127a541cf8
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x046D
  Version:              3.1
  Created:              2017-11-14

USB Camera
  DeviceID:             usb:00:01
  Guid:                 5d05c9df-81ca-5fa8-b754-52b2c5236170
  Guid:                 59bc9d42-7a4a-5ebd-8225-99901baaf83c
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x1BCF
  Version:              0.6
  Created:              2017-11-14

Unknown Device
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-11-14

GP104 [GeForce GTX 1070]
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_1c_4_0000_02_00_0_0000_03_01_0_0000_05_00_0_0000_06_01_0_0000_07_00_0
  Guid:                 43d43547-78f3-552d-b2b6-0cabb4d4d52e
  Plugin:               udev
  Flags:                internal|registered
  DeviceVendor:         NVIDIA Corporation
  DeviceVendorId:       PCI:0x10DE
  Created:              2017-11-14

```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0001,0002,0003,0004
Boot0000* Fedora	HD(2,GPT,4a4a47db-a4e5-4ff2-af9a-0fa20c78a28c,0x1770800,0x32000)/File(\EFI\FEDORA\SHIM.EFI)
Boot0001* Windows Boot Manager	HD(2,GPT,4a4a47db-a4e5-4ff2-af9a-0fa20c78a28c,0x1770800,0x32000)/File(\EFI\MICROSOFT\BOOT\BOOTMGFW.EFI)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0002* UEFI:CD/DVD Drive	BBS(129,,0x0)
Boot0003* UEFI:Removable Device	BBS(130,,0x0)
Boot0004* UEFI:Network Device	BBS(131,,0x0)

```

```shell
$ efivar -l | grep fw

```

Please answer the following questions:

- Operating system and version: Fedora 27
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): It was pre installed
- Have you tried rebooting? Yes
- Are you using an NVMe disk? 
- Is secure boot enabled (only for the UEFI plugin)? It's disabled at the bios level

hughsie, 2017-11-14:
Hey, thanks for reporting this. Can you attach the output of `sudo /usr/libexec/fwupd/fwupd --verbose` please -- if there's nothing spewing junk and the daemon is still at 100% CPU then if you can try and get a backtrace of that using `gdb` that would be super helpful too. Thanks.

ArtOfSettling, 2017-11-14:
Hey! I actually saw this response in another bug report and tried the command, unfortunately, running this command seems to resolve the issue, the cpu usage of fwupd returns to normal.

hughsie, 2017-11-14:
Can you try running that in a VT, and then logging out and then in again? Thanks.

ArtOfSettling, 2017-11-14:
It seems as though the issue is unrelated to session, log out and log in again doesn't seem to reproduce the issue, it seems to occur only after a reboot

hughsie, 2017-11-14:
Does the 10% cpu issue happen if the eGPU is detached? If not, can you reproduce it if you unplug-replug?

ArtOfSettling, 2017-11-14:
It does not reproduce if the eGPU is not unplugged, so it's definitely related, unfortunately hot plugging the GPU does not currently work.

One thing of note is that the eGPU also has a bunch of USB ports and an ethernet port. I'm not sure if that additional information helps at all :)

hughsie, 2017-11-14:
Okay, I guess we need to reboot with the eGPU plugged in to reproduce the problem, then attach to the 100% CPU fwupd process using gdb and then try and get a backtrace. Are you able to give that a go?

ArtOfSettling, 2017-11-14:
Actually attaching to the running process, appears to kill it immediately, I have the debug information for fwupd, but nothing useful displays in the console.

hughsie, 2017-11-14:
> Actually attaching to the running process, appears to kill it immediately

I guess when you attach it's frozen all threads -- do you get a huge backtrace if you type: `t a a bt`?

ArtOfSettling, 2017-11-14:
Thread 5 (Thread 0x7ff56ace6700 (LWP 2291)):
#0  0x00007ff581ec98bb in poll () from target:/usr/lib64/libc.so.6
#1  0x00007ff58359ded9 in g_main_context_iterate.isra ()
   from target:/usr/lib64/libglib-2.0.so.0
#2  0x00007ff58359e272 in g_main_loop_run ()
   from target:/usr/lib64/libglib-2.0.so.0
#3  0x00007ff583285b36 in gdbus_shared_thread_func ()
   from target:/usr/lib64/libgio-2.0.so.0
#4  0x00007ff5835c54c6 in g_thread_proxy ()
   from target:/usr/lib64/libglib-2.0.so.0
#5  0x00007ff5821a8609 in start_thread ()
   from target:/usr/lib64/libpthread.so.0
#6  0x00007ff581ed5e6f in clone () from target:/usr/lib64/libc.so.6

Thread 4 (Thread 0x7ff5728e7700 (LWP 2280)):
#0  0x00007ff581ecadc7 in ioctl () from target:/usr/lib64/libc.so.6
#1  0x00007ff57eb40733 in reap_for_handle ()
   from target:/usr/lib64/libusb-1.0.so.0
#2  0x00007ff57eb42720 in op_handle_events ()
   from target:/usr/lib64/libusb-1.0.so.0
#3  0x00007ff57eb3c5f9 in handle_events ()
   from target:/usr/lib64/libusb-1.0.so.0

ArtOfSettling, 2017-11-14:
from the trace, I can't really see anything suspicious, it doesn't look dead to me.

ArtOfSettling, 2017-11-14:
actually, it seems like for whatever reason, after this restart, the process is staying alive whilst I attach.

hughsie, 2017-11-14:
I assume you're attaching when fwupd is spinning in a 100% loop, which is confusing as it looks all in poll, which should be 0% CPU... If you `c` to continue, and then `bt` again, does anything else pop up? You might need to do this a few times. I'm looking for something with a `fu_` prefix. It would also be useful to have the debuginfo packages installed if that's possible.

ArtOfSettling, 2017-11-14:
I'll give this a bash now, I installed the debug-info packages for fwupd, will happily install them for any other libraries you'd need. I'll try this command now

ArtOfSettling, 2017-11-14:
I couldn't find anything like you said, but grabbed a couple of steps, with a bit of debug information I'll have a deeper look

0x00007f032abb88bb in poll () from target:/usr/lib64/libc.so.6
(gdb) bt

#0  0x00007f032abb88bb in poll () from target:/usr/lib64/libc.so.6
#1  0x00007f032c28ced9 in g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4187
#2  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#3  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#4  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) 
#0  0x00007f032abb88bb in poll () from target:/usr/lib64/libc.so.6
#1  0x00007f032c28ced9 in g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4187
#2  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#3  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#4  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
Single stepping until exit from function poll,
which has no line number information.
g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4188
4188	      errsv = errno;
(gdb) bt
#0  g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4188
#1  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
4189	      if (ret < 0 && errsv != EINTR)
(gdb) bt
#0  g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4189
#1  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
4188	      errsv = errno;
(gdb) bt
#0  g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4188
#1  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
4189	      if (ret < 0 && errsv != EINTR)
(gdb) bt
#0  g_main_context_poll (priority=<optimized out>, n_fds=4, fds=0x7f0308005450, timeout=<optimized out>, context=0x55ff27247670) at gmain.c:4189
#1  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3881
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3883
3883	  some_ready = g_main_context_check (context, max_priority, fds, nfds);
(gdb) bt
#0  g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3883
#1  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#2  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
g_main_context_check (context=context@entry=0x55ff27247670, max_priority=2147483647, fds=fds@entry=0x7f0308005450, n_fds=n_fds@entry=4) at gmain.c:3632
3632	{
(gdb) bt
#0  g_main_context_check (context=context@entry=0x55ff27247670, max_priority=2147483647, fds=fds@entry=0x7f0308005450, n_fds=n_fds@entry=4) at gmain.c:3632
#1  0x00007f032c28ce80 in g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3883
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
3639	  LOCK_CONTEXT (context);
(gdb) bt
#0  g_main_context_check (context=context@entry=0x55ff27247670, max_priority=2147483647, fds=fds@entry=0x7f0308005450, n_fds=n_fds@entry=4) at gmain.c:3639
#1  0x00007f032c28ce80 in g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3883
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb) step
[... 133 lines cut ...]
[... cut ...]
#0  g_main_context_check (context=context@entry=0x55ff27247670, max_priority=2147483647, fds=fds@entry=0x7f0308005450, n_fds=n_fds@entry=4) at gmain.c:3687
#1  0x00007f032c28ce80 in g_main_context_iterate (context=0x55ff27247670, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3883
#2  0x00007f032c28d272 in g_main_loop_run (loop=0x55ff272293a0) at gmain.c:4082
#3  0x000055ff26dfc926 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:947
(gdb)

ArtOfSettling, 2017-11-14:
and then, after a continue : 

(gdb) continue
Continuing.

Thread 3 "fwupd" received signal SIGINT, Interrupt.
[Switching to Thread 0x7f031bdd7700 (LWP 2259)]
0x00007f032abb88bb in poll () from target:/usr/lib64/libc.so.6
(gdb) continueQuit
(gdb)

hadess, 2017-11-28:
@gicmo on this.

hughsie, 2018-09-12:
@ArtOfSettling do you still get this? I'd be really interested if you could try 1.1.2 as it has a lot of changes compared to 1.0.x, although I admit none that should affect this. Thanks.

superm1, 2018-10-15:
I'm closing older bugs with `needsinfo` tags such as this one.  Can you please retry with a current release (1.1.3) and if the issues continue to exist, please re-open.

Even if the current release is not yet packaged for your distro, you can install it via snap like this:
```
# snap install --classic fwupd
```

closed 2018-10-15 by superm1