fwupd/fwupd #318: appstream-glib 0.7.4 causing CI failures

superm1, 2017-11-15:
appstream-glib 0.7.4 just finished testing migration to debian-testing and CI is starting to show failures.
Builds that passed with 0.7.1 (last version in debian-testing) started to fail now with 0.7.4.

```[0/1] /usr/bin/python3 -u /usr/share/meson/meson test --no-rebuild --print-errorlogs
1/7 fwupd-self-test                         OK       0.12 s
2/7 fu-self-test                            FAIL     0.45 s
3/7 dfu-self-test                           OK       0.34 s
4/7 raspberrypi-self-test                   OK       0.29 s
5/7 nitrokey-self-test                      OK       0.05 s
6/7 udev-self-test                          OK       0.10 s
7/7 unifying-self-test                      OK       0.08 s
OK:         6
FAIL:       1
SKIP:       0
TIMEOUT:    0
The output from the failed tests:
2/7 fu-self-test                            FAIL     0.45 s
--- command ---
qemu-s390x /build/build/src/fu-self-test
--- stdout ---
/fwupd/device-locker{success}: OK
/fwupd/device-locker{fail}: OK
/fwupd/device{metadata}: OK
/fwupd/engine: (/build/build/src/fu-self-test:1148): Fu-DEBUG: loading config values from /build/data/tests/daemon.conf
(/build/build/src/fu-self-test:1148): Fu-DEBUG: using config path of /build/data/tests
(/build/build/src/fu-self-test:1148): Fu-DEBUG: loading config from /build/data/tests/remotes.d/stable.conf
(/build/build/src/fu-self-test:1148): Fu-DEBUG: loading config from /build/data/tests/remotes.d/broken.conf
(/build/build/src/fu-self-test:1148): Fu-DEBUG: loading config from /build/data/tests/remotes.d/testing.conf
(/build/build/src/fu-self-test:1148): Fu-DEBUG: loading quirks from /build/data/tests/quirks.d/tests.quirk
(/build/build/src/fu-self-test:1148): Fu-DEBUG: now 4 quirk entries
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no /tmp/fwupd-self-test/var/lib/fwupd/quirks.d, skipping
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:store-from-file{/tmp/fwupd-self-test/broken.xml.gz}
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:store-from-file{/tmp/fwupd-self-test/stable.xml}
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:store-from-root
(/build/build/src/fu-self-test:1148): As-DEBUG: using icon path /tmp/fwupd-self-test/icons
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:match-addons
(/build/build/src/fu-self-test:1148): As-DEBUG: Emitting ::changed() [from-root]
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:store-from-file{/tmp/fwupd-self-test/testing.xml}
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:store-from-root
(/build/build/src/fu-self-test:1148): As-DEBUG: using icon path /tmp/fwupd-self-test/icons
(/build/build/src/fu-self-test:1148): As-DEBUG: run AsStore:match-addons
(/build/build/src/fu-self-test:1148): As-DEBUG: Emitting ::changed() [from-root]
(/build/build/src/fu-self-test:1148): Fu-DEBUG: adding release 1.2.4 to existing test
(/build/build/src/fu-self-test:1148): Fu-DEBUG: devices now in store:
(/build/build/src/fu-self-test:1148): Fu-DEBUG: 1	test	Test Device [1.2.3,1.2.2,1.2.4,?]
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property Manufacturer=LENOVO
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property EnclosureKind=10
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property Family=ThinkPad T440s
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property ProductName=20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property ProductSku=LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BiosVendor=LENOVO
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BiosVersion=GJET75WW (2.25 )
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BiosMajorRelease=2
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BiosMinorRelease=25
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BaseboardManufacturer=LENOVO
(/build/build/src/fu-self-test:1148): Fu-DEBUG: smbios property BaseboardProduct=20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {d37363b8-5ec4-5725-b618-b75368a1ad28}   <- LENOVO&ThinkPad T440s&20ARS19C0C&LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s&LENOVO&GJET75WW (2.25 )&2&25
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {309d9985-e453-587e-8486-ff7c835a9ef2}   <- LENOVO&ThinkPad T440s&20ARS19C0C&LENOVO&GJET75WW (2.25 )&2&25
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {705f45c6-fbca-5245-b9dd-6d4fab25e262}   <- LENOVO&20ARS19C0C&LENOVO&GJET75WW (2.25 )&2&25
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {3faec92a-3ae3-5744-be88-495e90a7d541}   <- LENOVO&ThinkPad T440s&20ARS19C0C&LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s&LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {660ccba8-1b78-5a33-80e6-9fb8354ee873}   <- LENOVO&ThinkPad T440s&20ARS19C0C&LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {8dc9b7c5-f5d5-5850-9ab3-bd6f0549d814}   <- LENOVO&ThinkPad T440s&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {178cd22d-ad9f-562d-ae0a-34009822cdbe}   <- LENOVO&LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s&LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {da1da9b6-62f5-5f22-8aaa-14db7eeda2a4}   <- LENOVO&LENOVO_MT_20AR_BU_Think_FM_ThinkPad T440s
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {059eb22d-6dc7-59af-abd3-94bbe017f67c}   <- LENOVO&20ARS19C0C&LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {0cf8618d-9eff-537c-9f35-46861406eb9c}   <- LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {f4275c1f-6130-5191-845c-3426247eb6a1}   <- LENOVO&ThinkPad T440s&LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {db73af4c-4612-50f7-b8a7-787cf4871847}   <- LENOVO&ThinkPad T440s
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {5e820764-888e-529d-a6f9-dfd12bacb160}   <- LENOVO&10
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {f8e1de5f-b68c-5f52-9d1a-f1ba52f1f773}   <- LENOVO&LENOVO&20ARS19C0C
(/build/build/src/fu-self-test:1148): Fu-DEBUG: {6de5d951-d755-576b-bd09-c5cf66b27234}   <- LENOVO
(/build/build/src/fu-self-test:1148): As-DEBUG: run FuMain:setup
(/build/build/src/fu-self-test:1148): As-DEBUG: run FuMain:coldplug
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on id{org.freedesktop.fwupd}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on firmware{(null)}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{bootloader}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{vendor-id}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on id{org.freedesktop.fwupd}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on firmware{(null)}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{bootloader}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{vendor-id}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: ignoring 1.2.3 as the same as 1.2.3
(/build/build/src/fu-self-test:1148): Fu-DEBUG: ignoring 1.2.2 as older than 1.2.3
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on id{org.freedesktop.fwupd}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no requirement on firmware{(null)}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{bootloader}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: no parameter given for firmware{vendor-id}
(/build/build/src/fu-self-test:1148): Fu-DEBUG: ignoring 1.2.3 as the same as 1.2.3
(/build/build/src/fu-self-test:1148): Fu-DEBUG: ignoring 1.2.4 as newer than 1.2.3
OK
/fwupd/engine{require-hwid}: (/build/build/src/fu-self-test:1148): Fu-DEBUG: reading /build/build/data/tests/missing-hwid/hwid-1.2.3.cab with 8571 bytes
(/build/build/src/fu-self-test:1148): Fu-DEBUG: Emitting PropertyChanged('Status'='decompressing')
--- stderr ---
**
Fu:ERROR:../src/fu-self-test.c:76:fu_engine_require_hwid_func: assertion failed (error == NULL): failed to extract .cab file: incorrect checksum detected (FwupdError, 7)
-------
Full log written to /build/build/meson-logs/testlog.txt
FAILED: meson-test 
/usr/bin/python3 -u /usr/share/meson/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
```

hughsie, 2017-11-15:
The string is coming from "libgcab/cabinet.c" in the  gcab project -- and if it's s390 specific I assume it's an endian problem. Was gcab updated in Debian at the same time perhaps? Is it also possible to get the *built* `missing-hwid/hwid-1.2.3.cab` from the CI server so we know if it's problem with the generation or the parsing?

superm1, 2017-11-15:
gcab hasn't changed since August. http://tracker.debian.org/pkg/gcab

The CI for s390x actually uses an x86-64 ELF binary to generate the cab file according to dependencies.xml (the s390x dependency isn't installed in the chroot).  So I'd say it's got to be parsing.

Getting the binary from Travis through artifacts (https://docs.travis-ci.com/user/uploading-artifacts/) looks more painful than just doing a local docker build.  It should be reproducible just the same.

superm1, 2017-11-15:
Here's the file from s390x docker build
[hwid-1.2.3.cab.zip](https://github.com/hughsie/fwupd/files/1476241/hwid-1.2.3.cab.zip)

hughsie, 2017-11-15:
That looks valid, so parsing indeed. I'm not immediately sure why this is broken tbh.

superm1, 2017-11-15:
Confirmed the hwid-1.2.3.cab is the same on x86-64 with appstream-glib 0.7.4.

superm1, 2017-11-16:
I checked with downgrading to 0.7.1-2 in the docker container and it works again.  Definitely introduced by appstream-glib 0.7.4.
@ximion any ideas?

hughsie, 2017-11-16:
I don't think this is a problem with appstream-glib -- the reason you see the problem go away when you downgrade that is that we skip the fu_engine_require_hwid_func() test if appstream-glib is less than 0.7.4 -- as before that we didn't have the `hardware` requires type.

I think this is the first time we're actually *decompressing* the .cab file in any of the fwupd tests, the other tests involving .cab files have all been checking signatures. I really think this is a latent bug in gcab that means it's been broken on s390 since the beginning of time. Maybe we should skip this test just on s390 so the CI server continues to be useful in the meantime?

So, gcab: The `Rx` functions in https://github.com/GNOME/gcab/blob/master/libgcab/cabinet.c#L81 don't appear to have any kind of endian-fixing going on, so I'm not hugely surprised it's breaking.

If you tell me how (I'm not super knowledgeable about docker) I'm happy to build a container and start to debug this. I'm kinda hoping sprinkling a few `GUINT32_FROM_LE`'s around will get things working.

superm1, 2017-11-16:
so once you've got docker installed, use this to build the container:
```
# OS=debian-s390x ./contrib/ci/generate_dockerfile.py
# docker build -t fwupd-debian-s390x -f contrib/ci/Dockerfile .
```
then use this to break into it:
```# docker run -i -t -v `pwd`:/build fwupd-debian-s390x bash```

once you're broken in you can run: `# ./contrib/ci/build_debian_s390x.sh` to repro.

cross compiling gcab for some test fixes is going to be a little more challenging, but you should have all the various compiler pieces installed already in the container.
in autotools set up the target to match what we use for cross compiling fwupd (details in _contrib/ci/s390x_cross.txt_)

I think yeah skipping this test for s390x until it's sorted is probably a good idea though so CI does continue to be useful.

hughsie, 2017-11-16:
Thanks for pushing that workaround. I've managed to reproduced this locally:

    # qemu-s390x .libs/gcab -x ../build/build/data/tests/missing-hwid/hwid-1.2.3.cab 
    gcab: error during extraction: incorrect checksum detected

...just looking at fixes now.

hughsie, 2017-11-17:
After a long night of hacking, this is the gcab fix you need to make the tests pass: https://git.gnome.org/browse/gcab/commit/?h=wip/hughsie/meson&id=18557a4c7811355987abf9efe91c61b1d17327b8

I've pinged the gcab maintainer and asked if he wants some help maintaining it. I'll let you know what happens.

superm1, 2017-11-17:
Awesome!
I've filed something with the Debian packager as well to pick up that patch.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=882013

hughsie, 2017-11-17:
I'm planning a new gcab tarball on the 27th as Marc-Andre agreed to let me co-maintain the library and modernize it a bit.

superm1, 2017-11-17:
OK.  Once you land that we should probably change that test for disabling to test the gcab version compiled against instead.

closed 2017-11-17 by hughsie

jtojnar, 2017-11-25:
The tests fail for me on NixOS even with the gcab patched:

```
3/9 fu-self-test                            FAIL     0.18 s

--- command ---
/tmp/nix-build-fwupd-1.0.1.drv-0/fwupd-1.0.1/build/src/fu-self-test
--- stdout ---
/fwupd/device-locker{success}: OK
/fwupd/device-locker{fail}: OK
/fwupd/device{metadata}: OK
/fwupd/engine: 
--- stderr ---

(/tmp/nix-build-fwupd-1.0.1.drv-0/fwupd-1.0.1/build/src/fu-self-test:5160): GLib-WARNING **: GError set over the top of a previous GError or uninitialized memory.
This indicates a bug in someone's code. You must ensure an error is NULL before it's set.
The overwriting error message was: Failed to parse /tmp/fwupd-self-test/stable.xml file: cannot process file of type text/plain
-------
```

hughsie, 2017-11-25:
Can you get a backtrace please. Thanks!

jtojnar, 2017-11-25:
@hughsie How  would I obtain that?

jtojnar, 2017-11-25:
@hughsie 

```
#0  0x00007ffff6c89a61 in _g_log_abort () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#1  0x00007ffff6c8ad7d in g_logv () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#2  0x00007ffff6c8aedf in g_log () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#3  0x00007ffff6c6f7f2 in g_set_error () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#4  0x00007ffff71d99b0 in as_store_from_file_internal () from /nix/store/njgfih4xmi0ng0vy9rh759p2lcfsp312-appstream-glib-0.7.2/lib/libappstream-glib.so.8
#5  0x00007ffff71d9cdf in as_store_from_file () from /nix/store/njgfih4xmi0ng0vy9rh759p2lcfsp312-appstream-glib-0.7.2/lib/libappstream-glib.so.8
#6  0x0000000000416856 in fu_engine_load_metadata_store.isra ()
#7  0x000000000041aa3e in fu_engine_load ()
#8  0x0000000000411acf in fu_engine_func ()
#9  0x00007ffff6caaf3a in g_test_run_suite_internal () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#10 0x00007ffff6caadef in g_test_run_suite_internal () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#11 0x00007ffff6cab0ce in g_test_run_suite () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#12 0x00007ffff6cab0f1 in g_test_run () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#13 0x000000000040c153 in main ()
```

jtojnar, 2017-11-25:
A better trace

```
#0  0x00007efe6c41ea61 in _g_log_abort () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#1  0x00007efe6c41fd7d in g_logv () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#2  0x00007efe6c41fedf in g_log () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#3  0x00007efe6c4047f2 in g_set_error () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#4  0x00007efe6c96e9b0 in as_store_from_file_internal () from /nix/store/njgfih4xmi0ng0vy9rh759p2lcfsp312-appstream-glib-0.7.2/lib/libappstream-glib.so.8
#5  0x00007efe6c96ecdf in as_store_from_file () from /nix/store/njgfih4xmi0ng0vy9rh759p2lcfsp312-appstream-glib-0.7.2/lib/libappstream-glib.so.8
#6  0x0000000000416076 in fu_engine_load_metadata_from_file (error=0x7ffff0ccc570, remote_id=0x1607100 "stable", path=0x1607590 "/tmp/fwupd-self-test/stable.xml", self=0x15f0850) at ../src/fu-engine.c:1585
#7  fu_engine_load_metadata_store (self=self@entry=0x15f0850, error=0x7ffff0ccc690) at ../src/fu-engine.c:1640
#8  0x000000000041a25e in fu_engine_load (self=self@entry=0x15f0850, error=error@entry=0x7ffff0ccc690) at ../src/fu-engine.c:2975
#9  0x00000000004112ef in fu_engine_func () at ../src/fu-self-test.c:177
#10 0x00007efe6c43ff3a in g_test_run_suite_internal () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#11 0x00007efe6c43fdef in g_test_run_suite_internal () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#12 0x00007efe6c4400ce in g_test_run_suite () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#13 0x00007efe6c4400f1 in g_test_run () from /nix/store/0cgwsihhxi6vzix5vdchrvj8m4qzb3r1-glib-2.54.1/lib/libglib-2.0.so.0
#14 0x000000000040b973 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-self-test.c:1020
```

superm1, 2017-11-27:
That particular failure is different then the one that this bug was raised on.  I see that @hughsie added a commit that should fix this one you reported.  if you still have some other problems, can you please open a separate issue with the new stdout/stderr and a backtrace?

Thanks!

apexo, 2017-12-21:
I'm now getting this error (failed to extract .cab file: incorrect checksum detected) when trying to upgrade to firmware 0.2.4.2 on a Dell 9360 (amd64). fwupd 1.0.2-1 with libappstream-glib8=0.7.4-1 on debian sid. The shasums of the .cab and of the extracted (cabextract) firmware.bin match those in the metadata.

superm1, 2017-12-21:
It's a bug in the gcab  0.7-5 upload that migrated. The maintainer has a fix in place for 0.7-6 but it hasn't been uploaded yet. You can build it from their VCS for now.

apexo, 2017-12-21:
Built & installed libgcab-1-0-0 0.7.6-1 from https://anonscm.debian.org/git/collab-maint/gcab.git, restarted fwupd -> works. Thanks!

superm1, 2017-12-21:
Great! Thanks for confirmation. Its probably worth pinging the maintainer asking them to upload.

commit 59cfba61 (2017-11-17): Test version of gcab to determine which tests to run (Closes: #318)

Adjust the disabling of the s390x CI test to only run when on newer
gcab.