fwupd/fwupd #339: Wrong firmware downloaded for Dell XPS 13 ?

jamiedbennett, 2017-12-12:
On a Dell XPS 13 fwupdmgr get-updates shows:

```
XPS 13 9350 TPM 1.2 has firmware updates:
ID:                      com.dell.uefi5034bac4.firmware
GUID:                    d433959e-03ca-524b-92b7-5022eff81a31
Update Version:          5.81.2.1
Update Remote ID:        lvfs
Update Checksum:         SHA1(77ae490f6cf7b03dd6dd485b782bde8274329af7)
Update Location:         https://fwupd.org/downloads/f7375df3c5f903f55ffd64e9ce891da3aa535355-DellTpm1.2_Fw5.81.2.1.cab
XPS 13 9350 System Firmware has firmware updates:
ID:                      com.dell.uefi33773727.firmware
GUID:                    33773727-8ee7-4d81-9fa0-57e8d889e1fa
Update Version:          0.1.5.1
Update Remote ID:        lvfs
Update Checksum:         SHA1(1bb95787c07612137e6cd00ea0bfde61b9a7b7a9)
Update Location:         https://fwupd.org/downloads/8f8215fdf6bb569f98c6ff76b5a48c168b158df8-firmware_XPS_9350_Bios_1.5.1.wu.cab
```
When I try to apply the update through Gnome Software the laptop reboots into the bios flash tool but complains that the firmware is not for this model. It then reboots and boots normally.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.7.1
	gusb:	0.2.11
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
XPS 13 9350 TPM 1.2
  DeviceID:             DELL-d433959e-03ca-524b-92b7-5022eff81a31lu
  Guid:                 d433959e-03ca-524b-92b7-5022eff81a31
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              5.81.0.0
  Created:              2017-12-12

XPS 13 9350 TPM 2.0
  DeviceID:             DELL-b62a2412-5ac4-5350-b16e-7e8f4655d096lu
  Guid:                 b62a2412-5ac4-5350-b16e-7e8f4655d096
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Created:              2017-12-12

XPS 13 9350 Thunderbolt Controller
  DeviceID:             c7010000-0092-8518-222c-f4d45c217118
  Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  DeviceVendor:         Dell
  DeviceVendorId:       TBT:0x00D4
  Version:              15.0
  Created:              2017-12-12

XPS 13 9350 System Firmware
  DeviceID:             UEFI-33773727-8ee7-4d81-9fa0-57e8d889e1fa-dev0
  Guid:                 33773727-8ee7-4d81-9fa0-57e8d889e1fa
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
[... 379 lines cut ...]
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.13.0-17-generic
├── initrd.img-4.13.0-19-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.13.0-17-generic
├── System.map-4.13.0-19-generic
├── vmlinuz-4.13.0-17-generic
├── vmlinuz-4.13.0-17-generic.efi.signed
├── vmlinuz-4.13.0-19-generic
└── vmlinuz-4.13.0-19-generic.efi.signed

```

Please answer the following questions:

- Operating system and version: Ubuntu 17.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2017-12-12:
Just to be clear, when you're applying updates - are you trying to apply both TPM update and UEFI BIOS update at the same time?  Or just the UEFI BIOS update?  If you were doing both, can you please try to only apply the UEFI BIOS update?  That should help clarify this issue.

jamiedbennett, 2017-12-12:
I applied the System update cleanly on its own. The TPM update complains that the TPM is already owned and to clear the owner, then complains it is the wrong software for this device, then fails to update. 

Previously the TPM update would try to be applied first triggering the failure and subsequent reboot.

superm1, 2017-12-12:
Can you confirm if the TPM really is owned?  You can check in BIOS setup on the TPM page.

jamiedbennett, 2017-12-13:
It looks like it is owned by default on this model as I do not remember doing anything in the bios to claim it.The solution was to clear the TPM in the bios which allowed the firmware to be updated as expected.

The main problems were that the first time I tried this the firmware flashing tool must have tweaked something in the bios (turn UEFI off?) which resulted in an unbootable device. I had to reset the bios to defaults to enable a successful boot. The second more minor problem is that the flashing too is incorrectly reporting that the firmware is not for this model when in fact it is.

superm1, 2017-12-13:
Thanks. Did the machine ship with windows or Linux? When shipped with Windows it will own the tpm by default.

superm1, 2017-12-13:
Regarding causing an unbootable machine, there was a problem with fwupdate in 17.10 that has been fixed very recently, a few days ago.

jamiedbennett, 2017-12-13:
It came with Windows 10 to start with.

superm1, 2017-12-13:
Alright, well that explains why it was owned already.  On Linux it won't be owned until provisioned with the various TPM tools.  I'll confer with some folks to see if we should be blocking in mode upgrades too when owned.  From your behavior described, I think the answer is yes.

jamiedbennett, 2017-12-13:
The other issue was that the TPM update failing blocked the system update
from happening which means that no firmware updates occur until you figure
out how to clear the TPM.

<https://github.com/hughsie/fwupd/issues/339#issuecomment-351393428>, or mute
the thread

superm1, 2017-12-13:
Good point.  I think that if we block the TPM update when owned that should fix that problem.

Another solution can be to set priority of the update relative to the plugins (eg if you call `fwupdmgr update` run **UEFI** update before **DELL** updates) and don't allow both **UEFI** and **DELL** to both stage updates for the same reboot. @hughsie what do you think of this?

hughsie, 2017-12-13:
I think blocking the TPM update when owned is probably the most pragmatic.

superm1, 2017-12-13:
I confirmed that we should have been blocking owned upgrades too.  I've filed a PR for this.

superm1, 2017-12-13:
@jamiedbennett thanks for your help in reporting and providing the details that root caused this.  I've committed to master and 0.9.x branches and it will be in the next fwupd releases.

closed 2017-12-13 by superm1

jamiedbennett, 2017-12-13:
@superm1 Thank you.

8none1, 2017-12-14:
@superm1 could the next release be SRU'd in to Xenial as well?

superm1, 2017-12-14:
@8none1 This particular issue won't affect xenial.  It's only an issue when the version of fwupd supports both Dell TPM and UEFI capsule updates.
Dell TPM updates weren't introduced until after the version in xenial.  I'd love to backport the latest release of fwupd into xenial but it's blocked by a mess of other dependencies that need to be backported.  See https://github.com/hughsie/fwupd/issues/301#issuecomment-342153899 for more details on the stuff that needs to come back.  Some of the things that need to come back aren't really SRU candidates either (like glib2).