fwupd/fwupd #343: thunderbolt update failed

superm1, 2017-12-14:
I was trying to upgrade from NVM 26 to NVM 29.  The process seemed to be going OK, but the update never "finished" from the fwupd side.  Fwupd eventually timed out and the controller was gone.
I did the update with a TBT dock connected.  Even after the update was done (timed out) I unplugged and plugged back in the dock and it never showed back up in sysfs.

I'm unsure which is blame (kernel, NVM update, HW, or fwupd) but I think it's good to look at all parts.
I'm working on a pre-release hardware and pre-release NVM, so excuse me if some details are obfuscated.

```shell
$ fwupdmgr --version
client: 0.9.7
daemon: 0.9.7
```

Can't share all of this.  here' the important one:
```shell
$ fwupdmgr get-devices
*** Thunderbolt Controller
  DeviceID:             d8030000-0070-6f18-a324-419709d21a1e
  Guid:                 a35a21e3-7eb7-56d5-89c8-6e8b28d33ac2
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  DeviceVendor:         Dell
  DeviceVendorId:       TBT:0x00D4
  Version:              26.0
  Created:              2017-12-14
```
- Operating system and version: Ubuntu 17.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting? After this happened I rebooted and things look back to normal.  The TBT controller did update to NVM 29.0.

Here's the fwupd daemon log while the update was running:
```
14:32:16:0994 Fu  Called Install(*,7)
14:32:16:0994 Fu  got option reason
14:32:16:0994 Fu  got option filename
14:32:16:0994 Fu  Emitting PropertyChanged('Status'='decompressing')
14:32:16:0994 Fu  Emitting PropertyChanged('Status'='decompressing')
14:32:16:0994 As  removed path prefix for DriverPackage\0x087C.metainfo.xml
14:32:16:0995 As  removed path prefix for DriverPackage\0x087C_nonsecure.bin
14:32:16:0996 As  removed path prefix for DriverPackage\0x087D.metainfo.xml
14:32:16:0996 As  removed path prefix for DriverPackage\0x087D_nonsecure.bin
14:32:16:0997 As  found file 0	0x087C.metainfo.xml
14:32:16:0999 As  found file 1	0x087C_nonsecure.bin
14:32:16:0999 As  found file 2	0x087D.metainfo.xml
14:32:17:0000 As  Emitting ::changed() [add-app]
14:32:17:0000 As  found file 3	0x087D_nonsecure.bin
14:32:17:0002 Fu  Emitting PropertyChanged('Status'='idle')
14:32:17:0002 Fu  Emitting PropertyChanged('Status'='idle')
14:32:17:0002 Fu  firmware archive contained no signature
14:32:17:0004 Fu  no requirement on id{org.freedesktop.fwupd}
14:32:17:0004 Fu  no requirement on firmware{(null)}
14:32:17:0004 Fu  no parameter given for firmware{bootloader}
14:32:17:0005 Fu  no requirement on firmware{vendor-id}
14:32:17:0005 Fu  performing update_prepare() on dell
14:32:17:0005 Dell preparing/cleaning update for thunderbolt
14:32:17:0006 Dell no dock plugged in
14:32:17:0006 Dell Toggled TBT GPIO to 1
14:32:17:0007 Dell SMI execution returned error: -1
14:32:17:0007 Dell Unable to toggle MST hub GPIO to 1
14:32:17:0007 Fu  performing update_prepare() on upower
14:32:17:0007 Fu  FuPending: trying to open database '/var/lib/fwupd/pending.db'
14:32:17:0007 Fu  FuPending: get res
14:32:22:0851 Fu  Emitting PropertyChanged('Status'='device-write')
14:32:22:0851 Fu  Emitting PropertyChanged('Status'='device-write')
14:32:22:0851 Thunderbolt written 0/237568 bytes [0.0%]
14:32:22:0851 Thunderbolt written 4096/237568 bytes [1.7%]
14:32:22:0852 Fu  Emitting PropertyChanged('Percentage'='1%')
14:32:22:0852 Thunderbolt written 8192/237568 bytes [3.4%]
14:32:22:0852 Fu  Emitting PropertyChanged('Percentage'='3%')
14:32:22:0852 Thunderbolt written 12288/237568 bytes [5.2%]
14:32:22:0852 Fu  Emitting PropertyChanged('Percentage'='5%')
[... 286 lines cut ...]
[182129.176192] pcieport 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
[182129.176193] pcieport 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
[182129.176196] pcieport 0000:02:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
[182129.176197] pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
[182129.176199] pcieport 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
[182129.176200] pcieport 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
[182129.176394] pci_bus 0000:02: Allocating resources
[182129.176462] pcieport 0000:02:02.0: bridge window [io  0x1000-0x0fff] to [bus 6b] add_size 1000
[182129.176464] pcieport 0000:02:02.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 6b] add_size 200000 add_align 100000
[182129.176482] pcieport 0000:02:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
[182129.176483] pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
[182129.176485] pcieport 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
[182129.176486] pcieport 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
[182129.176489] pcieport 0000:02:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
[182129.176490] pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
[182129.176492] pcieport 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
[182129.176493] pcieport 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
```

Here was the client while installing:
```
$ fwupdmgr install blah.cab 
Decompressing…         [-                                       ]
Restarting device…     [****************************************]could not detect device after update: timed out while waiting for device
```

gicmo, 2017-12-14:
As far as I can see from the fwupd log, from the thunderbolt plugins POV the devices disconnected but did not reappear. The kernel logs seems to indicate some issues. You did not, by any chance, check sysfs if the devices files were properly created?

superm1, 2017-12-14:
I did check sysfs the device files were missing after update.  So yes, signs are pointing to not fwupd.  @ybernat could this behavior happen if the operating mode for controller didn't match the mode for NVM?  I'm checking with the team what mode that NVM was built for.

If so, then I think we should resume discussion about encoding that mode into the GUID.
Also, loop @westeri

hughsie, 2017-12-14:
> the devices disconnected but did not reappear

This is also my conclusion.

ybernat, 2017-12-14:
@superm1 
That was exactly what I wanted to ask, when I saw the ACPI errors...
The controller is in native mode or legacy (BIOS-assist) mode?
What is the BIOS mode?
Does the installed kernel support native enumeration?

superm1, 2017-12-14:
@ybernat 
* BIOS mode is BIOS assist on this, so controller should have been legacy mode previously.
* The installed kernel should support native (4.15-rc1ish), however since BIOS assist is how the BIOS is configured that might not actually work

I'll have confirmation tomorrow how the NVM was built.  If you can tell me the configuration item from the XML export from the NVM build tool, I might be able to find out sooner (that's archived from the build process).

@hughsie  @gicmo 
@ybernat and I were discussing the schema has to be expanded in fwupd needs to be expanded for accounting for the mode (that's part of why I think fwupd might end up being partially implicated if this caused by mode).

hughsie, 2017-12-14:
> the schema has to be expanded in fwupd needs to be expanded for accounting for the mode

Can't we just *add* another GUID that includes the mode? Or we could use requirements like the new hardware is doing...

superm1, 2017-12-14:
From our last discussions we were leaning on was appending a number to the string to indicate it's in native mode.
Eg: `TBT-$(VID)$(DID)1` for native mode and `TBT-$(VID)$(DID)` for bios assist mode

ybernat, 2017-12-14:
@superm1 
I afraid I'm not familiar with the tool you mentioned.
OTOH, I hope to upload soon a PR for comparing the controller mode to the NVM mode so you'll be able to use for testing it :)

And yes, I forgot the fact that if the BIOS is set for BIOS-assist it'll reset the NVM to this mode too is probably why reboot helped.

@hughsie 
I suggested adding `1` to the end of the identifier we use today for generating the GUID but maybe adding something like `native` would make more sense?

superm1, 2017-12-14:
OK, then I guess I'll find out tomorrow.  It's not immediately obvious which field in the XML leads to native mode.

westeri, 2017-12-15:
```
[182115.608527] ACPI Exception: AE_ALREADY_EXISTS, During name lookup/catalog (20170831/psobject-252)
[182115.608529] ACPI Error: Method parse/execution failed \_GPE.XTBT, AE_ALREADY_EXISTS (20170831/psparse-550)
[182115.608534] ACPI Error: Method parse/execution failed \_GPE.XTBT, AE_ALREADY_EXISTS (20170831/psparse-550)
[182115.608545] ACPI Error: Method parse/execution failed \_GPE._E42, AE_ALREADY_EXISTS (20170831/psparse-550)
[182115.608549] ACPI: Marking method _E42 as Serialized because of AE_ALREADY_EXISTS error
[182115.608552] ACPI Exception: AE_ALREADY_EXISTS, while evaluating GPE method [_E42] (20170831/evgpe-646)
```

These look suspicious. Are you able to send me acpidump of the machine with full dmesg as well? I need to take a deeper look.

westeri, 2017-12-15:
Actually it would be better if you reproduce this directly by poking the nvmem nodes without fwupd because I don't think this has anything to do with fwupd, more like BIOS or kernel issue.

superm1, 2017-12-15:
@ybernat I didn't get confirmation from team yet, but I used Intel NVM reading tool myself and it claims that NVM is indeed set to "native".
Regarding what to append, I think the word native will be easier to understand when debugging in the future.
So BIOS assist ID's (what everything is "today" in the PC space) would be the same: `TBT-$(VID)-$(DID)`.
Native ID's would be: `TBT-$(VID)-$(DID)-native`

So yes, I don't think there is much that kernel can do in this case.  @westeri I guess not much kernel can do in this case.  Once @ybernat's PR is ready for fwupd fwupd can prevent fallout from this.

superm1, 2017-12-15:
Oh and as further evidence of the word _native_ making sense, it's the word that's output in the XML from that Intel tool.  I think it would make sense to match that one and the one that's for native + RTD3 in your PR if possible.

ybernat, 2017-12-16:
1. I'm not sure about the comment for "native + RTD3". On Windows, we don't differentiate between "native" and "native + RTD3" on FW update, as each platform is either "native" or "native + RTD3".
1. I'm not sure what to do on safe-mode. Actually, the question is about Dell plugin that sets the vid/did for TBT on those cases. On the regular case, there is no image validation and the user must select the correct one, but for Dell case, we must generate the correct GUID for selecting the relevant NVM from LVFS. Maybe you want to look there and query the BIOS status somehow.

superm1, 2017-12-16:
1. The image build tool seemed to have an option that showed both native or native +rtd3. Since we archive XML output from it I was thinking to use that output in cab building when the guid was built. If it doesn't matter to flashing then I'll make it binary instead.

2. In safe mode can you tell which mode the controller is operating in? If so that can be metadata passed to Dell plugin. If you can't tell then yes Dell plugin might be able to query bios for the mode, but this interface isn't yet available.

ybernat, 2017-12-16:
2. No. I read it from the NVM in the controller, which is impossible in safe-mode. Even if the driver would expose the info it gets from ICM in sysfs, in safe-mode it wouldn't get this info (for the same reason, the NVM is to be ignored on safe-mode as it isn't trusted or corrupted). Only BIOS info is relevant in this case.

ybernat, 2017-12-16:
Uploaded PR #345

superm1, 2018-01-02:
So some updates from testing PR #345.

I did a test with a BIOS that initialized the controller in native mode.  I confirmed in Windows that it was in native mode.  I also confirmed that fwupd was building a GUID with "-native" in the string.

When I flashed it looked successful, but the controller never came back.  dmesg had some hints.
Same kernel as previous (4.15-rc1ish)
```
Jan  2 16:58:29 test-Precision- kernel: [  180.994151] dpc 0000:00:1b.0:pcie010: DPC containment event, status:0x1f00 source:0x0000
Jan  2 16:58:29 test-Precision- kernel: [  180.994161] pciehp 0000:00:1b.0:pcie004: Slot(20): Link Up
Jan  2 16:58:30 test-Precision- kernel: [  181.100117] pci 0000:01:00.0: [8086:15da] type 01 class 0x060400
Jan  2 16:58:30 test-Precision- kernel: [  181.100217] pci 0000:01:00.0: enabling Extended Tags
Jan  2 16:58:30 test-Precision- kernel: [  181.100387] pci 0000:01:00.0: supports D1 D2
Jan  2 16:58:30 test-Precision- kernel: [  181.100389] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
Jan  2 16:58:30 test-Precision- kernel: [  181.100649] pci 0000:01:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
Jan  2 16:58:30 test-Precision- kernel: [  181.100788] pci 0000:02:00.0: [8086:15da] type 01 class 0x060400
Jan  2 16:58:30 test-Precision- kernel: [  181.100888] pci 0000:02:00.0: enabling Extended Tags
Jan  2 16:58:30 test-Precision- kernel: [  181.101041] pci 0000:02:00.0: supports D1 D2
Jan  2 16:58:30 test-Precision- kernel: [  181.101043] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
Jan  2 16:58:30 test-Precision- kernel: [  181.101205] pci 0000:02:01.0: [8086:15da] type 01 class 0x060400
Jan  2 16:58:30 test-Precision- kernel: [  181.101304] pci 0000:02:01.0: enabling Extended Tags
Jan  2 16:58:30 test-Precision- kernel: [  181.101447] pci 0000:02:01.0: supports D1 D2
Jan  2 16:58:30 test-Precision- kernel: [  181.101449] pci 0000:02:01.0: PME# supported from D0 D1 D2 D3hot D3cold
Jan  2 16:58:30 test-Precision- kernel: [  181.101604] pci 0000:02:02.0: [8086:15da] type 01 class 0x060400
Jan  2 16:58:30 test-Precision- kernel: [  181.101701] pci 0000:02:02.0: enabling Extended Tags
Jan  2 16:58:30 test-Precision- kernel: [  181.101845] pci 0000:02:02.0: supports D1 D2
Jan  2 16:58:30 test-Precision- kernel: [  181.101847] pci 0000:02:02.0: PME# supported from D0 D1 D2 D3hot D3cold
Jan  2 16:58:30 test-Precision- kernel: [  181.102061] pci 0000:01:00.0: PCI bridge to [bus 02-39]
Jan  2 16:58:30 test-Precision- kernel: [  181.102075] pci 0000:01:00.0:   bridge window [io  0x0000-0x0fff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102082] pci 0000:01:00.0:   bridge window [mem 0x00000000-0x000fffff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102093] pci 0000:01:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
Jan  2 16:58:30 test-Precision- kernel: [  181.102098] pci 0000:02:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
Jan  2 16:58:30 test-Precision- kernel: [  181.102112] pci 0000:02:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
Jan  2 16:58:30 test-Precision- kernel: [  181.102125] pci 0000:02:02.0: bridge configuration invalid ([bus 00-00]), reconfiguring
Jan  2 16:58:30 test-Precision- kernel: [  181.102254] pci 0000:03:00.0: [8086:15d9] type 00 class 0x088000
Jan  2 16:58:30 test-Precision- kernel: [  181.102302] pci 0000:03:00.0: reg 0x10: [mem 0x00000000-0x0003ffff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102316] pci 0000:03:00.0: reg 0x14: [mem 0x00000000-0x00000fff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102391] pci 0000:03:00.0: enabling Extended Tags
Jan  2 16:58:30 test-Precision- kernel: [  181.102540] pci 0000:03:00.0: supports D1 D2
Jan  2 16:58:30 test-Precision- kernel: [  181.102542] pci 0000:03:00.0: PME# supported from D0 D1 D2 D3hot D3cold
Jan  2 16:58:30 test-Precision- kernel: [  181.102760] pci 0000:02:00.0: PCI bridge to [bus 03-39]
Jan  2 16:58:30 test-Precision- kernel: [  181.102771] pci 0000:02:00.0:   bridge window [io  0x0000-0x0fff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102778] pci 0000:02:00.0:   bridge window [mem 0x00000000-0x000fffff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102789] pci 0000:02:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
Jan  2 16:58:30 test-Precision- kernel: [  181.102793] pci_bus 0000:03: busn_res: [bus 03-39] end is updated to 03
Jan  2 16:58:30 test-Precision- kernel: [  181.102893] pci 0000:02:01.0: PCI bridge to [bus 04-39]
Jan  2 16:58:30 test-Precision- kernel: [  181.102903] pci 0000:02:01.0:   bridge window [io  0x0000-0x0fff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102910] pci 0000:02:01.0:   bridge window [mem 0x00000000-0x000fffff]
Jan  2 16:58:30 test-Precision- kernel: [  181.102921] pci 0000:02:01.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
Jan  2 16:58:30 test-Precision- kernel: [  181.102924] pci_bus 0000:04: busn_res: [bus 04-39] end is updated to 39
Jan  2 16:58:30 test-Precision- kernel: [  181.102944] pci_bus 0000:3a: busn_res: [bus 3a] end is updated to 3a
Jan  2 16:58:30 test-Precision- kernel: [  181.102950] pci_bus 0000:3a: [bus 3a] partially hidden behind bridge 0000:02 [bus 02-39]
Jan  2 16:58:30 test-Precision- kernel: [  181.102952] pci_bus 0000:3a: [bus 3a] partially hidden behind bridge 0000:01 [bus 01-39]
Jan  2 16:58:30 test-Precision- kernel: [  181.102957] pci_bus 0000:02: busn_res: [bus 02-39] end can not be updated to 3a
[... cut ...]
I rebooted into Windows though and it seemed to recover OK and it was in native mode (checked from message analyzer).

So I guess there is either a system firmware or kernel problem going on still related to native however.

@westeri
So I guess my question would be when the system firmware has configured the controller for native mode (but no RTD3) how should the experience be?  Should I be seeing PCI devices in lspci with nothing plugged in the type C or still need to force power or plug something in to wake?  I was guessing no, that should only be with RTD3.

westeri, 2018-01-03:
It native non-RTD3 mode it should be pretty similar than in BIOS assisted mode.

That said, there is a kernel bug that specifically happens in this mode (it does not affect RTD3 mode) where ACPI Notify() races with PCIe hotplug driver and breaks things. I have partial fix for it already but requires more testing.

superm1, 2018-01-10:
There are some kernel fixes coming later for this, but at least the user space problems are fixed from 1.0.3 release. Closing this issue

closed 2018-01-10 by superm1