fwupd/fwupd #349: Error on Ubuntu Gnome 16.04.3

Aisyk, 2017-12-24:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**0.7.0-0ubuntu4.3**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.179" (uid=1000 pid=27070 comm="fwupdmgr get-devices ") interface="org.freedesktop.fwupd" member="GetDevices" error name="(unset)" requested_reply="0" destination=":1.177" (uid=0 pid=26852 comm="/snap/uefi-fw-tools/10/libexec/fwupd/fwupd ")
**
```

```shell
$ efibootmgr -v
**efibootmgr: EFI variables are not supported on this system.**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**efivar: error listing variables: Function not implemented
**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Ubuntu Gnome 16.04.3 LTS
- How did you install fwupd : `apt-get`
- Have you tried rebooting?
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? Don't have UEFI.

superm1, 2017-12-24:
It looks like you might have installed via a snap. Is that correct? If so on Ubuntu gnome I believe you should just install using apt and not a snap.

hughsie, 2017-12-24:
The snap version of fwupd is 100% unsupported by upstream, sorry.

Aisyk, 2017-12-25:
I have installed fwupd with synaptic, it is the version 0.7.0-0ubuntu4.3 (xenial-updates)

superm1, 2017-12-25:
I think you also installed from snap according to your logs though. Can you look at the output of your installed snaps? It seems like it's a confined snap.

Aisyk, 2017-12-25:
```
sudo snap list

Name                Version          Rev   Developer      Notes
anbox-installer     1                17    morphis        classic
core                16-2.29.4.2      3604  canonical      core
fwloader            0.0.1            8     freshmc        -
gnome-3-26-1604     3.26.0           27    canonical      -
gradio              6.0.2            24    haecker-felix  -
mattermost-desktop  3.7.1            37    snapcrafters   -
uefi-fw-tools       1.5.2-0.7.2+git  10    canonical      -
vectr               0.1.15           2     vectr          -

```

superm1, 2017-12-25:
Yep you have uefi-fw-tools. That's the snap of fwupd.

closed 2017-12-25 by superm1

Aisyk, 2017-12-25:
Ok,

I've unistalled the snap package uefi-fw-tools
It works now, thanks !