fwupd/fwupd #351: Dell XPS 9360 System Update doesn't show in Gnome Software and doesn't update via fwupdmgr

FMuro, 2017-12-30:
As the title says. There was a long time since I last saw a firmware update in Gnome Software, so I check fupupdmgr and saw the new BIOS version 2.4.2. I tried `fwupdmgr update`, which donwloaded the cap file and rebooted. The file was detected by the computer didn't apply the update.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.2
daemon version:	1.0.2
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.7.0
  VersionBootloader:    11.7.0
  Icon:                 computer
  Created:              2017-12-30

XPS 13 9360 System Firmware
  DeviceId:             3f0ecc7823ee99c718693e8ecca38f54e8738dc5
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.3.1
  VersionLowest:        0.2.3.1
  Icon:                 computer
  Created:              2017-12-30

Unknown Device
  DeviceId:             ca9f93cb97772027f45ba73c86cf4980fb02745b
  Guid:                 a2721130-d6d5-51aa-a710-0323525ca469
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2017-12-30
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001
Boot0001* Fedora	HD(1,GPT,4d6ec14e-e854-4f6c-907d-e913a3cab3d3,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-5ffdbc0d-f340-441c-a803-8439c8c0ae10-0
```

```shell
$ tree /boot
/boot
├── 520f44320ac640c68996d2e055036d31
│   └── 4.13.9-300.fc27.x86_64
├── config-4.14.6-300.fc27.x86_64
├── config-4.14.7-300.fc27.x86_64
├── config-4.14.8-300.fc27.x86_64
├── efi
│   ├── 7bae0d8606c34b91902a6129ce7d84e5
│   │   ├── 4.14.6-300.fc27.x86_64
│   │   ├── 4.14.7-300.fc27.x86_64
│   │   └── 4.14.8-300.fc27.x86_64
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTIA32.EFI
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fallback.efi
│   │   │   ├── fbia32.efi
│   │   │   └── fbx64.efi
│   │   └── fedora
│   │       ├── BOOT.CSV
│   │       ├── BOOTIA32.CSV
│   │       ├── BOOTX64.CSV
│   │       ├── fonts
│   │       │   └── unicode.pf2
│   │       ├── fw
│   │       │   └── fwupdate-GfuBQT.cap
│   │       ├── fwupia32.efi
│   │       ├── fwupx64.efi
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── MokManager.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       └── shimx64-fedora.efi
│   ├── mach_kernel
│   └── System
│       └── Library
│           └── CoreServices
│               └── SystemVersion.plist
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-7bae0d8606c34b91902a6129ce7d84e5.img
├── initramfs-4.14.6-300.fc27.x86_64.img
├── initramfs-4.14.7-300.fc27.x86_64.img
├── initramfs-4.14.8-300.fc27.x86_64.img
├── lost+found
├── memtest86+-5.01
├── System.map-4.14.6-300.fc27.x86_64
├── System.map-4.14.7-300.fc27.x86_64
├── System.map-4.14.8-300.fc27.x86_64
├── vmlinuz-0-rescue-7bae0d8606c34b91902a6129ce7d84e5
├── vmlinuz-4.14.6-300.fc27.x86_64
├── vmlinuz-4.14.7-300.fc27.x86_64
└── vmlinuz-4.14.8-300.fc27.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 27
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Preinstalled
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Don't know.
- Is secure boot enabled (only for the UEFI plugin)? Yes.

superm1, 2017-12-31:
There is something wrong with the fwupdate package in Fedora. Please see https://bugzilla.redhat.com/show_bug.cgi?id=1506609
For more details.

FMuro, 2017-12-31:
Oh, I see. Nice to see it's been worked on. BTW could this also be the cause that the update shows up in the command line but not in gnome software?

Klaas-, 2018-01-01:
meanwhile you can download the update from dell.com and apply it manually via uefi boot menu. just put the .exe you downloaded from dell into /boot/efi

at the moment dell.com also has a newer version than fwupd.org (2.5.0 on dell.com)

superm1, 2018-01-02:
@FMuro i would expect that to be a different issue.  You'll want to file a separate bug against the gnome-software bugzilla to debug further.

Since this problem isn't with fwupd, but is with the fwupdate version specifically in Fedora (and also discussed here: https://github.com/rhboot/fwupdate/issues/80) i'm going to close this fwupd issue.

closed 2018-01-02 by superm1

FMuro, 2018-01-02:
Done!

https://bugzilla.gnome.org/show_bug.cgi?id=792136