fwupd/fwupd #356: DELL Precision 7510 on Fedora 27 update fails

saschpe, 2018-01-05:
Hi, similar to issue #54 and #246 I'm struggling to update DELL TPM / System Firmware. This time it's on Fedora 27 with Kernel  4.14.11-300.fc27.x86_64:

```shell
$ fwupdmgr update
Downloading 5.81.2.1 for Precision 7510 TPM 1.2...
Updating 5.81.2.1 on Precision 7510 TPM 1.2...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
Downloading 0.1.15.4 for Precision 7510 System Firmware...
Updating 0.1.15.4 on Precision 7510 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
```

```shell
$ fwupdmgr --version
client version:	1.0.2
daemon version:	1.0.2
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
```

```shell
$ fwupdmgr get-devices
Precision 7510 TPM 1.2
  DeviceId:             30db8a35bc8fc40ebb8fb94bcaa64ae4a5a3e21e
  Guid:                 057c9e26-7e99-5afd-b128-93d91cc0e957
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              5.81.0.0
  Icon:                 computer
  Created:              2018-01-05
  Modified:             2018-01-05

Precision 7510 TPM 2.0
  DeviceId:             d92d077c2126d7757052d14e36d2f39738180b47
  Guid:                 80771650-33a1-5959-b6b6-d3090227fec5
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-01-05

Precision 7510 System Firmware
  DeviceId:             976dbbcfa11ee4c4579cbd5dd7b83ed740c0379a
  Guid:                 5b7ab884-7cf7-489d-ba15-460dffd64aba
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.12.4
  VersionLowest:        0.1.12.4
  Icon:                 computer
  Created:              2018-01-05
  Modified:             2018-01-05

...
```

```shell
$ efibootmgr -v
BootNext: 0001
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,000D,0005,000F,0001
Boot0000  Windows Boot Manager	HD(1,GPT,1a7f9616-05be-413c-9052-500593c31271,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0002* Fedora	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(\EFI\fedora\shim.efi)
Boot0005  Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(847beb21b0e7,0)/IPv6([::]:<->[::]:,0,0)..BO
Boot000D  UEFI: SM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(EFI\boot\bootx64.efi)..BO
Boot000F* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(847beb21b0e7,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-5b7ab884-7cf7-489d-ba15-460dffd64aba-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-057c9e26-7e99-5afd-b128-93d91cc0e957-0
```

```shell
$ tree /boot
/boot/efi
├── a98a208ae616463d9ad38db21ba50988
│   ├── 4.11.10-300.fc26.x86_64
│   ├── 4.11.11-300.fc26.x86_64
│   ├── 4.11.9-300.fc26.x86_64
│   ├── 4.12.13-300.fc26.x86_64
│   ├── 4.12.14-300.fc26.x86_64
│   ├── 4.12.5-300.fc26.x86_64
│   ├── 4.12.8-300.fc26.x86_64
│   ├── 4.12.9-300.fc26.x86_64
│   ├── 4.13.4-200.fc26.x86_64
│   ├── 4.14.11-300.fc27.x86_64
│   ├── 4.14.7-300.fc27.x86_64
│   └── 4.14.8-300.fc27.x86_64
├── EFI
│   ├── BOOT
│   │   ├── BOOTX64.EFI
│   │   ├── fallback.efi
│   │   └── fbx64.efi
│   ├── Dell
│   │   └── Bios
│   │       └── Recovery
│   │           └── BIOS_CUR.RCV
│   └── fedora
│       ├── BOOT.CSV
│       ├── BOOTX64.CSV
│       ├── fonts
│       │   └── unicode.pf2
│       ├── fw
│       │   ├── fwupdate-8VEdvZ.cap
│       │   └── fwupdate-rSC3LA.cap
│       ├── fwupia32.efi
│       ├── fwupx64.efi
│       ├── grub.cfg
│       ├── grubenv
│       ├── grubx64.efi
│       ├── mmx64.efi
│       ├── MokManager.efi
│       ├── shim.efi
│       ├── shimx64.efi
│       └── shimx64-fedora.efi
├── mach_kernel
└── System
    └── Library
        └── CoreServices
            └── SystemVersion.plist
```
Please answer the following questions:

- Operating system and version: Fedora 27 / Kernel 4.14.11-300.fc27.x86_64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? Several times
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

```shell
$ rpm -qa fwupdate-efi fwupd-labels fwupdate-libs fwupd  efivar-libs efibootmgr mokutil pesign
pesign-0.112-20.fc27.x86_64
efibootmgr-15-3.fc27.x86_64
fwupdate-efi-10-0.2.fc27.x86_64
fwupdate-libs-10-0.2.fc27.x86_64
fwupd-labels-1.0.2-1.fc27.x86_64
efivar-libs-32-2.fc27.x86_64
mokutil-0.3.0-7.fc27.x86_64
fwupd-1.0.2-1.fc27.x86_64
```

```shell
$ sudo file /boot/efi/EFI/fedora/fwupx64.efi                                                  
/boot/efi/EFI/fedora/fwupx64.efi: PE32+ executable (EFI application) x86-64 (stripped to external PDB), for MS Windows
$ sudo ls -lah /boot/efi/EFI/fedora/fwupx64.efi                                               
-rwx------. 1 root root 75K 20. Sep 21:56 /boot/efi/EFI/fedora/fwupx64.efi
```

saschpe, 2018-01-05:
When rebooting, I briefly see the updater and it's listing both capsules. I'll check TPM in the BIOS Setup soon.

superm1, 2018-01-05:
By chance can you catch the message it shows with both capsules with camera or a video? If it's too quick may need to turn on more debugging that can slow it down.

superm1, 2018-01-05:
btw the reason i'm looking for that is because you should be hitting the exact same failure as https://bugzilla.redhat.com/show_bug.cgi?id=1506609 with Fedora 27, but you're not.  Additional debug messages should aide in figuring out what's happening.

cobratbq, 2018-01-06:
I seem to have the same issue on a XPS 9360. I had this with previous firmware too. Found out that extracting the firmware package and manually pointing to the firmware file works fine. But I'd like to fix the issue. I'm on Fedora 27, have already tried reinstalling 'fwupd' and 'fwupdate-efi'. TPM on.

I don't think there is an error message visible. For me, at least, it lists 2 entries and a general message "File <...fw-capsule-correctly-written...> searched." Then it becomes dark like it's starting something ... screen brightness ... dark again, then shows Dell logo and continues booting Fedora. As if something was started that finished very quickly.

superm1, 2018-01-06:
So I posted a potential fix and someone has done a copr build on https://bugzilla.redhat.com/show_bug.cgi?id=1506609
Can you please try it?
You'll have to turn off secure boot to try it if you have it on.

cobratbq, 2018-01-06:
That bug fix works. Successfully starts firmware update process and afterwards fwupdmgr results indicate successful upgrade. Behavior is the same again as a few months ago with the 2.2.1 update and before.

superm1, 2018-01-06:
That's great. Can you please leave some comments on the Fedora bug confirming there?

saschpe, 2018-01-08:
👎 Didn't work for me unfortunately.

superm1, 2018-01-08:
@saschpe Sorry to hear that.  If it didn't work for you, can you help provide a few more details?

1. Do you have secure boot enabled?
2. Are you sure that the package was properly installed?  Can you please provide md5 or sha1 of the EFI binary on your EFI system partition?
3. Can you share `efibootmgr -v` output before rebooting after running the update?
4. Please describe your behavior more closely.  Need to make sure it's the same bug.

saschpe, 2018-01-08:
1) No
2) sha1:
92f5815e39bfc4235cbac1ab206d2c1cc9e90ae8  /boot/efi/EFI/fedora/fw/fwupdate-rSC3LA.cap
52271e0f439f76f3ef94ed19e0bde61c8637f978  /boot/efi/EFI/fedora/fw/fwupdate-8VEdvZ.cap
3)
BootNext: 0001
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,000D,0005,000F,0001
Boot0000  Windows Boot Manager	HD(1,GPT,1a7f9616-05be-413c-9052-500593c31271,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0002* Fedora	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(\EFI\fedora\shim.efi)
Boot0005  Onboard NIC(IPV6)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(847beb21b0e7,0)/IPv6([::]:<->[::]:,0,0)..BO
Boot000D  UEFI: SM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,6ed24369-1be2-4cd0-81d5-6647c9947c9e,0x800,0x64000)/File(EFI\boot\bootx64.efi)..BO
Boot000F* Onboard NIC(IPV4)	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(847beb21b0e7,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO

4) Machine reboots, seemingly select the correct BootNext entry. Then a bit of black and a normal reboot. I didn't see the Linux Firmware Manager flashing up this time but it might have been to fast or the display didn't init fast enough. Not sure. After logging in again, fwupdmgr update downloads the same files again thus the update didn't happen.

superm1, 2018-01-08:
So something notable about your boot entries is that it's choosing a different shim for fwupdate than it is for standard fedora.

Can you please provide the output of `tree /boot/efi`?
Also sorry it wasn't clear - i mean the sha1 of `/boot/efi/efi/fedora/fwupx64.efi`.
And confirm you have this version of fwupdate installed: `10-0.2.1.git91351b8.fc27`?

W73, 2018-01-08:
I did update to 2.5.0 on XPS 9360 in the morning on today. Worked smoothly. Thank you @superm1

superm1, 2018-01-08:
OK so since we're confirmed that this is not a fwupd issue and does resolve with my fix i'm closing it against the fwupd project.  @saschpe if things still aren't working when you confirm you're installed the correct version lets debug on a separate issue.  Feel free to "@" me on Github on an issue against https://github.com/rhboot/fwupdate or open another bugzilla entry for Fedora and CC me.

closed 2018-01-08 by superm1

saschpe, 2018-01-09:
@superm1 
1) See above, didn't change
2) d2d0f65f13bb94ff376875eb2d5004635c1a0b69  /boot/efi/efi/fedora/fwupx64.efi
3) Yes:
pesign-0.112-20.fc27.x86_64
efibootmgr-15-3.fc27.x86_64
fwupd-labels-1.0.2-1.fc27.x86_64
efivar-libs-32-2.fc27.x86_64
fwupdate-efi-10-0.2.1.git91351b8.fc27.x86_64
fwupdate-libs-10-0.2.1.git91351b8.fc27.x86_64
mokutil-0.3.0-7.fc27.x86_64
fwupd-1.0.2-1.fc27.x86_64

I did a short video on the boot sequence: https://peilicke.de:5001/mo/sharing/3w4S9KcAz#/item/5229?_k=7rselv (Download link is top right) Not much to see there. Not sure how I can debug this further.

superm1, 2018-01-09:
Hmm, everything there looks correct.  OK so next steps to me:
1. Install the new package with the tagged release (just added to that RHBZ).  I don't think this is going to fix your problem, but you should at least try it.
2. Re-run the systemd unit for cleanup.  This will remove stale binaries and NVRAM entries so they can be rebuilt.  You might have to remove `/var/cache/fwupdate/done `or `/var/lib/fwupdate/done` for it to run (or something similar - I don't know what the path is on Fedora for it).
3. If neither of those help, we'll need to turn on the verbose debugging for the fwupdate EFI application.  I don't know off hand the right way to write out that variable, will need to look into it if 1 or 2 don't help.

cobratbq, 2018-01-09:
FWIW, the difference I see compared to my own case, is that 2 capsules were found i.s.o. only one. It isn't very clear, but it looks like the boot sequence when it skips firmware updating is the same behavior as I experienced. (A bit of screen flashing followed by the normal boot process.)

superm1, 2018-01-09:
You know what - that's a really good point.

When a TPM and Firmware update are run at the same time it should be an EFI scatter/gather list passed to firmware.  If firmware doesn't understand it for some reason that behavior would match.

We've been seeing problems with the UX capsule (See https://github.com/rhboot/fwupdate/issues/79 for more information).
The UX capsule and firmware capsule get passed down together.

So yes, please run the cleanup script that will remove both capsules and try to install just "one".  if that fixes it then we know it should be most likely same root cause as the UX capsule problem.

saschpe, 2018-01-17:
1. I tried with the copr version. Meanwhile an update hit me from the Fedora channel. Supposedly it contains the patch?

```shell
$ rpm -q --changelog fwupdate.x86_64 | head      
* Mo Jan 08 2018 Peter Jones <pjones@redhat.com> - 10-1
- Update to the final released version 10.
- Make everything under /boot/efi be mode 0700, since that's what FAT will
  show anyway.

* Di Sep 12 2017 Peter Jones <pjones@redhat.com> - 10-0.2
- Update for version 10
- test release for ux capsule support; to enable UX capsules define
  LIBFWUP_ADD_UX_CAPSULE=1 in your environment.
```
2. These cache folders where empty for me
3. How do I enable verbose debugging?

I also tried just keeping one capsure but it didn;t work either.

superm1, 2018-01-17:
Yeah that's the fully patched version (10-1).

Regarding the cleanup stuff the cache directory might be somewhere else or the unit isn't running properly on Fedora, can you please try to manually:
```
# rm -f /boot/efi/EFI/fedora/fw/*
# chattr -i /sys/firmware/efi/efivars/fwupdate*
# rm -f /sys/firmware/efi/efivars/fwupdate*
```
Regarding verbose debugging you may refer to https://github.com/rhboot/fwupdate/wiki/Debugging-UEFI-Capsule-updates, specifically this:
```
printf "\x07\x00\x00\x00\x01" > /sys/firmware/efi/efivars/FWUPDATE_VERBOSE-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

To later turn it off:
```
chattr -i /sys/firmware/efi/efivars/FWUPDATE_VERBOSE-0abba7dc-e516-4167-bbf5-4d9d1c739416
rm -f  /sys/firmware/efi/efivars/FWUPDATE_VERBOSE-0abba7dc-e516-4167-bbf5-4d9d1c739416
```