fwupd/fwupd #359: fwupd service startup fails (fwupd 1.0.3)

superm1, 2018-01-10:
This came in from Debian on the new release.

> fwupd startup fails with the default systemd service configuation - the
> service seems to require libffi generated code, and the systemd
> directive 'MemoryDenyWriteExecute=yes' prevents libffi from executing
> generated code.
> 
> Disabling this in the systemd service MemoryDenyWriteExecute allows the
> service startup to proceed as expected.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=886878

hughsie, 2018-01-10:
Why are we using ffi code? Can you reproduce this?

superm1, 2018-01-10:
Hmm, well not in my Debian docker container I can't.  Let me see if I can on a real system.

superm1, 2018-01-10:
I didn't try on Debian unstable, but had ubuntu bionic handy and tried on that.  can't reproduce.  I'll get more details from the reporter.

hughsie, 2018-01-10:
Maybe get the reporter to run valgrind on the daemon too?

superm1, 2018-01-10:
Since it's tied to MemoryDenyWrite=yes, they'd have to valgrind it from systemd right?  Just putting it in the systemd unit launch command should hopefully be enough.

hughsie, 2018-01-10:
Well, MemoryDenyWrite is denying the crazyness (I think correctly) but valgrind on the /usr binary directly should show something funky. The only time I've ever seen ffi issues like this it's been detected as an invalid write by valgrind.

superm1, 2018-01-11:
So they reported back more details.
```
Jan 10 13:45:17 ahoskeri-lnx systemd[1]: Starting Firmware update daemon...
Jan 10 13:45:17 ahoskeri-lnx fwupd[23072]: /usr/lib/fwupd/fwupd: error while loading shared libraries: libffi.so.6: cannot enable executable stack as shared object requires: Operation not permitted
```

Indeed fwupd does link with it (although probably not directly).
```
# ldd /usr/lib/fwupd/fwupd | grep libffi
        libffi.so.6 => /lib/x86_64-linux-gnu/libffi.so.6 (0x00007f60bed3b000)
```
Because I notice also appstream glib does:
```
# ldd /usr/lib/x86_64-linux-gnu/libappstream-glib.so.8.0.10 | grep ffi
        libffi.so.6 => /lib/x86_64-linux-gnu/libffi.so.6 (0x00007f590e748000)
```

And then from glib, gio does.
```
# ldd /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 | grep ffi
        libffi.so.6 => /lib/x86_64-linux-gnu/libffi.so.6 (0x00007efd8bd55000)
```

So reading MemoryDenyWriteExecute at https://www.freedesktop.org/software/systemd/man/systemd.exec.html
I noticed this:
> Note that this option is incompatible with programs and libraries that generate program code dynamically at runtime, including JIT execution engines, executable stacks, and code "trampoline" feature of various C compilers. 

Some of the meltdown and spectre fixes do actually set up a return trampoline.  Is it possible that's causing problems?

hughsie, 2018-01-11:
> Some of the meltdown and spectre fixes do actually set up a return trampoline

Urgh, it could be this. I don't understand enough about compilers and assembly, but it's certainly plausible.

superm1, 2018-01-16:
I tried with a fully updated debian unstable VM but couldn't reproduce it.

hughsie, 2018-01-17:
I think close this, if the reporter comes back with more info we can reopen.

Toniob, 2018-01-31:
Hi,

I've got the same problem. When trying to start fwupd.service, I've got the following logs
> janv. 31 13:45:36 moss systemd[1]: Starting Firmware update daemon...
> janv. 31 13:45:36 moss fwupd[4063]: /usr/lib/fwupd/fwupd: error while loading shared libraries: libffi.so.6: cannot enable executable stack as shared object requires: Operation not permitted
> janv. 31 13:45:36 moss systemd[1]: fwupd.service: Main process exited, code=exited, status=127/n/a
> janv. 31 13:45:36 moss systemd[1]: fwupd.service: Failed with result 'exit-code'.
> janv. 31 13:45:36 moss systemd[1]: Failed to start Firmware update daemon.

If I switch **MemoryDenyWrite** from yes to no in the unit file, the daemon starts just fine.

I'm running debian unstable, fwupd 1.0.4-3, kernel 4.15.0-rc8 (from experimental), all this on an AMD Ryzen processor.

I'm available if you need more informations.

superm1, 2018-01-31:
I'm assuming x86-64 right?
Can you please double check output of this:

```
# execstack -q /lib/x86_64-linux-gnu/libffi.so.6
```

If you haven't intentionally touched the file in any way can you please share the sha1sum of it too?

Lastly where did your `libffi6 `come from?  I see there are newer snapshots in experimental, so I'm wondering if yours possibly was from there too.

Toniob, 2018-02-01:
You're assuming right
> $ uname -a
> Linux moss 4.15.0-rc8-amd64 #1 SMP Debian 4.15~rc8-1~exp1 (2018-01-15) x86_64 GNU/Linux

The result of execstack
> $ execstack -q /lib/x86_64-linux-gnu/libffi.so.6
> X /lib/x86_64-linux-gnu/libffi.so.6

It didn't know it but I've got an old version from experimental : 3.3~20160224-1 (installed in august because of steam from what I see in my logs). So I've just downgraded libffi6 with the package from unstable and ... everything is fine. The daemon starts without any issue.

Problem solved (at least, for me). If you need the version I had from experimental to make some tests, I've still got the deb file.

superm1, 2018-02-01:
Okay well that's good we're closer to the bottom of this.
Can you please try the latest from experimental and see if it happens with that again?

If it does then a bug should be filed against libffi in Debian about this.

Toniob, 2018-02-01:
There is no more package libffi6 in experimental.

superm1, 2018-02-01:
Oh I see it's libffi7 now. If you installed that can you check execstack on that library from it?

Toniob, 2018-02-01:
And here it is with the lastest version of libffi7 from experimental (3.3~20180131-2)
> $ execstack -q /lib/x86_64-linux-gnu/libffi.so.7
> \- /lib/x86_64-linux-gnu/libffi.so.7

superm1, 2018-02-01:
Alright great thanks so it was a problem in experimental on the old version of libffi6 that's been resolved then.

Thanks!

closed 2018-02-01 by superm1