fwupd/fwupd #364: Duplicate unknown device on Dell Precision 5510

nickaubert, 2018-01-12:
`fwupdmgr get-updates` exits because it detects a duplicate device ID:

```shell
$ fwupdmgr get-updates
device ID 8330a096d9f1af8567c7374cb8403e1ce9cf3163 was not unique
device ID 8330a096d9f1af8567c7374cb8403e1ce9cf3163 was not unique
```

I'm running Arch Linux on a Dell Precision 5510.

```shell
$ fwupdmgr --version
client version:	1.0.3
daemon version:	1.0.3
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
```

`get-devices` shows a duplicate Unknown device as well as a duplicate Intel AMT

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.0.26
  VersionBootloader:    11.0.26
  Icon:                 computer
  Created:              2018-01-12

Unknown Device
  DeviceId:             8330a096d9f1af8567c7374cb8403e1ce9cf3163
  Guid:                 2d47f29b-83a2-4f31-a2e8-63474f4d4c2e
  Plugin:               uefi
  Flags:                updatable|locked|supported|registered
  Version:              0
  Icon:                 computer
  Created:              2018-01-12

Synaptics VMM3332 inside Dell WD15/TB16 wired Dock
  DeviceId:             7b3cbc344094ace839263dd4711c84ea99bb6dc3
  Guid:                 cdc9e151-3c55-5f56-adbc-88babd6d358b
  Summary:              Multi-Stream Transport Device
  Plugin:               synapticsmst
  Flags:                updatable|registered
  Vendor:               Synaptics
  Version:              3.10.002
  Icon:                 computer
  Created:              2018-01-12

Dell WD15 Passive Cable
  DeviceId:             d1a437fb5f7f8bffea1b2184e909662d61105384
  Guid:                 fece1537-d683-4ea8-b968-154530bb6f73
  Summary:              A USB type-C docking station
  Plugin:               dell
  Flags:                require-ac|registered
  Vendor:               Dell Inc.
  Version:              0.0.3.18
  Icon:                 computer
  Created:              2018-01-12

Dell WD15 Port Controller 1
  DeviceId:             3ec64779e294dc058f2393efaa5c8db713aaaf67
  Guid:                 8ba2b709-6f97-47fc-b7e7-6a87b578fe25
  Summary:              A USB type-C docking station
  Plugin:               dell
  Flags:                require-ac|registered
  Vendor:               Dell Inc.
  Version:              0.1.1.8
  Icon:                 computer
  Created:              2018-01-12

Dell WD15 
  DeviceId:             8e5d18ee7aa0401befa8a18c4d84c573ee249e22
  Guid:                 e7ca1f36-bf73-4574-afe6-a4ccacabf479
  Summary:              A USB type-C docking station
  Plugin:               dell
  Flags:                require-ac|registered
  Vendor:               Dell Inc.
  Version:              0.0.0.0
  Icon:                 computer
  Created:              2018-01-12

GM107GLM [Quadro M1000M]
  DeviceId:             8e5e2722183462284e8eaa972ef21138395b6adc
  Guid:                 2b07aa6b-6b41-5e92-8f2f-ff742f711a81
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Icon:                 audio-card
  Created:              2018-01-12

HD Graphics 530
  DeviceId:             ca9f93cb97772027f45ba73c86cf4980fb02745b
  Guid:                 8ec3e242-65bd-58ee-9c5e-b28f1e32e7ae
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-01-12

Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.0.26
  VersionBootloader:    11.0.26
  Icon:                 computer
  Created:              2018-01-12

Unknown Device
  DeviceId:             8330a096d9f1af8567c7374cb8403e1ce9cf3163
  Guid:                 2d47f29b-83a2-4f31-a2e8-63474f4d4c2e
  Plugin:               uefi
  Flags:                updatable|locked|supported|registered
  Version:              0
  Icon:                 computer
  Created:              2018-01-12
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0002
Boot0000* Windows Boot Manager	HD(2,GPT,b6f18460-887a-4b62-b2ca-52cc994d11c5,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...,................
Boot0001* arch	HD(1,GPT,de1b47e9-42ae-4bde-b8c7-6a90a6d11c17,0x800,0x2f000)/File(\EFI\arch\grubx64.efi)
```

`efivar -l | grep fw` shows empty, no plugin.

This is Arch Linux with kernel `4.14.13-1-ARCH`.  I noticed this issue after rebooting to the new kernel.  The issue persisted after another reboot.

I am not using an NVMe disk, and secure boot is disabled.  In case it makes a difference, I am encrypting the drive with `dm-crypt`.

I installed fwupd via pacman.
```shell
pacman -Ss fwupd
community/fwupd 1.0.3-1 [installed]
    A simple daemon to allow session software to update firmware
community/fwupdate 9-3 [installed]
    Tools for using the ESRT and UpdateCapsule() to apply firmware updates
```

After I installed, I copied `/usr/lib/fwupdate/EFI/` to `/boot/efi/`
```shell
$ tree /boot/efi
/boot/efi/
└── EFI
    ├── Dell
    │   └── logs
    │       └── diags_current.xml
    └── arch
        ├── fw
        ├── fwupx64.efi
        └── grubx64.efi
```
I understand incomplete install of `fwupd` on Arch has been a problem.  If there was anything else I was supposed to do after pacman install, please let me know.  The arch wiki doesn't contain much information on this.

I'm not sure if this is related, but the `.cab` file for the Precision 5510 laptop I downloaded from Dell also shows as an unkown device.

```shell
$ fwupdmgr get-details e8f31049f612f83358234a67638ec99241c50cd5-firmware_xps_9550_1.6.1.wu.cab
Decompressing…         [***************************************]
Unknown Device
  Guid:                 124c207d-5db8-4d95-bd31-34fd971b34f9
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                none
  
  [Release]
  AppstreamId:          com.dell.uefi124c207d.firmware
  Summary:              Firmware for the Dell XPS 15 9550/Precision 5510
  Version:              0.1.1537
  Filename:             firmware.bin
  Checksum:             SHA1(a2cbcecc4250ce6035bbe6a0392c83bedd811c3b)
  License:              proprietary
  Size:                 10.2 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  TrustFlags:           payload
```

```shell
$ fwupdmgr install packages/dell-precision5510/e8f31049f612f83358234a67638ec99241c50cd5-firmware_xps_9550_1.6.1.wu.cab
Decompressing…         [***************************************]
Firmware is not for any attached hardware: got 124c207d-5db8-4d95-bd31-34fd971b34f9
```

That file came from here: https://fwupd.org/lvfs/device/124c207d-5db8-4d95-bd31-34fd971b34f9

hughsie, 2018-01-12:
I'm very confused how you got a duplicate device, `fu_device_list_add` is supposed to check for duplicates, and the `amt` plugin only ever creates one instance of the device. Can you get the daemon `--verbose` log please? Thanks!

superm1, 2018-01-12:
Also suspicious in addition to the dual amt:
* 2 UEFI devices
* The titles for the UEFI device(s) are wrong
* The version is "0" for the UEFI device(s)

I'd like to also see the verbose log to find out what's going on.

Do you have your old kernel lying around?  I'd like to know if it happens only specifically on this new kernel or it was an upgrade of the kernel and fwupd and other things all at once.
If it's specifically the new kernel only, can you please show the output of this:
```
    # cat /sys/kernel/debug/x86/pti_enabled
    # cat /sys/kernel/debug/x86/ibpb_enabled
    # cat /sys/kernel/debug/x86/ibrs_enabled
```

superm1, 2018-01-12:
Oh... the reason that the UEFI device isn't showing up is that UEFI capsule updates are currently turned off.

It shows up as "Unknown device" rather than the system name. I get it.  We should probably clean that up.  You can run 
```
# fwupdmgr unlock 8330a096d9f1af8567c7374cb8403e1ce9cf3163
```

With a reboot and that should turn them back on.

The duplicate devices and AMT reccoldplugging are showing up due to the re-coldplug from the Dell dock event showing up.  I'm surprised they're not filtered though.

nickaubert, 2018-01-12:
I tried unlocking the device and rebooting, but no change in status.

```shell
$ fwupdmgr unlock 8330a096d9f1af8567c7374cb8403e1ce9cf3163
Authenticating…        [***************************************]
device ID 8330a096d9f1af8567c7374cb8403e1ce9cf3163 was not unique
$ echo $?
1
```
...reboot...
```
$ fwupdmgr update
device ID 8330a096d9f1af8567c7374cb8403e1ce9cf3163 was not unique
device ID 8330a096d9f1af8567c7374cb8403e1ce9cf3163 was not unique
```

I also installed Arch's linux-lts kernel 4.9.75 and I had the same behavior.

nickaubert, 2018-01-12:
The issue is certainly related to the Dell docking device.  If I boot the machine without the dock there are no more duplicate devices.  Unfortunately, this USB-C dock doesn't support hot-plugging from Linux.  I was hoping the newer firmware update would fix that problem.

superm1, 2018-01-12:
The hotplugging fixes for Thunderbolt come in kernel 4.14 IIRC.

Can you please share the log from fwupd verbose output when you run the unlock command?  That should be turning UEFI capsule updates on.

hughsie, 2018-01-12:
> showing up due to the re-coldplug from the Dell dock 

I can look at this tonight.

hughsie, 2018-01-12:
@superm1 what plugins need to support recoldplug? dell? synapticsmst? thunderbolt? thunderbolt-power? I'm tempted to make this opt-in for a complete fix. Most of the USB-based plugins don't need a recoldplug.

superm1, 2018-01-12:
Yeah all of those you outlined should need it.

closed 2018-01-16 by hughsie