fwupd/fwupd #367: how to get dell usb c fix

uriva, 2018-01-14:
In regards to https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1741509
My fwupd version is 0.9.7 and I'm experiencing this bug.
Is there anyway I can get the fix now? (or any idea when it will reach my ubuntu 17.10)
I've tried compiling this repo but it didn't work (meson, ninja.. too complicated).

hughsie, 2018-01-14:
I'm not sure what you're asking us to do...

uriva, 2018-01-14:
ppa?
Also do you have an estimation of when this update might reach me naturally?

superm1, 2018-01-15:
From an upstream perspective there's nothing else we can do but wait.  The fix is already committed upstream.  It's all a Ubuntu problem now.

It's currently blocked by Ubuntu archive administration accepting the upload into artful.  You can view the queue [here](https://launchpad.net/ubuntu/artful/+queue?queue_state=1&queue_text=).

Once accepted there is a minimum 7 day aging period as described [here](https://wiki.ubuntu.com/StableReleaseUpdates).

If you'd like to manually build it, you can follow these steps:
```
dget https://launchpad.net/ubuntu/artful/+upload/17415104/+files/fwupd_0.9.7-2ubuntu1.dsc
cd fwupd-0.9.7
dpkg-buildpackage
```

closed 2018-01-15 by superm1

superm1, 2018-01-15:
FYI, I pinged an archive admin and they accepted into -proposed.  Please follow up on the bug to get them from proposed and test.
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1741509/comments/10

uriva, 2018-01-16:
OK I've somehow installed the latest version, when running `fwupdmgr --version` I'm getting:

> client version:	0.9.7
> daemon version:	0.9.7
> compile-time dependency versions
> 	appstream-glib:	0.7.1
> 	gusb:	0.2.11
> 

But when running `fwupmgr update` I'm now getting 

> No devices can be updated

uriva, 2018-01-16:
Actually this after several iterations each one yielding a different error, partly from the UI software updater and partly from the commandline. Then I rebooted a couple of times, then the update no longer appears and I'm getting the above message.

Since I've updated the last xps 9360 firmware update the screen will not connect via an hdmi adapter connected to the usb c port, if that helps.

uriva, 2018-01-16:
Any command to make sure I have the latest update installed?

superm1, 2018-01-16:
Can you please run the following:

```
# fwupdmgr get-devices
dpkg -l fwupd
```

uriva, 2018-01-16:
> Intel AMT (unprovisioned)
>   DeviceID:             /dev/mei
>   Guid:                 <SOME_ID>
>   Plugin:               amt
>   Flags:                internal|registered
>   Version:              11.6.1
>   VersionBootloader:    11.6.1
>   Created:              2018-01-16
> 
> XPS 13 9360 TPM 1.2
>   DeviceID:             DELL-<SOME_ID>
>   Guid:                 <SOME_ID>
>   Plugin:               dell
>   Flags:                internal|require-ac|supported|registered
>   Version:              5.81.2.1
>   Created:              2018-01-16
> 
> XPS 13 9360 TPM 2.0
>   DeviceID:             DELL-<SOME_ID>
>   Guid:                 <SOME_ID>
>   Plugin:               dell
>   Flags:                internal|require-ac|locked|registered
>   Created:              2018-01-16
> 
> XPS13 9360 Thunderbolt Controller
>   DeviceID:             <SOME_ID>
>   Guid:                 <SOME_ID>
>   Plugin:               thunderbolt
>   Flags:                internal|updatable|supported|registered
>   DeviceVendor:         Dell
>   DeviceVendorId:       TBT:0x00D4
>   Version:              21.0
>   Created:              2018-01-16
> 
> Integrated Webcam HD
>   DeviceID:             usb:00:05
>   Guid:                 <SOME_ID>
>   Guid:                 <SOME_ID>
>   Plugin:               usb
>   Flags:                registered
>   DeviceVendorId:       USB:0x0C45
>   Version:              86.38
>   Created:              2018-01-16
> 
> Microsoft? 2.4GHz Transceiver v8.0
>   DeviceID:             usb:00:02:01:01
>   Guid:                 <SOME_ID>
>   Guid:                 <SOME_ID>
>   Plugin:               usb
>   Flags:                registered
>   DeviceVendorId:       USB:0x045E
>   Version:              6.86
>   Created:              2018-01-16
> 
> HD Graphics 620
>   DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
>   Guid:                 <SOME_ID>
>   Plugin:               udev
>   Flags:                internal|registered
>   DeviceVendor:         Intel Corporation
>   DeviceVendorId:       PCI:0x8086
>   Created:              2018-01-16
> 

and

> Desired=Unknown/Install/Remove/Purge/Hold
> | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
> |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
> ||/ Name           Version      Architecture Description
> +++-==============-============-============-=================================
> ii  fwupd          0.9.7-2ubunt amd64        Firmware update daemon

superm1, 2018-01-16:
According to that your controller is fully updated now and you have the new version.

> XPS13 9360 Thunderbolt Controller
> DeviceID: <SOME_ID>
> Guid: <SOME_ID>
> Plugin: thunderbolt
> Flags: internal|updatable|supported|registered
> DeviceVendor: Dell
> DeviceVendorId: TBT:0x00D4
> **Version: 21.0**
> Created: 2018-01-16

So I'd say the only other thing we can do is check the journal log to see if anything stands out from the errors you were getting.
You can check:
# journalctl -u fwupd.service

uriva, 2018-01-16:
> -- Logs begin at Tue 2018-01-16 12:57:34 IST, end at Tue 2018-01-16 21:38:04 IST. --
> Jan 16 12:58:18 uri-XPS-13-9360 systemd[1]: Starting Firmware update daemon...
> Jan 16 12:58:20 uri-XPS-13-9360 fwupd[1738]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
> Jan 16 12:58:21 uri-XPS-13-9360 fwupd[1738]: disabling plugin because: failed to coldplug uefi: Cannot create fwup iter
> Jan 16 12:58:21 uri-XPS-13-9360 fwupd[1738]: using plugins: altos, amt, colorhug, dell, dfu, ebitdo, steelseries, synapticsmst, thunderbolt, udev, unifying, upower, usb
> Jan 16 12:58:21 uri-XPS-13-9360 fwupd[1738]: Daemon ready for requests
> Jan 16 12:58:22 uri-XPS-13-9360 systemd[1]: Started Firmware update daemon.
>

superm1, 2018-01-16:
Nothing older than that?

uriva, 2018-01-16:
That's all I see there

superm1, 2018-01-16:
Alright, nothing else that can be done here.  If when a future TBT update is posted you continue to have any problems crop up please feel free to file another issue.