fwupd/fwupd #370: Arch CI has started failing

superm1, 2018-01-15:
I don't believe this to be anything that's changed in fwupd recently to be causing this.

Maybe something related to another package in arch.
Any ideas @ArchangeGabriel ?

```
==> Starting check()...
ninja: Entering directory `build'
[0/1] Running all tests.
 1/11 fwupd-exported-api                      OK       0.01 s
 2/11 fwupd-self-test                         OK       0.02 s
 3/11 fu-self-test                            OK       4.49 s
 4/11 dfu-self-test                           OK       0.04 s
 5/11 raspberrypi-self-test                   OK       0.03 s
 6/11 nitrokey-self-test                      OK       0.01 s
 7/11 udev-self-test                          OK       0.02 s
 8/11 unifying-self-test                      OK       0.02 s
 9/11 thunderbolt-self-test                   TIMEOUT 120.09 s
10/11 dell-self-test                          OK       0.04 s
11/11 synapticsmst-self-test                  OK       0.01 s
OK:        10
FAIL:       1
SKIP:       0
TIMEOUT:    1
The output from the failed tests:
 9/11 thunderbolt-self-test                   TIMEOUT 120.09 s
--- command ---
LD_PRELOAD='libumockdev-preload.so.0' /build/build/src/build/plugins/thunderbolt/thunderbolt-self-test
--- stdout ---
/thunderbolt/basic: 
--- stderr ---
ERROR: uevent_sender_send: No such device /sys/devices/domain0
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/nvm_non_active0
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/nvm_active0
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1/nvm_non_active1
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1/nvm_active1
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3/nvm_non_active3
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3/nvm_active3
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1/0-2
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1/0-2/nvm_non_active2
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-1/0-2/nvm_active2
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3/0-4
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3/0-4/nvm_non_active4
ERROR: uevent_sender_send: No such device /sys/devices/domain0/0-0/0-3/0-4/nvm_active4
-------
Full log written to /build/build/src/build/meson-logs/testlog.txt
FAILED: meson-test 
/usr/bin/python -u /usr/sbin/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
==> ERROR: A failure occurred in check().
    Aborting...
The command "if [[ "$OS" == "arch" ]];   then docker run -e CI=true -t -v `pwd`:/build fwupd-$OS ./contrib/ci/build_and_install_pkgs.sh; fi" exited with 2

```

superm1, 2018-01-15:
Worth noting it's probably not the 0.10 release of umockdev alone doing it.  That's in both debian/testing and arch now.  Debian/testing isn't failing.

ArchangeGabriel, 2018-01-16:
Do we have an estimate of when it started to fail?

ArchangeGabriel, 2018-01-16:
At least in the direct dependency tree, umockdev seems to be the only changed package. Will look at the whole tree later when I get time to do so.

superm1, 2018-01-16:
Between January 12 and January 18

superm1, 2018-01-16:
With the same commit:

[Good build](https://travis-ci.org/hughsie/fwupd/jobs/328217488)
[Bad build](https://travis-ci.org/hughsie/fwupd/jobs/329188223)

superm1, 2018-01-16:
Actually from those logs you can see more specific times that the breakage happened.  Between these two dates:
Fri Jan 12 17:58:50 UTC 2018
Mon Jan 15 19:29:24 UTC 2018

ArchangeGabriel, 2018-01-16:
Thanks, those logs are very helpful. Like I expected, it’s systemd (or Mesa does funny things…):
```
-libsystemd-236.0-3
-mesa-17.3.1-2
-systemd-236.0-3
+libseccomp-2.3.2-2
+libsystemd-236.80-1
+mesa-17.3.2-2
+systemd-236.80-1
```
`libseccomp` being a new dependency of systemd.

I’ve would have bet on https://github.com/systemd/systemd-stable/commit/c46365e5c958a79ab12cf164735a6433282ca7df (because “breaks things”), but it was reverted in https://github.com/systemd/systemd-stable/commit/242f14c5a807bb50bfda18568887deae1eb8a3ec which is included in `systemd-236.81-1` which was used by the latest failing build (today’s one).

The only things I see else could be https://github.com/systemd/systemd-stable/commit/d0467ba7ccf9480cda66fd6f3536538839bf8a20 and https://github.com/systemd/systemd-stable/commit/a1cf10a644d809b782a34bab7c37bf06bfaec3ae, but at this point I lack the necessary knowledge and understanding of how things work to get further.

superm1, 2018-01-16:
@ArchangeGabriel since `libseccomp ` is a new dependency was some other option that was previously not enabled turned on now?

Given those were stable commits from systemd that changed, Arch does seem like an early indicator in this instance of an upcoming failure from systemd that hasn't hit Fedora, Ubuntu, Debian, etc.

I don't know a lot about this area either, but that second commit looks suspicious to me.  Is it undoing the mock environment that umockdev setup?

@gicmo any thoughts on this?

superm1, 2018-01-16:
Can you reproduce the failure locally with an updated arch?  Maybe can you bisect the failing commit?

ArchangeGabriel, 2018-01-16:
I can reproduce locally yes (however I’m at one week of uptime, so maybe a reboot could change things), I’ve tried that first as a matter of fact (maybe I should have mentioned it). I have no available time right now to go into bisecting systemd…

superm1, 2018-01-16:
Can you file something with the systemd maintainer in Arch at least?  Perhaps they will be able to comment further and/or recommend what commit they believe is causing it.

gicmo, 2018-01-19:
This breaks for me (bolt) too. Arch has umockdev 0.10 and that has two changes related to thunderbolt (now treated differently): [88d2cc3](https://github.com/martinpitt/umockdev/commit/88d2cc3520c6006244374fdf40feb89fd9c91744), [bd14fb6](https://github.com/martinpitt/umockdev/commit/bd14fb607b99d253204cf7946c8d1da1634f6968)
I might have a closer look later

gicmo, 2018-01-19:
I just tested with a downgraded umockdev on Arch and it seems to not fix it ...

ArchangeGabriel, 2018-01-19:
@gicmo We have already determined that systemd was the culprit, not umockdev. If you want to help and have some free time available, you should try bisecting systemd-stable between the 236.0 and 236.80 releases.

gicmo, 2018-01-19:
@ArchangeGabriel sorry, had the problem come up for me and only skimmed this issue. Having had a quick look at the systemd changes this one seems indeed a good candidate, because umockdev creates a mock sysfs inside `/tmp`: https://github.com/systemd/systemd-stable/commit/a1cf10a644d809b782a34bab7c37bf06bfaec3ae
Might have a look on the weekend

ArchangeGabriel, 2018-01-19:
OK, note that this is the reason of the commit: https://github.com/systemd/systemd/issues/7676.

gicmo, 2018-01-20:
I think I was wrong again, after doing a bit more in-depth analysis I think it is indeed https://github.com/systemd/systemd-stable/commit/d0467ba7ccf9480cda66fd6f3536538839bf8a20 I filed an issue with umockdev because hopefully we can fix it there.

ArchangeGabriel, 2018-02-12:
Update was released thanks to @gicmo: https://github.com/martinpitt/umockdev/releases/tag/0.11. It should be in Arch soon and then tests can be enabled again.

superm1, 2018-02-12:
Looks like 0.11 is in the repos for arch:
https://www.archlinux.org/packages/community/x86_64/umockdev/

So safe to turn on again?

commit b00ae96d (2018-02-13): contrib: re-enable test suite on arch (Fixes #370)

Arch has a new umockdev now with the fixes
commit 91192564 (2018-02-13): contrib: re-enable test suite on arch (Fixes #370)

Arch has a new umockdev now with the fixes
closed 2018-02-13 by superm1