fwupd/fwupd #372: gcab: Uncomparable version string 'UNKNOWN'.

paulmenzel, 2018-01-17:
gcab 0.7 build from the source archive is built incorrectly, that the version string return *UNKNOWN*.

```
$ gcab --version
gcab UNKNOWN
```

Here is the snippet from the configuration.

```
[…]
Native dependency gio-2.0 found: YES 2.54.2
Native dependency gmodule-2.0 found: YES 2.54.2
Native dependency gio-unix-2.0 found: YES 2.54.2
Native dependency polkit-gobject-1 found: YES 0.104
Native dependency gudev-1.0 found: YES 164
Native dependency appstream-glib found: YES 0.7.4
Native dependency gusb found: YES 0.2.11
Native dependency sqlite3 found: YES 3.19.3
Native dependency libarchive found: YES 3.2.1
Native dependency valgrind found: YES 3.13.0
Native dependency libsoup-2.4 found: YES 2.53.2
Native dependency gnutls found: YES 3.4.8
Library gpgme found: YES
Library gpg-error found: YES
Library m found: YES
Native dependency udev found: YES 164
Native dependency uuid found: YES 2.29.
Native dependency libgcab-1.0 found: YES UNKNOWN

Meson encountered an error in file meson.build, line 167, column 21:
Uncomparable version string 'UNKNOWN'.

A full log can be found at /dev/shm/bee-root/fwupd/fwupd-1.0.3-0/build/meson-logs/meson-log.txt
[…]
```

fwupd should gracefully deal with this, and just return a warning.

paulmenzel, 2018-01-17:
Using the version from https://ftp.gnome.org/pub/GNOME/sources/gcab/0.7/gcab-0.7.tar.xz works. Very confusing, as the search engine shows the GitHub URLs first.

hughsie, 2018-01-17:
github.com/GNOME/gcab isn't an official mirror of gnome stuff.

> fwupd should gracefully deal with this, and just return a warning.

It can't, it doesn't know what features to enable.

closed 2018-01-17 by hughsie