fwupd/fwupd #373: python3-gi required despite passing `-Denable-introspection=false`

paulmenzel, 2018-01-17:
fwupd 1.0.3 fails to configure if python3-gi is not installed, but introspection is disabled.

```
$ meson --prefix /usr --buildtype=plain -Denable-introspection=false /dev/shm/bee-root/fwupd/fwupd-1.0.3-0/source
[…]
Found pkg-config: /usr/bin/pkg-config (0.26)
Native dependency gio-2.0 found: YES 2.54.2
Native dependency gmodule-2.0 found: YES 2.54.2
Native dependency gio-unix-2.0 found: YES 2.54.2
Native dependency polkit-gobject-1 found: YES 0.104
Native dependency gudev-1.0 found: YES 164
Native dependency appstream-glib found: YES 0.7.4
Native dependency gusb found: YES 0.2.11
Native dependency sqlite3 found: YES 3.19.3
Native dependency libarchive found: YES 3.2.1
Native dependency valgrind found: YES 3.13.0
Native dependency libsoup-2.4 found: YES 2.53.2
Native dependency gnutls found: YES 3.4.8
Library gpgme found: YES
Library gpg-error found: YES
Library m found: YES
Native dependency udev found: YES 164
Native dependency uuid found: YES 2.29.
Native dependency libgcab-1.0 found: YES 0.7
Native dependency cairo found: YES 1.15.6
Library fontconfig found: YES
Library freetype found: YES

Meson encountered an error in file meson.build, line 180, column 4:
Error encountered: Error: missing dependency python gobject introspection (python3-gi)

A full log can be found at /dev/shm/bee-root/fwupd/fwupd-1.0.3-0/build/meson-logs/meson-log.txt
```

paulmenzel, 2018-01-17:
The error is thrown by `po/test-deps`.

hughsie, 2018-01-17:
Right, we use python3 in various other places in the build system too. The introspection isn't anything to do with the python3 requirement, sorry.

closed 2018-01-17 by hughsie