fwupd/fwupd #378: Dell Precision 5520 update failed

sghosh151, 2018-01-31:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.3
daemon version:	1.0.3
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.6.13
  VersionBootloader:    11.6.13
  Icon:                 computer
  Created:              2018-01-30

Precision 5520 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.2.4
  VersionLowest:        0.1.2.4
  Icon:                 computer
  Created:              2018-01-30

Unknown Device
  DeviceId:             ca9f93cb97772027f45ba73c86cf4980fb02745b
  Guid:                 3ab5fe9e-bef7-5421-920a-ef7301884933
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-01-30

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0008,0000
Boot0000* ubuntu	HD(1,GPT,40bf34c7-0d30-4777-a524-36e7f404b77d,0x800,0xfa000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Fedora	HD(1,GPT,40bf34c7-0d30-4777-a524-36e7f404b77d,0x800,0xfa000)/File(\EFI\fedora\shim.efi)
Boot0002* M.2 PCIe SSD	BBS(HD,SM961 NVMe SAMSUNG 1024GB ,0x0)..BO
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC	BBS(Network,Onboard NIC,0x0)..BO
Boot0006* USB NIC	BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0007* UEFI: SM961 NVMe SAMSUNG 1024GB, Partition 1	HD(1,GPT,a56d2873-82ae-4f3d-9bad-d2121f8308eb,0x800,0xfa000)/File(EFI\boot\bootx64.efi)..BO
Boot0008* CentOS Linux	HD(1,GPT,40bf34c7-0d30-4777-a524-36e7f404b77d,0x800,0x64000)/File(\EFI\centos\shim.efi)
Boot0009* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,40bf34c7-0d30-4777-a524-36e7f404b77d,0x800,0xfa000)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
```

[... 208 lines cut ...]
├── vmlinuz-4.14.13-300.fc27.x86_64
└── vmlinuz-4.14.14-300.fc27.x86_64

19 directories, 99 files

**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 27
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):  dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk?  YES
- Is secure boot enabled (only for the UEFI plugin)? NO

#fwupdmgr update
Downloading 0.1.5.0 for Precision 5520 System Firmware...
Fetching firmware https://fwupd.org/downloads/e69073e5403ee433ddfdd40b53781e91074b07b2-firmware_XPS_9560_1_5_0.wu.cab
Downloading…           [***************************************]
Updating 0.1.5.0 on Precision 5520 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} efivarfs.c:277 efivarfs_del_variable(): unlink failed: Invalid argument	{error #1} lib.c:132 efi_del_variable(): ops->del_variable() failed: Invalid argument	{error #2} efivarfs.c:312 efivarfs_set_variable(): efi_del_variable failed: Invalid argument	{error #3} lib.c:81 _efi_set_variable_mode(): ops->set_variable() failed: Invalid argument	{error #4} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:609 put_info(): efi_set_variable(fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0) failed: Invalid argument	{error #5} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:1920 fwup_set_up_update_with_buf(): put_info failed.: Invalid argument

superm1, 2018-01-31:
It looks like you might have some stale entries from a previous OS possibly.

Can you please share:
1) your previous OS installation history.
2) did you install any fw updates in that previous OS?
3) have you successfully installed a fw update in Fedora yet?
4) do you see any fwupdate-* entries in /sys/firmware/efi/efivars?

superm1, 2018-02-19:
So since there hasn't been any activity on this from OP to further sort this out I'm going to close it.  If you want to re-open with more details please feel free.

sghosh151, 2018-03-02:
Sorry - been away for a bit.

1 Previous OS history - Fedora 26 > upgrade F27
2. Not that I recal explicitly doing like here 
3. No
4. yes
```
$ ls /sys/firmware/efi/efivars/ | grep fwup
fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
```

superm1, 2018-03-02:
OK awesome thanks for the details.

Can you readily reproduce this over and over?

sghosh151, 2018-03-12:
Yes - reproducible consistently

```
$ fwupdmgr get-updates
Precision 5520 System Firmware has firmware updates:
GUID:                    34578c72-11dc-4378-bc7f-b643866f598c
ID:                      com.dell.uefi34578c72.firmware
Update Version:          0.1.6.2
Update Name:             XPS 15 9560/Precision 5520 System Update
Update Summary:          Firmware for the Dell XPS 15 9560/Precision 5520
Update Remote ID:        lvfs
Update Checksum:         SHA1(7935710f16d7ff627e0bdd3fe6a5ada50abfc5db)
Update Location:         https://fwupd.org/downloads/05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab
ID:                      com.dell.uefi34578c72.firmware
Update Version:          0.1.5.0
Update Name:             XPS 15 9560/Precision 5520 System Update
Update Summary:          Firmware for the Dell XPS 15 9560/Precision 5520
Update Remote ID:        lvfs
Update Checksum:         SHA1(dd472551e4c6f80c5cfd0ffcb3f477ef120fe358)
Update Location:         https://fwupd.org/downloads/e69073e5403ee433ddfdd40b53781e91074b07b2-firmware_XPS_9560_1_5_0.wu.cab
ID:                      com.dell.uefi34578c72.firmware
Update Version:          0.1.4.0
Update Name:             XPS 15 9560/Precision 5520 System Update
Update Summary:          Firmware for the Dell XPS 15 9560/Precision 5520
Update Remote ID:        lvfs
Update Checksum:         SHA1(9113a98356abde59cf9bf454ab1313acc570c1ee)
Update Location:         https://fwupd.org/downloads/5b8e9cbc79dd4f37df0f792196f246d9b7da1e0d-firmware_XPS_9560_1_4_0.wu.cab
No upgrades for Precision 5520  Thunderbolt Controller, current is 21.00: 21.00=same
________________________________________________

Devices that were not updated correctly:

 • Precision 5520 System Firmware (0.1.2.4 → 0.1.6.2)

Upload report now? (Requires internet connection) [Y|n]: n

$ curl -C - -O https://fwupd.org/downloads/05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 9784k  100 9784k    0     0  1956k      0  0:00:05  0:00:05 --:--:-- 2123k

$ sha1sum 05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab 
7935710f16d7ff627e0bdd3fe6a5ada50abfc5db  05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab

$ sudo fwupdmgr install 05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab 
[sudo] password for xxx: 
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} efivarfs.c:277 efivarfs_del_variable(): unlink failed: Invalid argument	{error #1} lib.c:132 efi_del_variable(): ops->del_variable() failed: Invalid argument	{error #2} efivarfs.c:312 efivarfs_set_variable(): efi_del_variable failed: Invalid argument	{error #3} lib.c:81 _efi_set_variable_mode(): ops->set_variable() failed: Invalid argument	{error #4} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:609 put_info(): efi_set_variable(fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0) failed: Invalid argument	{error #5} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:1920 fwup_set_up_update_with_buf(): put_info failed.: Invalid argument

$ sudo fwupdmgr install 05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab 
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} efivarfs.c:277 efivarfs_del_variable(): unlink failed: Invalid argument	{error #1} lib.c:132 efi_del_variable(): ops->del_variable() failed: Invalid argument	{error #2} efivarfs.c:312 efivarfs_set_variable(): efi_del_variable failed: Invalid argument	{error #3} lib.c:81 _efi_set_variable_mode(): ops->set_variable() failed: Invalid argument	{error #4} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:609 put_info(): efi_set_variable(fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0) failed: Invalid argument	{error #5} /builddir/build/BUILD/fwupdate-10/linux/libfwup.c:1920 fwup_set_up_update_with_buf(): put_info failed.: Invalid argument
```

sghosh151, 2018-03-12:
Is there any issue with skipping intermediate releases?

superm1, 2018-03-13:
Something odd is going on with your efivarfs filesystem.

There shouldn't be any issue with skipping intermediary releases though.

I think you should go to BIOS setup and try to restore BIOS defaults and see if this helps.
Keep in mind you will probably need to change back over to AHCI mode from RAID mode if you changed this in the past. You will probably also need to manually create a boot entry again.

sghosh151, 2018-03-22:
Reset BIOS to Factory defualts
fwupdmgr install xxxx.cab (1.6.2)
--- No errors
Reboot
-- firmware upgrade in progress 
Reboot
-- F27 came up clean

anyway to debug the efivars filesystem?

Will try the 1.7.1 firmware update after I tweak the BIOS to enable VTd

sghosh151, 2018-03-22:
```
# fwupdmgr get-history
Precision 5520 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|needs-reboot
  Version:              0.1.2.4
  Created:              2018-03-22
  Modified:             2018-03-22
  UpdateState:          needs-reboot
  
  [Release]
  RemoteId:             lvfs
  Version:              0.1.6.2
  Checksum:             SHA1(7935710f16d7ff627e0bdd3fe6a5ada50abfc5db)
  TrustFlags:           none
  CpuArchitecture:      x86_64
  FwupdVersion:         1.0.5
  GUsbVersion:          0.2.11
  SecureBoot:           Enabled
  AppstreamGlibVersion: 0.7.6
  EfivarVersion:        32
  FwupdateVersion:      10
  KernelVersion:        4.15.7-300.fc27.x86_64
  BootTime:             1521686690

```

Interesting that the older firmware entry still says _needs-reboot_

...

# fwupdmgr get-updates
Precision 5520 System Firmware has firmware updates:
GUID:                    34578c72-11dc-4378-bc7f-b643866f598c
ID:                      com.dell.uefi34578c72.firmware
Update Version:          0.1.7.1
Update Name:             XPS 15 9560/Precision 5520 System Update
Update Summary:          Firmware for the Dell XPS 15 9560/Precision 5520
Update Remote ID:        lvfs
Update Checksum:         SHA1(5d8887c2cb54c981393fa110c178810ce895a4b0)
Update Location:         https://fwupd.org/downloads/ab137fa0823d7de159163eb48bdd7a4619aa729b-Signed_1152921504627483679.cab
No upgrades for Precision 5520  Thunderbolt Controller, current is 21.00: 21.00=same

...

closed 2018-03-22 by sghosh151

superm1, 2018-03-26:
So am I correct to understand you've resolved this by the reset BIOS defaults and were able to successfully update then also up to 1.7.1?