fwupd/fwupd #381: since version 1.0.3 PS4 controller is not usable after connection error.

giankun, 2018-02-02:
The bug appeared with 1.0.3 version while the controller was correctly recognized with the previous version. I opened this bug:

[(https://bugs.archlinux.org/task/57262)]

I also opened a thread on the Arch support Forum:

[(https://bbs.archlinux.org/viewtopic.php?id=233814)]

On the Arch linux package and was asked to report upstream had the problem persisted (which it does). 
I attach logs and results from fwupd status. 

As for the required info:

`[PC@User ~]$ fwupdmgr --version
client version:	1.0.4
daemon version:	1.0.4
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
[PC@User ~]$ fwupdmgr get-devices
GF108 [GeForce GT 420]
  DeviceId:             8e5e2722183462284e8eaa972ef21138395b6adc
  Guid:                 f4efc432-13cc-5578-b7f2-7f5673f04fb5
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Icon:                 audio-card
  Created:              2018-02-02

[PC@User ~]$ uname -a
Linux User 4.14.15-1-ARCH #1 SMP PREEMPT Tue Jan 23 21:49:25 UTC 2018 x86_64 GNU/Linux`

PC is not booting through UEFI.

- Operating system and version:

Linux User 4.14.15-1-ARCH #1 SMP PREEMPT Tue Jan 23 21:49:25 UTC 2018 x86_64 GNU/Linux

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):

pacman package installed when installing gnome (gnome software dependency)

- Have you tried rebooting?

Of course

- Are you using an NVMe disk?

No

- Is secure boot enabled (only for the UEFI plugin)?

No UEFI. 

[log_fwupd_02022018.txt](https://github.com/hughsie/fwupd/files/1690745/log_fwupd_02022018.txt)
[fwupd_bug_02022018.txt](https://github.com/hughsie/fwupd/files/1690746/fwupd_bug_02022018.txt)

superm1, 2018-02-05:
My guess is it's caused by https://github.com/hughsie/fwupd/commit/3955e6e97df7eccde222166ed13817b0f0af2ade which matches on the VID/PID of the PS4 controller too.  Can you please share lsusb -v output for your PS4 controller (with fwupd not running)?

giankun, 2018-02-05:
Here:

`````
Bus 002 Device 005: ID 054c:05c4 Sony Corp. DualShock 4 [CUH-ZCT1E]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x054c Sony Corp.
  idProduct          0x05c4 DualShock 4 [CUH-ZCT1E]
  bcdDevice            1.00
  iManufacturer           1 Sony Computer Entertainment
  iProduct                2 Wireless Controller
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     483
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

superm1, 2018-02-05:
Thanks.  it's the exact same VID/PID as the 8bitdo uses.  Since the 8bitdo can't seem to flash in that mode, may as well just block that VID/PID from matching the 8bitdo plugin too.
Can you please try to rebuild with the commit I linked in https://github.com/hughsie/fwupd/pull/383 to confirm it fixes it for you?

closed 2018-02-05 by hughsie

giankun, 2018-02-19:
Sorry for the delay... I managed to build the git version of fwupd today and can confirm the problem is solved.

PR #383 (2018-02-05): plugins/ebitdo: Stop matching DS4 controller VID/PID
This causes problems with the real DS4 controller.
Furthermore in MacOS mode fwupd doesn't actually work properly
anyhow (Closes #381)
commit ead5bf39 (2018-02-05): plugins/ebitdo: Stop matching DS4 controller VID/PID

This causes problems with the real DS4 controller.
Furthermore in MacOS mode fwupd doesn't actually work properly
anyhow (Closes #381)