fwupd/fwupd #382: Daemon restart required to update Thunderbolt controller version

fooishbar, 2018-02-05:
Running on F27, updated to latest as of Feb 5th:
```shell
$ fwupdmgr --version
client version:	1.0.4
daemon version:	1.0.4
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
```

This is purely a cosmetic issue. I can upgrade the firmware on my Thunderbolt controller just fine (e.g. not suffering from the missing wakeup), using `fwupdmgr update` on the command line. The update visibly takes place (monitor connected over Type-C drops link etc), and I get prompted to submit results after.

However, fwupd still reports the old version until the daemon is restarted. Output below:

```shell
strictly:~/tmp/gpuwave-1.0% sudo fwupdmgr get-devices
[... other devices elided ...]
XPS 13 9350 Thunderbolt Controller
  DeviceId:             [...]
  Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported
  Version:              8.0
  Created:              2018-01-31
  Modified:             2018-01-31
  UpdateState:          success
  
  [Release]
  Version:              16.00
  Checksum:             SHA1(62b734f69f93568faa929e16f860f4e7219d72eb)
  TrustFlags:           none
  AppstreamGlibVersion: 0.7.4
  CpuArchitecture:      x86_64
  FwupdVersion:         1.0.4
  GUsbVersion:          0.2.11
  BootTime:             [...]
  KernelVersion:        4.15.0-rc6-drm-tip+

________________________________________________

Devices that have been updated successfully:

 • XPS 13 9350 Thunderbolt Controller (8.0 → 16.00)

Upload report now? (Requires internet connection) [Y|n]: 
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "[elided]",
                           "Metadata" : {
                             "DistroId" : "fedora",
                             "DistroVersion" : "27"
                           },
                           "Reports" : [
                             {
                               "DeviceId" : "[elided]",
                               "Checksum" : "62b734f69f93568faa929e16f860f4e7219d72eb",
                               "UpdateState" : 2,
                               "Guid" : "dbbe0ed9-df91-529c-ac86-8976b78f726d",
                               "Plugin" : "thunderbolt",
                               "VersionOld" : "8.0",
                               "VersionNew" : "16.00",
                               "Flags" : 35,
                               "Created" : [elided],
                               "Modified" : [elided],
                               "Metadata" : {
                                 "AppstreamGlibVersion" : "0.7.4",
                                 "CpuArchitecture" : "x86_64",
                                 "FwupdVersion" : "1.0.4",
                                 "GUsbVersion" : "0.2.11",
                                 "BootTime" : "[elided]",
                                 "KernelVersion" : "4.15.0-rc6-drm-tip+"
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]: 
strictly:~/tmp/gpuwave-1.0% sudo fwupdmgr get-updates
No upgrades for device, current is 0.1.6.1: 0.1.6.1=same, 0.1.5.1=older, 0.1.4.18=older
XPS 13 9350 Thunderbolt Controller has firmware updates:
GUID:                    dbbe0ed9-df91-529c-ac86-8976b78f726d
ID:                      com.dell.tbtdbbe0ed9.firmware
Update Version:          16.00
Update Name:             Thunderbolt NVM for Xps Notebook 9350
Update Summary:          Xps Notebook 9350
Update Remote ID:        lvfs
Update Checksum:         SHA1(62b734f69f93568faa929e16f860f4e7219d72eb)
Update Location:         https://fwupd.org/downloads/75984aeb95d33ed7fab857b18b28ffe317cb870d-8FDJ2_NVM16.00.cab
strictly:~/tmp/gpuwave-1.0% sudo systemctl restart fwupd.service
strictly:~/tmp/gpuwave-1.0% sudo fwupdmgr get-updates
No upgrades for device, current is 0.1.6.1: 0.1.6.1=same, 0.1.5.1=older, 0.1.4.18=older
No upgrades for device, current is 16.0: 16.00=same
strictly:~/tmp/gpuwave-1.0% 
```

hughsie, 2018-02-05:
@gicmo can we get the new version from sysfs straight away? If so, there's a `fu_plugin_update_reload` that needs hooking up in the thunderbolt plugin.

superm1, 2018-02-05:
You should be able to.  When the NVM is flashed the AR controller resets and basically falls off the "thunderbolt bus".  The TBT plugin explicitly waits for it to come back but doesn't try to re-read sysfs values.

At least for me most of the testing done on it has been with `--allow-reinstall` and the same version so I probably wouldn't have noticed this myself.

superm1, 2018-02-05:
`fu_plugin_thunderbolt_wait_for_device` captures the kernel change and add events.  I think it just needs to call out to `fu_plugin_thunderbolt_change` which will re-read the `nvm_version`.

superm1, 2018-02-08:
I did some testing with this on an XPS 9370 going back and forth from NVM 23 to NVM 28 with master.  I couldn't reproduce this behavior.

```
20:12:09:0656 Fu  FuMain: acquired name: org.freedesktop.fwupd
20:12:16:0769 Fu  Called GetDevices()
20:12:16:0771 Fu  Called GetHistory()
20:12:16:0771 Fu  FuHistory: get devices
20:12:16:0772 Fu  FuHistory: got sql result 8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
20:12:16:0772 Fu  FuHistory: got sql result cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
20:12:30:0243 Fu  Called GetDevices()
20:12:30:0245 Fu  Called GetUpgrades(cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61)
20:12:30:0245 Fu  no requirement on id{org.freedesktop.fwupd}
20:12:30:0245 Fu  no requirement on firmware{(null)}
20:12:30:0245 Fu  no parameter given for firmware{bootloader}
20:12:30:0245 Fu  no requirement on firmware{vendor-id}
20:12:30:0246 Fu  Called GetRemotes()
20:12:33:0950 Fu  Called Install(cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61,10)
20:12:33:0950 Fu  got option reason
20:12:33:0950 Fu  got option filename
20:12:33:0951 Fu  Emitting PropertyChanged('Status'='decompressing')
20:12:33:0951 Fu  Emitting PropertyChanged('Status'='decompressing')
20:12:33:0951 Fu  processing folder: 1/1
20:12:33:0951 Fu  processing file: 0x07e6_secure.bin
20:12:33:0951 Fu  processing file: 0x07e6.metainfo.xml
20:12:33:0951 Fu  creating path /var/cache/fwupd
20:12:33:0955 Fu  processing release: 28.00
20:12:33:0956 Fu  processing file: 0x07e6_secure.bin.asc
20:12:33:0957 Fu  Emitting PropertyChanged('Status'='idle')
20:12:33:0957 Fu  Emitting PropertyChanged('Status'='idle')
20:12:33:0977 Fu  Using keyring at /var/lib/fwupd/gnupg
20:12:33:0982 Fu  Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
20:12:33:0989 Fu  importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
20:12:33:0989 Fu  Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Hughski-Limited
20:12:33:0999 Fu  importing key 7E5439F64986F7A9E973809BAD8A528FEC44881E [1] Success
20:12:34:0005 Fu  returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
20:12:34:0005 Fu  marking payload as trusted
20:12:34:0006 Fu  Emitting PropertyChanged('Status'='waiting-for-auth')
20:12:34:0008 Fu  Emitting PropertyChanged('Status'='idle')
20:12:34:0008 Fu  no requirement on id{org.freedesktop.fwupd}
20:12:34:0008 Fu  no requirement on firmware{(null)}
20:12:34:0008 Fu  no parameter given for firmware{bootloader}
20:12:34:0008 Fu  no requirement on firmware{vendor-id}
20:12:34:0008 Fu  performing update_prepare() on dell
20:12:34:0008 Fu  performing update_prepare() on upower
20:12:34:0008 Fu  performing update_prepare() on thunderbolt_power
20:12:34:0009 Fu  FuHistory: remove device cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
20:12:34:0009 Fu  FuHistory: add device cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
20:12:34:0020 Fu  FuHistory: trying to open database '/var/lib/fwupd/pending.db'
20:12:34:0021 Fu  FuHistory: got schema version of 2
20:12:34:0021 Fu  FuHistory: get device
20:12:34:0021 Fu  FuHistory: got sql result cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
20:12:38:0345 Fu  Emitting PropertyChanged('Status'='device-write')
20:12:38:0345 Fu  Emitting PropertyChanged('Status'='device-write')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='1%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='3%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='5%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='6%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='8%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='10%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='12%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='13%')
20:12:38:0345 Fu  Emitting PropertyChanged('Percentage'='15%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='17%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='18%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='20%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='22%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='24%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='25%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='27%')
20:12:38:0346 Fu  Emitting PropertyChanged('Percentage'='29%')
[... 78 lines cut ...]
20:13:14:0716 Fu  performing device_added() on synapticsmst
20:13:14:0716 Fu  ::added 2655d5bb1ae932d1b7359a60931b09519bfc48bd
20:13:15:0143 Thunderbolt uevent for /sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0/0-0/0-3/0-303: add
20:13:15:0143 Thunderbolt adding udev device: 1075e18c-414f-8680-ffff-ffffffffffff at /sys/devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:00.0/0000:05:00.0/domain0/0-0/0-3/0-303
20:13:15:0144 Fu  using 9bb130e39238201970bd39babf7e690d8608078e for 1075e18c-414f-8680-ffff-ffffffffffff:(null)
20:13:15:0144 Fu  using 76cc74d4-f062-5b93-a11c-8d2a58a25848 for TBT-00d4b054
20:13:15:0144 Fu  emit added from thunderbolt: 9bb130e39238201970bd39babf7e690d8608078e
20:13:15:0144 Fu  performing device_added() on dell
20:13:15:0144 Fu  performing device_added() on thunderbolt_power
20:13:15:0144 Fu  performing device_added() on synapticsmst
20:13:15:0145 Fu  ::added 9bb130e39238201970bd39babf7e690d8608078e
20:13:17:0444 Fu  Called GetDevices()
20:13:17:0449 Fu  Called GetUpgrades(cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61)
20:13:17:0449 Fu  no requirement on id{org.freedesktop.fwupd}
[... cut ...]
20:13:17:0449 Fu  no parameter given for firmware{bootloader}
20:13:17:0449 Fu  no requirement on firmware{vendor-id}
20:13:17:0449 Fu  ignoring 28.00 as the same as 28.0
20:14:11:0106 Fu  Called GetDevices()
20:14:11:0108 Fu  Called GetHistory()
20:14:11:0108 Fu  FuHistory: get devices
20:14:11:0108 Fu  FuHistory: got sql result 8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
20:14:11:0108 Fu  FuHistory: got sql result cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
20:14:11:0108 Fu  FuHistory: got sql result cb52b4cd86255a70d1d9e6b09c6bf00b9f876b61
```

superm1, 2018-02-19:
I'm wondering if this might have also been a timing related problem with force power since you were using 4.15 and didn't mention any Thunderbolt devices (just a monitor which IIRC might not wake the TBT controller).

If so the recently commited https://github.com/hughsie/fwupd/commit/7b19f7db5b7f537171a3193ff3981fb890110095 may actually have helped (ensure that the device remained force powered longer).

Since I couldn't reproduce and that commit in master may help i'm going to mark this closed.  If you see another update posted for your notebook please update to fwupd 1.0.6 or master before running the update to confirm.  If more problems come up, please re-open.

closed 2018-02-19 by superm1