fwupd/fwupd #387: master wants to keep reporting the same event over and over

superm1, 2018-02-08:
When I was trying to reproduce #382, I noticed the reporting mechanism was failing in two ways. 
 
1. I started out by removing `/var/lib/fwupd/pending.db`.
2. Then I downgraded from Thunderbolt NVM 28 to NVM 23 (NVM 23 is not on LVFS however).
3. I checked `fwupdmgr get-devices`.  Confirmed it showed update success and didn't want me to report to LVFS
4. I then did `fwupdmgr update`.  It Pulled the update from LVFS, installed it.  Looked successful.
5. Did `fwupdmgr get-updates` to check for updates again.  It nagged me to report the _failed_ update.
I reported it.
6. Did `fwupdmgr get-updates` to check for updates again.  It nagged me to report the _failed_ update.
I reported it.

So it keeps prompting me to report the updates and also it thinks the update failed when it was fine.
fwupd version:
```
client version:	1.0.5
daemon version:	1.0.5
checkout info:	1.0.4-36-gc0cd023
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
```

daemon log:
```
20:29:54:0724 Fu  Verbose debugging enabled (on console 0)
20:29:54:0724 As  run FuEngine:load
20:29:54:0724 Fu  loading config values from /etc/fwupd/daemon.conf
20:29:54:0726 Fu  using config path of /etc/fwupd
20:29:54:0726 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
20:29:54:0726 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
20:29:54:0726 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
20:29:54:0727 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
20:29:54:0727 Fu  using config path of /usr/lib/fwupd
20:29:54:0727 Fu  ordering lvfs=fwupd+1
20:29:54:0727 Fu  ordering lvfs-testing=lvfs+1
20:29:54:0727 As  run FuEngine:load-smbios
20:29:54:0727 As  run FuEngine:load-hwids
20:29:54:0727 Fu  smbios property Manufacturer=Dell Inc.
20:29:54:0727 Fu  smbios property EnclosureKind=9
20:29:54:0727 Fu  smbios property Family=XPS
20:29:54:0727 Fu  smbios property ProductName=XPS 13 9370
20:29:54:0727 Fu  smbios property ProductSku=07E6
20:29:54:0727 Fu  smbios property BiosVendor=Dell Inc.
20:29:54:0727 Fu  smbios property BiosVersion=1.1.3
20:29:54:0727 Fu  smbios property BiosMajorRelease=1
20:29:54:0727 Fu  smbios property BiosMinorRelease=1
20:29:54:0727 Fu  smbios property BaseboardManufacturer=Dell Inc.
20:29:54:0727 Fu  smbios property BaseboardProduct=0173S1
20:29:54:0727 Fu  {18a2a01e-0182-53bd-8176-e928f3b3df3b}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&1.1.3&1&1
20:29:54:0727 Fu  {639ff027-b0cd-5c78-b649-27d331fd1239}   <- Dell Inc.&XPS&XPS 13 9370&Dell Inc.&1.1.3&1&1
20:29:54:0727 Fu  {d21b771c-47e3-5c58-8969-ce6fc60b1aca}   <- Dell Inc.&XPS 13 9370&Dell Inc.&1.1.3&1&1
20:29:54:0727 Fu  {6cb1f0d4-054e-5d7c-a8c8-5358010d9f62}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&173S1
20:29:54:0727 Fu  {d72b8452-80b6-5b66-b95c-8dd75519a30c}   <- Dell Inc.&XPS&XPS 13 9370&7E6
20:29:54:0727 Fu  {184da599-9986-5212-9dba-226d2d233133}   <- Dell Inc.&XPS&XPS 13 9370
20:29:54:0727 Fu  {3701616e-7261-579b-8b9c-ec94e598b8dd}   <- Dell Inc.&7E6&Dell Inc.&173S1
20:29:54:0727 Fu  {74bb5080-7e51-5373-aea1-bfc218a2caaf}   <- Dell Inc.&7E6
20:29:54:0727 Fu  {b84621e1-d31e-5350-af5e-f6be8e73ce0f}   <- Dell Inc.&XPS 13 9370&Dell Inc.&173S1
20:29:54:0727 Fu  {f33883b9-e996-539e-b35a-73568b903d13}   <- Dell Inc.&XPS 13 9370
20:29:54:0727 Fu  {ab9ada1b-9939-56c8-b8a0-47edb7ee0e63}   <- Dell Inc.&XPS&Dell Inc.&173S1
20:29:54:0727 Fu  {61178075-a8fd-563c-9045-44227d8c121f}   <- Dell Inc.&XPS
20:29:54:0727 Fu  {29ca4080-4d68-5395-afe2-35fa9a4f13bf}   <- Dell Inc.&9
20:29:54:0727 Fu  {dd46addd-ee15-5e81-b63f-08f09898c622}   <- Dell Inc.&Dell Inc.&173S1
20:29:54:0727 Fu  {85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Dell Inc.
20:29:54:0727 As  run FuEngine:load-quirks
20:29:54:0727 Fu  loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai-h05.quirk
20:29:54:0728 Fu  now 6 quirk entries
20:29:54:0728 Fu  loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai-h60.quirk
20:29:54:0728 Fu  now 11 quirk entries
20:29:54:0728 Fu  loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
20:29:54:0728 Fu  now 14 quirk entries
[... 1192 lines cut ...]
                           "Reports" : [
                             {
                               "Checksum" : "296e0680554b3208326468041fe810eaaa167644",
                               "UpdateState" : 3,
                               "Guid" : "4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a",
                               "Plugin" : "thunderbolt",
                               "VersionOld" : "23.0",
                               "VersionNew" : "28.00",
                               "Flags" : 35,
                               "Created" : 1518121802,
                               "Modified" : 1518121941,
                               "Metadata" : {
                                 "AppstreamGlibVersion" : "0.7.4",
                                 "CpuArchitecture" : "x86_64",
                                 "FwupdVersion" : "1.0.5",
                                 "GUsbVersion" : "0.2.11",
                                 "BootTime" : "1518111097",
                                 "KernelVersion" : "4.15.2-041502-generic"
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]: Y

```

hughsie, 2018-02-09:
Very odd. I can't see what the problem is. Can you upload your pending.db somewhere please, preferably after the _up_grade. Thanks.

superm1, 2018-02-09:
I've done a few other runs since then that might further confuse the results.  I'll axe it and redo it first.

superm1, 2018-02-09:
Here's the fresh `pending.db`:
https://dell.box.com/s/s5g86w3mzvjnzi4t46yj2lye78k64fok

Something that stands out to me, the versions are 28.0 ->23.0**0** and 23.0 > 28.0**0**.  If you're doing straight string comparisons rather than using `as_util_vercmp` (which I thought I saw from the history PR) that might be leading to problems.

hughsie, 2018-02-09:
> 28.0 ->23.00 and 23.0 > 28.00

Well spotted! I think it's easy enough to fixup the device version, although I'd really much prefer the version string matched in the release and the hardware.

hughsie, 2018-02-09:
i.e. if the thunderbolt version format is xx.yy from Intel then that's fine, but we need to set it accordingly in the thunderbolt plugin. We still need the sanity check tho.

superm1, 2018-02-09:
@ybernat can you confirm if the version format should always be XX.Y or XX.YY might sometimes come?

closed 2018-02-09 by hughsie

superm1, 2018-02-09:
@ybernat it would still be useful to know how this should be set in TBT plugin each time.

ybernat, 2018-02-10:
Is this about what comes from LVFS metadata? Because I think it's better to not assume anything about leading zeros there, as this is manually created by whoever uploads it there, isn't it?
If this is about what comes from sysfs or TBT plugin, I'm surprised by the different between the formats unless this is a mistake in the plugin with using inconsistent format string, which should be fixed then.

Leading zeros aside, the minor version can be a 2-digit number.

hughsie, 2018-02-11:
@ybernat I think the issue is that `nvm_version` from sysfs should return AA.BB format, i.e. `28.00` -- at the moment it's AA.B and I'm pretty sure it's not supposed to be parsed as a float...

ybernat, 2018-02-12:
I guess this follows the way the kernel itself is versioned...
Anyway, this is something we can easily handle in the plugin/fwupd (which also helps covering the case that someone uploads metadata with AA.B format). Isn't it?

hughsie, 2018-02-12:
@ybernat where is the the `nvm_version` actually populated from? If it's 4 bytes in hardware then you need to include the correct number of leading zeros. If it's freeform text (e.g. `28.0\0`) then I guess we have to handle it in the plugin.

ybernat, 2018-02-12:
Depending on the HW type it can be a read from controller flash in a known location or from a message the driver gets from the FW with data about the controller. Anyway, the data is just 2 bytes, one for the major version and one for the minor one, and must be parsed as int (in hex).

The documentation for this entry from Documentation/ABI/testing/sysfs-bus-thunderbolt:

```
What:           /sys/bus/thunderbolt/devices/.../nvm_version
Date:           Sep 2017
KernelVersion:  4.13
Contact:        thunderbolt-software@lists.01.org
Description:    If the device has upgradeable firmware the version
                number is available here. Format: %x.%x, major.minor.
                If the device is in safe mode reading the file returns
                -ENODATA instead as the NVM version is not available.
```

hughsie, 2018-02-12:
@superm1 do you want to parse and output a `%02x.%02x` so the version is always `82.00` in fwupd, or do you want to change your metainfo?

superm1, 2018-02-12:
I'd prefer to see it parsed and always do a `%02x.%02x`.

hughsie, 2018-02-12:
@superm1 okay, i'll do that if you review my LVFS mega-patch :)