fwupd/fwupd #391: `fwupdmgr refresh` fails due to invalid signature

cocreature, 2018-02-09:
Running `fwupdmgr refresh` fails on my system with the following error message:
```shell
$ fwupdmgr refresh
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

This is a new issue, on wednesday (07.02.2018), refreshing worked just fine. I can reproduce this on another machine as well (also Archlinux).

```shell
$ fwupdmgr --version
client version: 1.0.4
daemon version: 1.0.4
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
```

- Operating system and version:
Archlinux 64bit
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
`pacman`
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled (only for the UEFI plugin)?
No

superm1, 2018-02-09:
Yep, I can also reproduce on Ubuntu with master.

hughsie, 2018-02-09:
I'll try regenerating the metadata first as a quick fix.

hughsie, 2018-02-09:
Right, that worked, I'll have a look at the old files and try to work out what went wrong. I have a suspicion it's my fault and the staging LVFS accidentally uploaded metadata with the test key.

hughsie, 2018-02-09:
Nope, it appears the right key was used:

    $ gpg2 --verify firmware.xml.gz.asc firmware.xml.gz
    gpg: Signature made Wed 07 Feb 2018 12:47:04 GMT
    gpg:                using RSA key 48A6D80E4538BAC2
    gpg: BAD signature from "Linux Vendor Firmware Service <sign@fwupd.org>" [unknown]

(I have the public key imported system wide on this machine)

cocreature, 2018-02-09:
Thanks for the fix, that was impressively fast! I’ll leave it up to you to decide when to close the issue.

hughsie, 2018-02-09:
In the event log:

    2018-02-07 12:47:03 | ???.???.???.??? | dellqa | Uploaded file 05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab to stable /lvfs/upload

...which is what triggered the metadata rebuild (correctly).

hughsie, 2018-02-09:
And nothing untoward in the uwsgi logs:

    [pid: 28217|app: 0|req: 1175/1175] ???.???.???.??? () {52 vars in 1302 bytes} [Wed Feb  7 12:47:00 2018] POST /lvfs/upload => generated 317 bytes in 10536 msecs (HTTP/1.1 302) 
4 headers in 526 bytes (2 switches on core 0)
uploading /var/www/lvfs/downloads/firmware-testing.xml.gz as downloads/firmware-testing.xml.gz
uploading /var/www/lvfs/downloads/firmware-testing.xml.gz.asc as downloads/firmware-testing.xml.gz.asc
uploading /var/www/lvfs/downloads/firmware.xml.gz as downloads/firmware.xml.gz
uploading /var/www/lvfs/downloads/firmware.xml.gz.asc as downloads/firmware.xml.gz.asc

hughsie, 2018-02-09:
$ sha1sum firmware.xml.gz*
6443eac9f838c6b12edaaf2fcd56cf140b1e5309  firmware.xml.gz
3268b224447fa12b5dcd95dc8f9c1be59ecf5fe9  firmware.xml.gz.1
e817446b3a96aadd1ad9c5a4de72d20a1f2db1d7  firmware.xml.gz.asc

..where the .1 is after the rebuild.

hughsie, 2018-02-09:
Ohh, that's expected, other firmwares got pushed to stable in the meantime (new `<releases>` are appearing... but that should have synced the cdn too -- but there is nothing in the logs.

hughsie, 2018-02-09:
I can't work this one out -- I'll close and we can reopen if this every happens again.

rpbikker, 2018-02-28:
Please reopen, it's happening again. Since yesterday I get "Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature" on two machines running Fedora 27.

rpbikker, 2018-02-28:
It's gone again. If nobody changed anything, it seems to be something intermittent...

brumle80, 2018-03-13:
reopening due to:

```
fwupdmgr get-updates -v                                                                                                                                                                                                                                                                              16:19:47  ✘ 1 
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
(fwupdmgr:15127): Fu-DEBUG: creating path /root/.cache/fwupdmgr
(fwupdmgr:15127): Fwupd-DEBUG: looking for os-release at /etc/os-release
(fwupdmgr:15127): Fu-DEBUG: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:15127): Fu-DEBUG: progress: 26%
Downloading…           [**********                             ](fwupdmgr:15127): Fu-DEBUG: progress: 54%
Downloading…           [*********************                  ](fwupdmgr:15127): Fu-DEBUG: progress: 81%
Downloading…           [*******************************        ](fwupdmgr:15127): Fu-DEBUG: progress: 100%
Downloading…           [***************************************]
(fwupdmgr:15127): Fu-DEBUG: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:15127): Fu-DEBUG: progress: 100%

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature

```

hughsie, 2018-03-13:
@brumle80 can you upload /root/.cache/fwupdmgr/firmware.xml.gz and /root/.cache/fwupdmgr/firmware.xml.gz to this ticket please.

hughsie, 2018-03-13:
I managed to find someone else having the same problem. This is what I see:

    $ shasum /home/hughsie/.cache/fwupdmgr/lvfs-metadata.xml.gz*
    f14067a80825e321ab36c9c0063e1766d4f78e30  /home/hughsie/.cache/fwupdmgr/lvfs-metadata.xml.gz
    e1a2f0b2402c6e9dc9b3b217366471a362460c4b  /home/hughsie/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc

    $ gpg /home/hughsie/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
    gpg: assuming signed data in `/home/hughsie/.cache/fwupdmgr/lvfs-metadata.xml.gz'
    gpg: Signature made Tue 06 Mar 2018 09:06:39 GMT using RSA key ID 4538BAC2
    gpg: Good signature from "Linux Vendor Firmware Service <sign@fwupd.org>"

Now, the other person provided me with files that were downloaded from the CDN just a few minutes ago:

    shasum /home/hughsie/Downloads/metadata.xml.gz*
    f14067a80825e321ab36c9c0063e1766d4f78e30  /home/hughsie/Downloads/metadata.xml.gz
    e0d1a5cbb1fa4dd0a7f9e79505b1b9bb57d50d28  /home/hughsie/Downloads/metadata.xml.gz.asc

Hmm, so the detached signature is different.

    $ gpg /home/hughsie/Downloads/metadata.xml.gz.asc
    gpg: assuming signed data in `/home/hughsie/Downloads/metadata.xml.gz'
    gpg: Signature made Wed 28 Feb 2018 20:07:37 GMT using RSA key ID 4538BAC2
    gpg: BAD signature from "Linux Vendor Firmware Service <sign@fwupd.org>"

A file generated on the 28th Feb?!

hughsie, 2018-03-13:
Some followup; it seems the CDN is shipping old files for some zones. I've purged all zones manually, and it should "fix" it, although we need some kind of root cause analysis and a plan to detect this again automatically in the future.

brumle80, 2018-03-14:
the purge fixed it as it's working fine now, thanks for the quick resolve.

HawkinsTheWizard, 2018-03-29:
Still/again Getting the same
```
(fwupdmgr:2051): Fu-DEBUG: 16:46:58.763: creating path /root/.cache/fwupdmgr
(fwupdmgr:2051): Fwupd-DEBUG: 16:46:58.765: looking for os-release at /etc/os-release
(fwupdmgr:2051): Fu-DEBUG: 16:46:58.765: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:2051): Fu-DEBUG: 16:46:59.386: progress: 24%
Downloading…           [*********                              ](fwupdmgr:2051): Fu-DEBUG: 16:46:59.386: progress: 50%
Downloading…           [*******************                    ](fwupdmgr:2051): Fu-DEBUG: 16:46:59.444: progress: 76%
Downloading…           [*****************************          ](fwupdmgr:2051): Fu-DEBUG: 16:46:59.444: progress: 100%
Downloading…           [***************************************]
(fwupdmgr:2051): Fu-DEBUG: 16:46:59.449: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /root/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:2051): Fu-DEBUG: 16:46:59.513: progress: 100%

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature

```

hughsie, 2018-04-12:
I've purged manually; which seems to fix things for me too, but we really need a way to purge triggered from the LVFS.

snowyu, 2018-07-10:
Again now:

```bash
> fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature

> shasum ~/.cache/fwupdmgr/lvfs-metadata.xml.gz*
d4a73a0924d9bed830ddff85f748dbe6bb3a3473  fwupdmgr/lvfs-metadata.xml.gz
be5d50b6fd62ed59edd0e10396a4f7499b56dfa8  fwupdmgr/lvfs-metadata.xml.gz.asc
```
It should refresh CDN automatically

hughsie, 2018-07-10:
Different error this time; a permissions error on the server. I think I've fixed up everything now, but I'll do a root cause after breakfast.

hughsie, 2018-07-10:
See https://github.com/hughsie/lvfs-website/issues/138 for the root cause.

ilya40umov, 2019-02-01:
I've run into the same problem today:
```
fwupdmgr refresh 
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

hughsie, 2019-02-01:
Works now, right? I broke the metadata sync when converting everything to Python 3.

ilya40umov, 2019-02-01:
@hughsie ran it a few more times and still the same thing, I'm guessing it's the cdn cache

ilya40umov, 2019-02-01:
@hughsie I've run it now and it has worked. Thanks a quick fix!

tjyrinki, 2019-02-01:
At the moment still broken around here, but probably just a matter of waiting for all of cdn to catch.

laptop006, 2019-02-15:
Broken right now (Accessing a Fastly node in Sydney)

laptop006, 2019-02-15:
... although a retry after a few minutes did work.

madsl, 2019-03-04:
Tested right now:

# fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************] Less than one minute remaining…
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature

jat255, 2019-04-16:
Also having this issue:

```sh
OS: Arch Linux 
Kernel: x86_64 Linux 5.0.7-arch1-1-ARCH
Shell: zsh 5.7.1
Resolution: 3840x1200
DE: KDE 5.56.0 / Plasma 5.15.4
WM: KWin
CPU: Intel Core i7-8650U @ 8x 4.2GHz
```

```
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************] Less than one minute remaining…
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

frankvw2017, 2019-05-23:
Today, 23 May 2019, Ubuntu 18.04:

```
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

laferuben, 2019-06-11:
The same thing, once again. @hughsie

lrvega, 2019-07-18:
I'm seeing the same thing again after enabling `/etc/fwupd/remotes.d/lvfs-testing.conf` :
```
$ sudo fwupdmgr refresh && fwupdmgr update
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Failed to update metadata for lvfs-testing: '48A6D80E4538BAC2' is not a valid signature
```

laptop006, 2019-07-18:
I see two obvious ways to fix this permanently.

1. Embed the signature in the metadata file.
2. Have the signature actually be a meta manifest, at a known name, listing the manifest files which themselves are named by a version, and possibly keep the last two or three so races can be handled.

This, of course, wouldn't be backwards compatible, but at least you could fix it for the future (plus it's a nice chance to make any other backwards incompatible changes that you might want).

hughsie, 2019-07-19:
So, the reason we don't bundle these together is so we can download the *tiny* detached signature, and then if it has changed from what we have already we download the few-hundred-k metadata file. I suppose one thing to check would be the number of days where the metadata *hasn't* changed as it seems to change much more frequently than when LVFS was a little project only used by one or two OEMs. If it indeed changes every day we could just embed the signature into the .xml.gz somehow, or just do a non-detached-signature at the risk of making the metadata harder to check without gpg2.

The best way of doing this is probably like you suggest, so we get the json manifest and use the manifest checksum to do the "has the remote changed" check. Then if changed, grab the metadata and signature which have a unique per-generation filename. Of course, that's 3 requests although with keep-alive maybe that's not such a big deal.

superm1, 2019-07-19:
>Then if changed, grab the metadata and signature which have a unique per-generation filename

And for backwards compatibility you'll need to have latest symlink pointing to the per-generation filename.

fethomm, 2019-07-25:
Same here just now:
 `fwupdmgr get-updates -v  
Firmware-Metadaten wurden seit 30 Tagen nicht aktualisiert und sind möglicherweise nicht auf dem neuesten Stand.

Jetzt aktualisieren? (Erfordert Internetverbindung) [y|N]: y
(fwupdmgr:13132): FuCommon-DEBUG: 07:19:16.491: creating path /root/.cache/fwupdmgr
(fwupdmgr:13132): Fwupd-DEBUG: 07:19:16.491: looking for os-release at /etc/os-release
(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.492: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.gz to /root/.cache/fwupdmgr/lvfs-testing-metadata.xml.gz
Metadaten werden abgerufen https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
(fwupdmgr:13132): GLib-GIO-DEBUG: 07:19:16.499: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.499: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.499: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:13132): GLib-DEBUG: 07:19:16.499: posix_spawn avoided (fd close requested) 
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.499: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.501: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.501: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.501: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.501: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.501: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.502: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.502: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:13132): GLib-GIO-DEBUG: 07:19:16.503: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.516: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.516: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.517: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.517: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:13132): dconf-DEBUG: 07:19:16.517: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:13132): GLib-GIO-DEBUG: 07:19:16.520: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ‘gio-tls-backend’
(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.694: progress: 8%
Herunterladen …          [***                                    ](fwupdmgr:13132): FuMain-DEBUG: 07:19:16.694: progress: 22%
Herunterladen …          [********                               ] Weniger als eine Minute verbleiben…(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.695: progress: 37%
Herunterladen …          [**************                         ] Weniger als eine Minute verbleiben…(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.696: progress: 51%
Herunterladen …          [*******************                    ](fwupdmgr:13132): FuMain-DEBUG: 07:19:16.697: progress: 65%
Herunterladen …          [*************************              ](fwupdmgr:13132): FuMain-DEBUG: 07:19:16.717: progress: 78%
Herunterladen …          [******************************         ](fwupdmgr:13132): FuMain-DEBUG: 07:19:16.718: progress: 93%
Herunterladen …          [************************************   ](fwupdmgr:13132): FuMain-DEBUG: 07:19:16.718: progress: 100%
Herunterladen …          [***************************************]
(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.729: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc to /root/.cache/fwupdmgr/lvfs-testing-metadata.xml.gz.asc
Signatur wird abgerufen https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
(fwupdmgr:13132): FuMain-DEBUG: 07:19:16.758: progress: 100%

Failed to update metadata for lvfs-testing: '48A6D80E4538BAC2' is not a valid signature`

schildbach, 2019-07-25:
I too get

```
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

fethomm, 2019-07-25:
Found the cause for the error in my case. The package _fwupd-amd64-signed_ had been removed on Debian Sid the other day. After reinstalling it, everything worked fine.

lrvega, 2019-07-25:
> Found the cause for the error in my case. The package _fwupd-amd64-signed_ had been removed on Debian Sid the other day. After reinstalling it, everything worked fine.

That worked for me. Thanks!

superm1, 2019-07-25:
@fethomm @Irvega
I don't see any connection between those two actions.  That package is only used when a UEFI capsule is installed, not a metadata refresh.

lrvega, 2019-07-25:
@superm1 then perhaps it was coincidental and the correct metadata/signatures have been pushed?

superm1, 2019-07-25:
I agree it's coincidental; it's likely a timing problem.  See the above comment for details and ideas on how to approach it.
https://github.com/hughsie/fwupd/issues/391#issuecomment-513218878

bestouff, 2019-08-01:
Had it just today:
```
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************] Less than one minute remaining…
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

arcivanov, 2019-08-20:
Same here for me on Fedora 30.

g2p, 2019-09-10:
Got this after upgrading to Ubuntu 19.10 eoan.
For anyone who needs a simple fix: `sudo rm -rf /root/.cache/fwupd`

superm1, 2019-09-10:
`/root/.cache`?  The cache is created by fwupdmgr I thought, so it should be `~/.cache` unless you've been running `fwupdmgr` as root (which generally you shouldn't be doing)

superm1, 2019-09-23:
I'm going to add the regression label to this since this keeps popping up from time to time.  The _proper_ fix is to do something like https://github.com/fwupd/fwupd/issues/391#issuecomment-513218878 which will take some time to do the plumbing on both sides.

gaeljw, 2019-12-06:
`rm -rf ~/.cache/fwupd` worked for me (indeed, not run as root)

LinuxOnTheDesktop, 2020-01-09:
```
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
$ rm -rf ~/.cache/fwupd
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

Dridi, 2020-02-18:
Same for me today, except that I removed this instead:

```
sudo rm -rv /root/.cache/fwupdmgr
```

After attempting a new refresh it failed again, but this time the cache appeared in `/root/.cache/fwupd`, but removing that directory also doesn't make a difference...

kflak, 2020-02-18:
Happened to me as well on arch linux this morning. /root/.cache/fwudmgr doesn't exist on my system, and removing ~/.cache/fwupd/* doesn't help either.

bigfreak85, 2020-02-18:
and next one same problem here@archlinux

adm-bome, 2020-02-18:
And ubuntu[19.10]

azpekt, 2020-02-18:
And another one on pop_os 19.10

akunszt, 2020-02-18:
The same problem here. This might be the root cause, at least on my machine:
```
trantor ~ # LANG=C gpg -v ~/.cache/fwupd/lvfs-metadata.xml.gz.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: armor header: Version: GnuPG v2.0.22 (GNU/Linux)
gpg: assuming signed data in '/root/.cache/fwupd/lvfs-metadata.xml.gz'
gpg: Signature made Mon Feb 17 13:26:24 2020 CET
gpg:                using RSA key 48A6D80E4538BAC2
gpg: using pgp trust model
gpg: BAD signature from "Linux Vendor Firmware Service <sign@fwupd.org>" [unknown]
gpg: binary signature, digest algorithm SHA1, key algorithm rsa2048
```

hughsie, 2020-02-18:
The PKCS signer exploded overnight. I'll do a root cause analysis later this evening, but I've manually restarted it, which caused the signing queue to become unstuck, which caused the cdn sync to hapen successfully. It should all work correctly now.

segadora, 2020-02-18:
Im having the problem now
```
(fwupdmgr:28732): FuMain-DEBUG: 08:51:40.446: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /home/cn/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:28732): FuMain-DEBUG: 08:51:40.450: progress: 100%

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

segadora, 2020-02-18:
works now

[23 more replies]

closed 2020-04-03 by hughsie