fwupd/fwupd #397: `get-updates` says there is an update, but `update` says "no upgrades for device"

maximbaz, 2018-02-12:
```shell
$ fwupdmgr --version
client version: 1.0.4
daemon version: 1.0.4
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
```

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.6.29
  VersionBootloader:    11.6.29
  Icon:                 computer
  Created:              2018-02-12

Precision 5520 TPM 2.0
  DeviceId:             de98f10c9b146ba4ced2fc37cad31ee5b8808f6d
  Guid:                 8bc551f7-f7ca-5ea8-a4cb-38ae16ecfa17
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.1.0
  Icon:                 computer
  Created:              2018-02-12

Precision 5520 TPM 1.2
  DeviceId:             e0527307c0aa37127c77c5c1f777cbe348591d1a
  Guid:                 61c84b2d-1924-5867-91e2-63ae6a4373af
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-02-12

Precision 5520 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  Icon:                 computer
  Created:              2018-02-12
  Modified:             2018-02-12
  UpdateState:          failed
  UpdateError:          failed to run update on reboot

GM107GLM [Quadro M1200 Mobile]
  DeviceId:             8e5e2722183462284e8eaa972ef21138395b6adc
  Guid:                 f201385b-abb8-522b-b393-a11c2bdb4ca8
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Icon:                 audio-card
  Created:              2018-02-12

Unknown Device
  DeviceId:             ca9f93cb97772027f45ba73c86cf4980fb02745b
  Guid:                 3ab5fe9e-bef7-5421-920a-ef7301884933
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-02-12

Precision 5520  Thunderbolt Controller
  DeviceId:             e6496053b475140ff702f37620e8c1028617dac7
  Guid:                 c5364422-2ec3-5092-b30e-03bdb1a53e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.0
  Icon:                 computer
  Created:              2018-02-12
```

```shell
$ efibootmgr -v
BootNext: 0003
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0003
Boot0000* Windows Boot Manager  HD(1,GPT,4ff8a9b2-1fe3-430e-b8dd-ef7ee5145358,0x800,0x1f4000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* arch  HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\grubx64.efi)
Boot0002* arch  PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-B3-71-B2-1A-24)/HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\grubx64.efi)
Boot0003* Linux-Firmware-Updater \fwupx64.efi   HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\fwupx64.efi)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0
```

```shell
$ tree /boot/efi/EFI/arch
/boot/efi/EFI/arch
├── fw
│   └── fwupdate-Wxz1mG.cap
├── fwupx64.efi
└── grubx64.efi
```

Please answer the following questions:

- Operating system and version: `Arch Linux 4.15.2-2-ARCH`
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman`
- Have you tried rebooting? `yes`
- Are you using an NVMe disk? `yes`
- Is secure boot enabled (only for the UEFI plugin)? `yes`

----------------

```bash
$ fwupdmgr refresh
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc

$ fwupdmgr get-updates
ignoring de98f10c9b146ba4ced2fc37cad31ee5b8808f6d [Precision 5520 TPM 2.0] as not updatable
Precision 5520 System Firmware has firmware updates:
GUID:                    34578c72-11dc-4378-bc7f-b643866f598c
ID:                      com.dell.uefi34578c72.firmware
Update Version:          0.1.6.2
Update Name:             XPS 15 9560/Precision 5520 System Update
Update Summary:          Firmware for the Dell XPS 15 9560/Precision 5520
Update Remote ID:        lvfs
Update Checksum:         SHA1(7935710f16d7ff627e0bdd3fe6a5ada50abfc5db)
Update Location:         https://fwupd.org/downloads/05e0d0906098a57c4d0aa71d5f5fc4e620e7ee03-Signed_1152921504627420111.cab
No upgrades for device, current is 21.0: 21.00=same

$ fwupdmgr update
ignoring de98f10c9b146ba4ced2fc37cad31ee5b8808f6d [Precision 5520 TPM 2.0] as not updatable
Downloading 0.1.6.2 for Precision 5520 System Firmware...
Updating 0.1.6.2 on Precision 5520 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
No upgrades for device, current is 21.0: 21.00=same
```

`get-updates` says that `0.1.6.2` is available, `get-devices` says I have `0.1.5.0` installed, and yet `update` says that there are no upgrades for the device.

I am not sure where the `21.00` comes from. I tried removing `/boot/efi/EFI/arch/fw` folder and trying again, tried rebooting a couple of times, nothing helped.

This inconsistency makes me think that there is a bug somewhere in the code. In meantime can you suggest a workaround perhaps?

superm1, 2018-02-12:
I think what needs to be done is the area that "No upgrades for device" is mentioned should call out the device name rather than "for device".

The 21.0 is coming from the Thunderbolt controller.
```
Precision 5520  Thunderbolt Controller
  DeviceId:             e6496053b475140ff702f37620e8c1028617dac7
  Guid:                 c5364422-2ec3-5092-b30e-03bdb1a53e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.0
  Icon:                 computer
  Created:              2018-02-12
```

maximbaz, 2018-02-12:
Ah, so that's unrelated and I do get to the update stage, but it fails...

I tried once again to be sure, rebooted and this is what I get on `get-device`:

```
Devices that were not updated correctly:

 • Precision 5520 System Firmware (0.1.5.0 → 0.1.6.2)

Upload report now? (Requires internet connection) [Y|n]:
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "794e4f277116d3ddfedc1f6c50c4a67e96e8b5f1bd73b1e31ea00acd5802f816",
                           "Metadata" : {
                             "DistroId" : "arch"
                           },
                           "Reports" : [
                             {
                               "DeviceId" : "f49999b6e3af5b4bf8ad1f2ed2922b76df457271",
                               "Checksum" : "7935710f16d7ff627e0bdd3fe6a5ada50abfc5db",
                               "UpdateState" : 3,
                               "UpdateError" : "failed to run update on reboot",
                               "Guid" : "34578c72-11dc-4378-bc7f-b643866f598c",
                               "Plugin" : "uefi",
                               "VersionOld" : "0.1.5.0",
                               "VersionNew" : "0.1.6.2",
                               "Flags" : 299,
                               "Created" : 1518468237,
                               "Modified" : 1518469014,
                               "Metadata" : {
                                 "AppstreamGlibVersion" : "0.7.4",
                                 "CpuArchitecture" : "x86_64",
                                 "FwupdVersion" : "1.0.4",
                                 "GUsbVersion" : "0.2.11",
                                 "BootTime" : "1518468207",
                                 "KernelVersion" : "4.15.2-2-ARCH"
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]:
```

Are there any more detailed logs to investigate the root cause? I'm pretty sure I upgraded to `0.1.5.0` using `fwupdmgr` a little while ago and I didn't experience any issues.

superm1, 2018-02-12:
Yeah, I submitted a PR to clarify that for the next fwupd version.

Well so if you're on Arch there are some known issues.  They're fixed in a new fwupdate version (10) but IIRC that's not yet in Arch yet (see https://bugs.archlinux.org/task/55815 which is blocked by https://bugs.archlinux.org/task/5704 which is blocked by https://github.com/rhboot/efivar/issues/96)

So have you already rebooted?  With that report I'm guessing so.  If you did reboot, can you describe what you saw?

What version of fwupdate do you have installed?  It's not part of the report by default until next fwupd version.

We can at least see if it's the same as those known issues or if you found something else new.

closed 2018-02-12 by superm1

maximbaz, 2018-02-12:
> can you describe what you saw?

This was a good question and a perfect clue - I did not see absolutely anything, as if `fwupd` is not installed. The culprit appeared to be Secure Boot - once I disabled it, the upgrade started. This makes sense, I only recently started using Secure Boot.

I haven't verified this yet, but I believe it didn't work because I use `cryptboot` and although I signed `grubx64.efi` as described in the documentation, I didn't sign the `fwupx64.efi` and thus Secure Boot prevented loading `fwupx64.efi`.

To answer the rest of the questions, I am on Arch and I have `fwupdmgr v1.0.4` and `fwupdate v9-3` from Arch repos, I didn't experience any issues with efi paths, probably because I use the default `/boot/efi`. 

Thanks for the hints and the improved message!

superm1, 2018-02-12:
@maximbaz OK wonderful, thanks for confirming.  I think it would also be useful to indicate the status of secure boot in our failure reports for next version.  I'll see what we can do.

commit 44ebdb82 (2018-02-12): fu-engine: Clarify which devices don't have upgrades (Fixes #397)