fwupd/fwupd #403: Building in sandbox fails due to missing /etc/machine-id

jtojnar, 2018-02-14:
I am trying to build fwupd with tests on NixOS and `fwupd-self-test` fails because `/etc/machine-id` is not available in the build sandbox.

```
 2/11 fwupd-self-test                         FAIL     0.13 s

--- command ---
/build/fwupd-1.0.4/build/libfwupd/fwupd-self-test
--- stdout ---
(/build/fwupd-1.0.4/build/libfwupd/fwupd-self-test:20134): Fwupd-DEBUG: D-Bus system bus unavailable, skipping tests.
/fwupd/enums: OK
/fwupd/common{machine-hash}: 
--- stderr ---
**
Fwupd:ERROR:../libfwupd/fwupd-self-test.c:384:fwupd_common_machine_hash_func: assertion failed (error == NULL): Failed to open file ?/etc/machine-id?: No such file or directory (g-file-error-quark, 4)
-------
```

- fwupd: 1.0.4

superm1, 2018-02-14:
Can you touch it during your build script that prepares the sandbox?

We do this for example with arch:
https://github.com/hughsie/fwupd/blob/master/contrib/ci/Dockerfile-arch.in

jtojnar, 2018-02-14:
No, the sandbox does not allow writing to arbitrary paths.

superm1, 2018-02-14:
Considering the other issue you filed - this begs the question is the daemon also unable to read /etc/machine-id? There are many other expectations about accessing things on filesystem. If fwupd daemon is not able to get to them I would expect other failures too.

closed 2018-02-14 by hughsie

jtojnar, 2018-02-14:
Is it normal that the test is listed as OK instead of SKIP?

```
 1/11 fwupd-exported-api                      OK       0.01 s
 2/11 fwupd-self-test                         OK       0.02 s
 3/11 fu-self-test                            OK       4.91 s
 4/11 dfu-self-test                           OK       0.03 s
 5/11 raspberrypi-self-test                   OK       0.03 s
 6/11 nitrokey-self-test                      OK       0.01 s
 7/11 udev-self-test                          OK       0.03 s
 8/11 unifying-self-test                      OK       0.02 s
 9/11 thunderbolt-self-test                   OK       5.65 s
10/11 dell-self-test                          OK       0.02 s
11/11 synapticsmst-self-test                  OK       0.02 s

OK:        11
FAIL:       0
SKIP:       0
TIMEOUT:    0
```

superm1, 2018-02-14:
AFICS the entire fwupd-self-test isn't skipped, just one of the tests in it.  If you manually run `fwupd-self-test `you should see the one test skipped.

commit bd608257 (2018-02-14): libfwupd: Allow skipping machine ID self test (Fixes #403)

Some sandbox environments won't allow creating this file, so skip
this test.

We're still running it on all Travis CI test cases though.