fwupd/fwupd #405: Installed tests assertion failed

jtojnar, 2018-02-14:
I am trying to run the installed tests in a container but some assertions fail and the tests just hang until `gnome-desktop-testing` times out, apparently during “Installing test firmware...”:

```
machine: connected to guest root shell
machine# [    5.856486] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input4
machine# [    5.974096] ppdev: user-space parallel port driver
machine# [    6.092852] dbus[577]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service'
machine# [    6.103867] systemd[1]: Starting Firmware update daemon...
machine# [    6.130509] systemd-logind[574]: Watching system buttons on /dev/input/event0 (AT Translated Set 2 keyboard)
machine# [    6.133982] systemd-logind[574]: Watching system buttons on /dev/input/event2 (Power Button)
machine# [    6.196961] systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
machine# [    6.203952] systemd-vconsole-setup[726]: /nix/store/77k738g0yav6flccf9smw0lp5ralfr6y-kbd-2.0.4/bin/loadkeys failed with error code 1.
machine# [    6.205849] systemd-vconsole-setup[726]: KD_FONT_OP_GET failed while trying to get the font metadata: Input/output error
machine# [    6.209437] systemd-vconsole-setup[726]: Fonts will not be copied to remaining consoles
machine# [    6.215942] systemd-udevd[607]: Process '/nix/store/9f8iga5pr7b59kxrldzmgap21ngh9f49-systemd-234/lib/systemd/systemd-vconsole-setup' failed with exit code 1.
machine# [    6.503284] .fwupd-wrapped[795]: disabling plugin because: failed to startup dell: Firmware updating not supported
machine# [    6.530686] .fwupd-wrapped[795]: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
machine# [    6.534162] .fwupd-wrapped[795]: disabling plugin because: failed to coldplug raspberrypi: Raspberry PI firmware updating not supported, no /boot/start.elf
machine# [    6.537614] .fwupd-wrapped[795]: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
machine# [    6.664355] .fwupd-wrapped[795]: disabling plugin because: failed to coldplug uefi: UEFI firmware updating not supported
machine# [    6.666217] .fwupd-wrapped[795]: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported by OEM
machine# [    6.669088] .fwupd-wrapped[795]: using plugins: steelseries, ebitdo, altos, unifying, colorhug, udev, upower, csr, nitrokey, thunderbolt, dfu
machine# [    6.705332] dbus[577]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service'
machine# [    6.714943] systemd[1]: Starting Authorization Manager...
machine# [    6.778258] polkitd[807]: Started polkitd version 0.113
machine# [    6.828257] polkitd[807]: Loading rules from directory /etc/polkit-1/rules.d
machine# [    6.829888] polkitd[807]: Loading rules from directory /var/run/current-system/sw/share/polkit-1/rules.d
machine# [    6.839843] polkitd[807]: Finished loading, compiling and executing 3 rules
machine# [    6.845860] dbus[577]: [system] Successfully activated service 'org.freedesktop.PolicyKit1'
machine# [    6.849544] systemd[1]: Started Authorization Manager.
machine# [    6.851607] polkitd[807]: Acquired the name org.freedesktop.PolicyKit1 on the system bus
machine# [    6.853422] .fwupd-wrapped[795]: Daemon ready for requests
machine# [    7.121950] dbus[577]: [system] Successfully activated service 'org.freedesktop.fwupd'
machine# [    7.128200] systemd[1]: Started Firmware update daemon.
machine# [    8.752761] .fwupd-wrapped[795]: fu_common_get_contents_fd: assertion 'count > 0' failed
machine# [    8.754175] .fwupd-wrapped[795]: g_dbus_method_invocation_return_gerror: assertion 'error != NULL' failed
machine: exit status 2
machine: output: Running test: fwupd/fwupdmgr.test
Getting the list of remotes...
Remote ID:               lvfs-testing
Title:                   Linux Vendor Firmware Service (testing)
Type:                    download
Keyring:                 gpg
Enabled:                 False
Priority:                2
Filename:                /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz
Filename Signature:      /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz.asc
Metadata URI:            https://s3.amazonaws.com/lvfsbucket/downloads/firmware-testing.xml.gz
Metadata URI Signature:  https://s3.amazonaws.com/lvfsbucket/downloads/firmware-testing.xml.gz.asc
Report URI:              https://fwupd.org/lvfs/firmware/report

Remote ID:               lvfs
Title:                   Linux Vendor Firmware Service
Type:                    download
Keyring:                 gpg
Enabled:                 True
Priority:                1
Filename:                /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz
Filename Signature:      /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz.asc
Metadata URI:            https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Metadata URI Signature:  https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc
Report URI:              https://fwupd.org/lvfs/firmware/report

Remote ID:               fwupd
Title:                   Core
Type:                    local
Keyring:                 none
Enabled:                 True
Filename:                /nix/store/c3vd3knlc8vx0wjbg0jzyhmbqb2v3qdi-fwupd-1.0.4/share/fwupd/remotes.d/fwupd/metadata.xml

[... 94 lines cut ...]
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Executing: fwupd/fwupdmgr.test
Test timed out after 300 seconds
Executing: fwupd/fwupdmgr.test
FAIL: fwupd/fwupdmgr.test (Child process killed by signal 9)
SUMMARY: total=1; passed=0; skipped=0; failed=1; user=0.1s; system=0.1s; maxrss=9272
FAIL: fwupd/fwupdmgr.test (Child process killed by signal 9)
```

- fwupd: 1.0.4
- Operating system and version: NixOS unstable
- How did you install fwupd: https://github.com/NixOS/nixpkgs/pull/32751

superm1, 2018-02-14:
I doubt it's just installed tests failing, I suspect the client doesn't work in this packaged arrangement.

Have you tried to update firmware at all?

jtojnar, 2018-02-14:
> Considering the other issue you filed - this begs the question is the daemon also unable to read /etc/machine-id?

This is a full-pledged container, which has `/etc/machine-id` and should have other important files.

> Have you tried to update firmware at all?

I am only packaging this, do not own any compatible hardware. Should not the installed tests run irregardless of hardware (at least some of them)?

hughsie, 2018-02-14:
>machine# [    8.752761] .fwupd-wrapped[795]: fu_common_get_contents_fd: assertion 'count > 0' failed
>machine# [    8.754175] .fwupd-wrapped[795]: g_dbus_method_invocation_return_gerror: assertion 'error != NULL' failed

The only way this can happen is if `fu_config_get_archive_size_max()` returns zero. Did you perhaps set `ArchiveSizeMax=0` in daemon.conf?

jtojnar, 2018-02-14:
This is the content of my `/etc/fwupd/daemon.conf`:

```
[fwupd]
BlacklistDevices=
BlacklistPlugins=
```

hughsie, 2018-02-14:
> I am trying to run the installed tests in a container

I see no `test` plugin in:

    machine# [    6.669088] .fwupd-wrapped[795]: using plugins: steelseries, ebitdo, altos, unifying, colorhug, udev, upower, csr, nitrokey, thunderbolt, dfu

I'm struggling to see how you don't have the test plugin enabled if `BlacklistPlugins=` is empty. Can you get the `--verbose` logs from fwupd in that container please?

jtojnar, 2018-02-14:
Hmm, I was not compiling with `-Dplugin_dummy=true`.

closed 2018-02-14 by hughsie