fwupd/fwupd #409: fwupd_client_remotes_func: assertion failed (error == NULL)

jtojnar, 2018-02-18:
When I try to build fwupd outside of sandbox and fwupd service is not running, `fu-self-test` fails. This is not a problem with sandboxing enabled or the service running.

```
 1/11 fwupd-exported-api                      OK       0.01 s
 2/11 fwupd-self-test                         FAIL     0.02 s
 3/11 fu-self-test                            OK       4.98 s
 4/11 dfu-self-test                           OK       0.03 s
 5/11 raspberrypi-self-test                   OK       0.02 s
 6/11 nitrokey-self-test                      OK       0.02 s
 7/11 udev-self-test                          OK       0.01 s
 8/11 unifying-self-test                      OK       0.01 s
 9/11 thunderbolt-self-test                   OK       5.60 s
10/11 dell-self-test                          OK       0.02 s
11/11 synapticsmst-self-test                  OK       0.02 s

OK:        10
FAIL:       1
SKIP:       0
TIMEOUT:    0

The output from the failed tests:

 2/11 fwupd-self-test                         FAIL     0.02 s

--- command ---
/tmp/nix-build-fwupd-1.0.5.drv-0/fwupd-1.0.5/build/libfwupd/fwupd-self-test
--- stdout ---
/fwupd/enums: OK
/fwupd/common{machine-hash}: OK
/fwupd/release: OK
í/fwupd/device: 
ColorHug2
  DeviceId:             USB:foo
  Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
  Guid:                 00000000-0000-0000-0000-000000000000
  Flags:                updatable|require-ac
  Checksum:             SHA1(beefdead)
  Icon:                 input-gaming,input-mouse
  Created:              1970-01-01
  Modified:             1970-01-02
  
  [Release]
  AppstreamId:          org.dave.ColorHug.firmware
  Description:          <p>Hi there!</p>
  Version:              1.2.3
  Filename:             firmware.bin
  Checksum:             SHA1(deadbeef)
  Size:                 1.0 kB
  Uri:                  http://foo.com
  TrustFlags:           payload
OK
/fwupd/remote{download}: OK
/fwupd/remote{base-uri}: OK
/fwupd/remote{no-path}: OK
/fwupd/remote{local}: OK
/fwupd/client{remotes}: 
--- stderr ---
**
Fwupd:ERROR:../libfwupd/fwupd-self-test.c:348:fwupd_client_remotes_func: assertion failed (error == NULL): The name org.freedesktop.fwupd was not provided by any .service files (FwupdError, 10)
-------

Full log written to /tmp/nix-build-fwupd-1.0.5.drv-0/fwupd-1.0.5/build/meson-logs/testlog.txt
```

- fwupd version: 1.0.5
- Operating system and version: NixOS unstable

jtojnar, 2018-02-18:
This is the test output with sandboxing enabled:

```
 2/11 fwupd-self-test                         OK       0.02 s

--- command ---
/build/fwupd-1.0.5/build/libfwupd/fwupd-self-test
--- stdout ---
(/build/fwupd-1.0.5/build/libfwupd/fwupd-self-test:16664): Fwupd-DEBUG: D-Bus system bus unavailable, skipping tests.
/fwupd/enums: OK
/fwupd/common{machine-hash}: SKIP
/fwupd/release: OK
/fwupd/device:
ColorHug2
  DeviceId:             USB:foo
  Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
  Guid:                 00000000-0000-0000-0000-000000000000
  Flags:                updatable|require-ac
  Checksum:             SHA1(beefdead)
  Icon:                 input-gaming,input-mouse
  Created:              1970-01-01
  Modified:             1970-01-02

  [Release]
  AppstreamId:          org.dave.ColorHug.firmware
  Description:          <p>Hi there!</p>
  Version:              1.2.3
  Filename:             firmware.bin
  Checksum:             SHA1(deadbeef)
  Size:                 1.0 kB
  Uri:                  http://foo.com
  TrustFlags:           payload
OK
/fwupd/remote{download}: OK
/fwupd/remote{base-uri}: OK
/fwupd/remote{no-path}: OK
/fwupd/remote{local}: OK
```

hughsie, 2018-02-19:
I'm wondering why `fwupd_client_connect()` didn't return if fwupd isn't available.

jtojnar, 2018-02-19:
Fwupd is not even installed. The only thing that crashes are tests; though for some reason only the `fwupd-self-test` fails.

```
Feb 19 12:52:48 kaiser systemd[1]: Started Process Core Dump (PID 8795/UID 0).
-- Subject: Unit systemd-coredump@35-8795-0.service has finished start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit systemd-coredump@35-8795-0.service has finished starting up.
-- 
-- The start-up result is RESULT.
Feb 19 12:52:48 kaiser systemd-coredump[8796]: Resource limits disable core dumping for process 8783 (fwupd-self-test).
Feb 19 12:52:48 kaiser systemd-coredump[8796]: Process 8783 (fwupd-self-test) of user 30001 dumped core.
-- Subject: Process 8783 (fwupd-self-test) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 8783 (fwupd-self-test) crashed and dumped core.
-- 
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
Feb 19 12:52:54 kaiser systemd[1]: Started Process Core Dump (PID 8981/UID 0).
-- Subject: Unit systemd-coredump@36-8981-0.service has finished start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit systemd-coredump@36-8981-0.service has finished starting up.
-- 
-- The start-up result is RESULT.
Feb 19 12:52:54 kaiser systemd-coredump[8982]: Resource limits disable core dumping for process 8978 (fwupd-self-test).
Feb 19 12:52:54 kaiser systemd-coredump[8982]: Process 8978 (fwupd-self-test) of user 30010 dumped core.
-- Subject: Process 8978 (fwupd-self-test) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 8978 (fwupd-self-test) crashed and dumped core.
-- 
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
Feb 19 12:52:54 kaiser kernel: traps: fu-self-test[8984] trap int3 ip:7ffff6c96a61 sp:7ffffffe1a30 error:0 in libglib-2.0.so.0.5400.3[7ffff6c47000+112000]
Feb 19 12:52:54 kaiser systemd[1]: Started Process Core Dump (PID 8989/UID 0).
-- Subject: Unit systemd-coredump@37-8989-0.service has finished start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit systemd-coredump@37-8989-0.service has finished starting up.
-- 
-- The start-up result is RESULT.
Feb 19 12:52:54 kaiser systemd-coredump[8990]: Resource limits disable core dumping for process 8984 (fu-self-test).
Feb 19 12:52:54 kaiser systemd-coredump[8990]: Process 8984 (fu-self-test) of user 30010 dumped core.
-- Subject: Process 8984 (fu-self-test) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 8984 (fu-self-test) crashed and dumped core.
-- 
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
Feb 19 12:52:54 kaiser systemd[1]: Started Process Core Dump (PID 9003/UID 0).
-- Subject: Unit systemd-coredump@38-9003-0.service has finished start-up
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit systemd-coredump@38-9003-0.service has finished starting up.
-- 
-- The start-up result is RESULT.
Feb 19 12:52:54 kaiser systemd-coredump[9006]: Resource limits disable core dumping for process 8998 (raspberrypi-sel).
Feb 19 12:52:54 kaiser systemd-coredump[9006]: Process 8998 (raspberrypi-sel) of user 30010 dumped core.
-- Subject: Process 8998 (raspberrypi-sel) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 8998 (raspberrypi-sel) crashed and dumped core.
-- 
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
```

hughsie, 2018-02-19:
This should have failed https://github.com/hughsie/fwupd/blob/master/libfwupd/fwupd-self-test.c#L338 before the remotes test. Is this reproducable on anything other than NixOS?

superm1, 2018-02-19:
I reproduced this on Ubuntu; steps I took:
1. apt remove all the fwupd related packages
2. `mkdir build; pushd build; meson ..; ninja; ninja test`

I got the same extended error about the service not being found.

superm1, 2018-02-19:
Where you linked to, that's the remotes test and seems to match where it's failing to me.  It should be failing too - the systemd unit isn't installed at that time.

Maybe the test needs to be amended [here  ](https://github.com/hughsie/fwupd/blob/master/libfwupd/fwupd-self-test.c#L416) to check `ListActivatableNames `to make sure that `fwupd` is there.

hughsie, 2018-02-19:
maybe just something like this?

    +       if (fwupd_client_get_daemon_version (client) == NULL) {
    +               g_test_skip ("no enabled fwupd daemon");
    +               return;
    +       }

That gets things working for me.

superm1, 2018-02-19:
That doesn't seem to to work for me on the box I reproduced, still shows the error.
Doesn't it need to be moved to before the asserts and also added to the `client_devices_func`?

hughsie, 2018-02-19:
> That doesn't seem to to work for me on the box I reproduced, still shows the error.

But how? If there's no fwupd, so no version, surely?

> Doesn't it need to be moved to before the asserts and also added to the client_devices_func?

It certainly needs the latter, I'm confused why that didn't fail for me just there too.

superm1, 2018-02-19:
Still fails to me with the last commit in that PR (0c4bd18).   It fails on either assert from the connection in both functions (the `fwupd_client_connect `will return false and will also set an error)

superm1, 2018-02-19:
As for why it didn't fail for you previously, did you call `systemd --daemon-reload `after removing packages or reboot?

hughsie, 2018-02-21:
> call systemd --daemon-reloadafter removing packages 

This was exactly the problem, thanks for the pointer.

hughsie, 2018-02-21:
@jtojnar can you try that please^^

superm1, 2018-02-21:
Still failing for me, line 348

jtojnar, 2018-02-21:
@hughsie Works for me with the patch:

```
 2/11 fwupd-self-test                         OK       0.04 s

--- command ---
/tmp/nix-build-fwupd-1.0.5.drv-0/fwupd-1.0.5/build/libfwupd/fwupd-self-test
--- stdout ---
/fwupd/enums: OK
/fwupd/common{machine-hash}: OK
/fwupd/release: OK
/fwupd/device:
ColorHug2
  DeviceId:             USB:foo
  Guid:                 2082b5e0-7a64-478a-b1b2-e3404fab6dad
  Guid:                 00000000-0000-0000-0000-000000000000
  Flags:                updatable|require-ac
  Checksum:             SHA1(beefdead)
  Icon:                 input-gaming,input-mouse
  Created:              1970-01-01
  Modified:             1970-01-02

  [Release]
  AppstreamId:          org.dave.ColorHug.firmware
  Description:          <p>Hi there!</p>
  Version:              1.2.3
  Filename:             firmware.bin
  Checksum:             SHA1(deadbeef)
  Size:                 1.0 kB
  Uri:                  http://foo.com
  TrustFlags:           payload
OK
/fwupd/remote{download}: OK
/fwupd/remote{base-uri}: OK
/fwupd/remote{no-path}: OK
/fwupd/remote{local}: OK
/fwupd/client{remotes}: SKIP
/fwupd/client{devices}: SKIP
```

superm1, 2018-02-21:
I'm really confused why that could possibly work for @hughsie and @jtojnar but fail for me.
The service shouldn't be able to start in line 347, and should return `false` set `error` setting off asserts in line 348 and 349 (which is what i'm still seeing).
https://gist.github.com/superm1/912eaa117c67496e963dbae42f0c4063

hughsie, 2018-02-21:
@superm1 I cant explain why you're getting a different error domain; does the fixup patch on that branch fix things?

superm1, 2018-02-21:
No that doesn't seem to help either.

superm1, 2018-02-27:
Since it's working for @jtojnar who reported it and I don't really need this I'd say merge it.  If we need to iterate later more we can.

closed 2018-02-27 by superm1