fwupd/fwupd #413: Cannot build Ubuntu packages using contrib

szszszsz, 2018-02-21:
I have tried to build Ubuntu packages with [contrib](https://github.com/hughsie/fwupd/tree/master/contrib), but if failed with a message `ModuleNotFoundError: No module named 'lsb_release'`.
It looks like this has been disabled in Travis CI.

Host: Ubuntu 18.04 (updated)
Docker: version 17.12.0-ce, build c97c6d6

Used commands:
```
sudo OS=ubuntu ./generate_docker.py
sudo docker run -t -v `pwd`:/build fwupd-ubuntu
```

Full output of the last command:
```
126 sz@sz-18:~/work/fwupd$ sudo docker run -t -v `pwd`:/build fwupd-ubuntu
+ '[' '' = debian-s390x ']'
+ export 'DEBFULLNAME=CI Builder'
+ DEBFULLNAME='CI Builder'
+ export DEBEMAIL=ci@travis-ci.org
+ DEBEMAIL=ci@travis-ci.org
++ sed 's/-/+r/;s/-/+/'
++ git describe
+ VERSION=1.0.5+r10+g02c4690
+ '[' -z 1.0.5+r10+g02c4690 ']'
+ rm -rf build/
+ mkdir -p build
+ shopt -s extglob
+ cp -lR AUTHORS CODE_OF_CONDUCT.md CONTRIBUTING.md COPYING MAINTAINERS NEWS README.md RELEASE contrib data docs libfwupd meson.build meson_options.txt meson_post_install.sh plugins po policy src build/
+ pushd build
/build/build /build
+ mv contrib/debian .
+ sed s/quilt/native/ debian/source/format -i
+ ./contrib/ci/generate_debian_control.py debian/control.in debian/control
Traceback (most recent call last):
  File "./contrib/ci/generate_debian_control.py", line 100, in <module>
    deps = parse_control_dependencies("build")
  File "./contrib/ci/generate_debian_control.py", line 41, in parse_control_dependencies
    import lsb_release
ModuleNotFoundError: No module named 'lsb_release'

```

superm1, 2018-02-21:
You're missing the OS variable in you docker run command.

`sudo docker run -e OS=ubuntu-clang -t -v `pwd`:/build fwupd-ubuntu`

szszszsz, 2018-02-21:
It works. Thank you!
Just FYI - this is different to [contrib's guide](https://github.com/hughsie/fwupd/tree/master/contrib#deb-packages) command.

closed 2018-02-21 by szszszsz

superm1, 2018-02-21:
Thanks!  I forgot we changed something in Travis related to this.  I'll update the guide when we rename ubuntu-clang to ubuntu-x86_64.

commit 0029221f (2018-02-21): contrib: Remove declaring OS variable in docker run command (Fixes: #413)

Since the containers are ephemeral just declare directly in the Dockerfile
This also simplifies the run command for users building fwupd
commit 8fd7d05b (2018-02-21): contrib: Don't declare OS variable in docker run command (Fixes: #413)

Since the containers are ephemeral just declare directly in the Dockerfile
This also simplifies the run command for users building fwupd
commit 66a51e81 (2018-02-21): contrib: Don't declare OS variable in docker run command (Fixes: #413)

Since the containers are ephemeral just declare directly in the Dockerfile
This also simplifies the run command for users building fwupd
commit a0f9f619 (2018-02-21): contrib: Don't declare OS variable in docker run command (Fixes: #413)

Since the containers are ephemeral just declare directly in the Dockerfile
This also simplifies the run command for users building fwupd