fwupd/fwupd #416: thunderbolt-self-test fails mysteriously

jtojnar, 2018-02-21:
In certain CI environment the test fails similarly to https://github.com/hughsie/fwupd/issues/280:

```
 9/11 thunderbolt-self-test                   FAIL     0.02 s
--- command ---
LD_PRELOAD='/nix/store/085rz1chj77nsw9zi08p13vx8lq87prv-umockdev-0.11/lib/libumockdev-preload.so.0' /build/fwupd-1.0.5/build/plugins/thunderbolt/thunderbolt-self-test
--- stdout ---
/thunderbolt/basic:
--- stderr ---
(/build/fwupd-1.0.5/build/plugins/thunderbolt/thunderbolt-self-test:16712): FuPluginThunderbolt-WARNING **: Need to run within umockdev wrapper (umockdev-wrapper thunderbolt-self-test)!
```

Locally and on CI builder, it works. The library path is correct, I even patched it to use absolute path to be sure; an error would be produced with a non-existing path:

```
--- command ---
LD_PRELOAD='/nix/store/jxw5md5vdk68iiqzswgnr48d1w6q6jd3-umockdev/lib/blibumockdev-preload.so.0' /build/fwupd-1.0.5/build/plugins/thunderbolt/thunderbolt-self-test
--- stdout ---
/thunderbolt/basic:
--- stderr ---
ERROR: ld.so: object '/nix/store/jxw5md5vdk68iiqzswgnr48d1w6q6jd3-umockdev/lib/blibumockdev-preload.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.

(/build/fwupd-1.0.5/build/plugins/thunderbolt/thunderbolt-self-test:16709): FuPluginThunderbolt-WARNING **: Need to run within umockdev wrapper (umockdev-wrapper thunderbolt-self-test)!
```

I am not sure how exactly the environments differ, only that it seems to fail deterministically.

- Downstream issue: https://github.com/NixOS/ofborg/issues/84
- fwupd: 1.0.5
- umockdev: 0.11
- Operating system and version: NixOS

superm1, 2018-02-21:
Can you try to emulate this environment locally to make it fail locally too?  It will be a bit hard to figure out otherwise
.

Or do you know anything about the artificial restrictions put in place by this environment?  Is something else LD_PRELOAD'ed that might be making umockdev fail?

superm1, 2018-03-01:
Is this still a problem for you?

jtojnar, 2018-03-01:
I will try to run another round of tests during the weekend.

superm1, 2018-04-23:
Any updates on this?

jtojnar, 2018-04-23:
Nope, I were not able to reproduce it due to other issue. Will re-open this when I encounter the problem again.

closed 2018-04-23 by jtojnar