fwupd/fwupd #417: Reducing RAM usage by shutting down the daemon

eloydegen, 2018-02-23:
```
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.7.2
	gusb:	0.2.11
```
It would be possible to reduce the RAM to shutdown the daemon when it is not being used. Systemd timers could be used to start the daemon once in a while, while ```udev``` rules could be used to spawn the daemon to start it up at hotplug events with vendor and device IDs of supported devices. Applications that depend on ```fwupd```  such as ```gnome-software``` are already autostarting the daemon, so the it should be shutdown afterwards, by itself or otherwise. 

I suggest adding a command line option that it would shutdown itself after all updating jobs are done.

hughsie, 2018-02-23:
This is something that's perhaps useful; of course by default I'd like to see fwupd running all the time to get device hotplug events and to be able to get results instantly. How much RAM is fwupd actually using on your system? On mine it's ~1Mb...

eloydegen, 2018-02-23:
I ran it through ```heaptrack``` and it is around 1.7MiB, but considering this daemon is also made for devices such as the Raspberry Pi it would still be useful to save those small amounts of RAM if that is feasible.

hughsie, 2018-02-23:
The only low-hanging fruit I can see is:

    ->22.57% (227,552B) 0x6C05EA2: sqlite3MemMalloc (sqlite3.c:21322)

where we could close the database after some timeout, rather than keeping it open. But that's only 200k...

superm1, 2018-02-23:
It would be some effort/duplication to bring VID/PID over into udev rules for the ones that should spawn the daemon on hotplug events.

Since you mention rpi, another thought is to have the daemon look at how much memory is on the system and decide to shut itself down if it does find a memory constrained environment.  If a proposed command line option for shutting down after some timeout was adopted the constrained environment can flip the default for it.

superm1, 2018-04-23:
Since rpi bootloader updating support was dropped from fwupd, it's not really a relevant target any more.  Closing this issue.

hughsie, 2018-11-14:
We were talking about this just today; this seems like something we might be able to do with some udev and systemd service integration. @aruiz

superm1, 2018-11-14:
Are you thinking optionally shutting it down (opt in)?

aruiz, 2018-11-14:
@gicmo and I just had a chat about this, he's going to explore the possibility of shutting everything down (fwupd and boltd)

hughsie, 2018-11-14:
> Are you thinking optionally shutting it down

I'm not sure; we could either have an explicit "fwupdmgr shutdown" command that requires root -- but more obvious would just be if fwupd shut down after some idle time, say 15 minutes. It would mean re-launching (and re-probing hardware) whenever any hardware was inserted -- and so I'm slightly skeptical it would be any more power efficient, but it would certainly help on low memory IoT devices that come with 1Gb of RAM.

superm1, 2018-11-14:
I'm not crazy about having to redo coldplug every time something is plugged in outside of a narrow window.

Example; thunderbolt.
If you plug in something in another USB port that would spawn fwupd again and fwupd would do the whole force power dance to turn on thunderbolt, turn it off etc. Powering up AR/TR controllers is going to use a bunch of power for what should be static information. It's going to cause boltd to spin up too if not already running.

Coldplugging synaptics again might cause a display flicker due to how GFX driver works.

So my preference would be that this be:
1. Offer via command line option that you normally need root for.
2. Use heuristics to decide when to do this automatically (less than 1GB ram, no thunderbolt, etc)

gicmo, 2018-11-14:
For the thunderbolt stuff: Since the bootacl support, bolt already keeps the domain representation around; it would not be *that* hard to also keep the device representing the host device around and add the needed information (firmware version, etcpp) that are currently missing. fwupd could use that if the controller is powered down. Would add extra complexity to boltd and the fwupd plugin.

hughsie, 2018-11-14:
I think an easier way to do this is just let the plugins "inhibit" the auto-shutdown. If the computer has a Thunderbolt controller it's realistically going to have more than 1GB RAM, and like @superm1 there is a lot going on just to get the firmware version from a powered down controller. If fwupd uses 1Mb of RAM on my 32Gb laptop then that's just insignificant and almost a rounding error.

superm1, 2018-11-14:
Sounds like a good approach to me.

On the udev rules to wake up it would be nice to only wakeup from stuff we know fwupd supports. Maybe during

closed 2018-11-21 by hughsie