fwupd/fwupd #420: issue updating fw ... nothings happens when i do fwupdmgr get-upates

laithrafid, 2018-02-26:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell fwupdmgr --version 
client version:	1.0.5
daemon version:	1.0.5
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.
ii  fwupd                1.0.5-1         amd64           Firmware update daemon

```shell
$ fwupdmgr get-devices
wupdmgr get-devices 
HP Spectre x360 Convertible 13 System Firmware
  DeviceId:             4e0edd024cc3d24b3749eadf2dfa33a58d046600
  Guid:                 39cf18a6-27a0-4cce-8323-f79679042ab1
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.15.16896
  VersionLowest:        0.10.0
  Icon:                 computer
  Created:              2018-02-26

HD Graphics 5500
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 9f10b466-1d3c-5e9a-bd0e-be4ee85de216
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-02-26

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,9999
Boot0001* kali	HD(1,GPT,abe9bda4-e8b9-4673-a453-3777792bc7fd,0x800,0x100000)/File(\EFI\kali\grubx64.efi)
Boot0002* USB Drive (UEFI) - Hard Drive	BBS(HD,,0x0)..GO..NO........o.A.X.N.S.3.8.1.E.-.1.2.8.G.M.-.B....................A...........................>..Gd-.;.A..MQ..L.F.2.4.1.0.2.3.0.0.2.8.2. . . . . . . . ........BO
Boot0005* USB Drive (UEFI) - Hard Drive	BBS(HD,,0x0)..GO..NO........o.A.X.N.S.3.8.1.E.-.1.2.8.G.M.-.B....................A...........................>..Gd-.;.A..MQ..L.F.2.4.1.0.2.3.0.0.2.8.2. . . . . . . . ........BO
Boot9999* USB Drive (UEFI)	PciRoot(0x0)/Pci(0x1d,0x0)/USB(16,0)..BO

```shell
$ efivar -l | grep fw
no output 

```shell
$ tree /boot
boot/
├── config-4.14.0-kali3-amd64
├── efi
│   └── EFI
│       ├── debian
│       │   ├── fw
│       │   └── fwupx64.efi
│       └── kali
│           └── grubx64.efi
├── grub
│   ├── fonts
│   │   └── unicode.pf2
[... 312 lines cut ...]
└── vmlinuz-4.14.0-kali3-amd64

9 directories, 315 files

**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version:  cat /etc/os-release 
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2018.1"
VERSION_ID="2018.1"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.kali.org/"
SUPPORT_URL="http://forums.kali.org/"
BUG_REPORT_URL="http://bugs.kali.org/"

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):apt-get
- Have you tried rebooting? yes 
- Are you using an NVMe disk? nope 
- Is secure boot enabled (only for the UEFI plugin)? nope

laithrafid, 2018-02-26:
below is the strace of the command 
`strace fwupdmgr get-updates
execve("/usr/bin/fwupdmgr", ["fwupdmgr", "get-updates"], 0x7ffc612a55f8 /* 44 vars */) = 0
brk(NULL)                               = 0x559a7b6bb000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=181366, ...}) = 0
mmap(NULL, 181366, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f9077590000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libfwupd.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0`w\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=108464, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f907758e000
mmap(NULL, 2203816, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f907717e000
mprotect(0x7f9077197000, 2097152, PROT_NONE) = 0
mmap(0x7f9077397000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19000) = 0x7f9077397000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libappstream-glib.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0pn\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=357104, ...}) = 0
mmap(NULL, 2452824, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9076f27000
mprotect(0x7f9076f7b000, 2093056, PROT_NONE) = 0
mmap(0x7f907717a000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x53000) = 0x7f907717a000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0p\257\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=342072, ...}) = 0
mmap(NULL, 2440040, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9076cd3000
mprotect(0x7f9076d25000, 2097152, PROT_NONE) = 0
mmap(0x7f9076f25000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x52000) = 0x7f9076f25000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libglib-2.0.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\256\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1125680, ...}) = 0
mmap(NULL, 3223656, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f90769bf000
mprotect(0x7f9076ad0000, 2097152, PROT_NONE) = 0
mmap(0x7f9076cd0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x111000) = 0x7f9076cd0000
mmap(0x7f9076cd2000, 104, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f9076cd2000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260v\3\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=1669944, ...}) = 0
mmap(NULL, 3772600, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9076625000
mprotect(0x7f90767b6000, 2093056, PROT_NONE) = 0
mmap(0x7f90769b5000, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x190000) = 0x7f90769b5000
mmap(0x7f90769bd000, 4280, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f90769bd000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libgusb.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@I\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=56496, ...}) = 0
mmap(NULL, 2151832, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9076417000
mprotect(0x7f9076424000, 2093056, PROT_NONE) = 0
mmap(0x7f9076623000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xc000) = 0x7f9076623000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/libsoup-2.4.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300\312\4\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=983248, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f907758c000
mmap(NULL, 3080464, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9076126000
mprotect(0x7f90761f2000, 2093056, PROT_NONE) = 0
mmap(0x7f90763f1000, 151552, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xcb000) = 0x7f90763f1000
mmap(0x7f9076416000, 272, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f9076416000
[... 748 lines cut ...]
futex(0x559a7b6fda50, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x559a7b6fd9e0, FUTEX_WAKE_PRIVATE, 1) = 1
[... cut ...]
futex(0x559a7b6f3078, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x559a7b6f3078, FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)
futex(0x7f9076f261f8, FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)
futex(0x7f9076f261f8, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x559a7b6f3078, FUTEX_WAKE_PRIVATE, 1) = 1
write(6, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x7f9076f26288, FUTEX_WAKE_PRIVATE, 1) = 0
exit_group(0)                           = ?
+++ exited with 0 +++
`

laithrafid, 2018-02-26:
[straceofcommand3.txt](https://github.com/hughsie/fwupd/files/1756908/straceofcommand3.txt)
[straceofcommand2.txt](https://github.com/hughsie/fwupd/files/1756909/straceofcommand2.txt)
[straceofcommand.txt](https://github.com/hughsie/fwupd/files/1756910/straceofcommand.txt)

laithrafid, 2018-02-26:
[strace-fwupdmgr-get-update.txt](https://github.com/hughsie/fwupd/files/1756930/strace-fwupdmgr-get-update.txt)

[strace-fwupdmgr-get-devices.txt](https://github.com/hughsie/fwupd/files/1756931/strace-fwupdmgr-get-devices.txt)

laithrafid, 2018-02-26:
[strace-fwupdmgr-get-updates.txt](https://github.com/hughsie/fwupd/files/1756934/strace-fwupdmgr-get-updates.txt)

closed 2018-02-26 by laithrafid

hughsie, 2018-02-26:
Maybe we need to make it more obvious that no devices are supported or have updates?

laithrafid, 2018-02-26:
@hughsie  what do you mean ?

laithrafid, 2018-02-26:
@have i want to upload cap files from my labtop's vendor to lvfs so that i can update regularly ... my laptop vendors doesn't support linux fw upgrades .

laithrafid, 2018-02-26:
@hughsie  nevermid my questions .... i just read it on LVFS