fwupd/fwupd #421: Question regarding fwupdate --esp-path

ArchangeGabriel, 2018-02-27:
This was introduced in `fwupdate` so that Arch users can finally have this working without compiling from source, but what I’m not sure of is when should this option be passed as arg? For `fwupdmgr update`? Or are users currently required to run `fwupdate` directly?

superm1, 2018-02-27:
Let me see if I understand correctly.  Normally with `fwupdate` 10+ you would set an argument on command line that would override the path.  You're wondering how `fwup_set_esp_mountpoint `gets called from fwupd?

I suppose it's a bit of a shift to allow setting this path at runtime that fwupd didn't comprehend yet.
My gut would say we should add this as a conf option to a new `uefi.conf` in `/etc/fwupd/`, although that might not really be necessary anymore after a fix for https://github.com/rhboot/fwupdate/issues/97 is developed.

@hughsie your thoughts on the matter?

superm1, 2018-02-27:
Yeah you're probably right it will take some time so it's worthwhile to let fwupd override it.  Unless @hughsie thinks it's the wrong approach i'll work on a PR for it soon.

https://github.com/rhboot/fwupdate/blob/master/linux/fwupdate.c#L234

superm1, 2018-02-27:
Submitted a PR above that should offer this via conffile.  if that's accepted I'd say you should backport that to your fwupd package.

closed 2018-02-27 by superm1

ArchangeGabriel, 2018-02-28:
Thanks, I’m backporting that.

ArchangeGabriel, 2018-03-06:
Done, as well as wiki update: https://wiki.archlinux.org/index.php/Fwupd#Running_fwupd

daenney, 2018-04-07:
@ArchangeGabriel I don't really understand the instructions on the Arch wiki. It suggests that if the ESP is not mounted at `/boot/EFI`, you need to set `OverrideESPMountPoint` to `esp`? Or the actual path to the ESP, i.e `/boot`?

Given that Arch seems to use systemd-bootd by default, and the [wiki recommends mounting the ESP on `/boot`](https://wiki.archlinux.org/index.php/systemd-boot#Installing_the_EFI_boot_loader), should the default on not be changed? The out-of-the-box experience is mildly broken now.

ArchangeGabriel, 2018-04-07:
> Given that Arch seems to use systemd-bootd by default, and the wiki recommends mounting the ESP on /boot, should the default on Arch not be changed? The out-of-the-box experience is mildly broken now.

There is no such thing as a default bootloader in Arch. No-one is expected to use `systemd-bootd`, and GRUB is a better choice if you ask me because of `/boot` encryption.

daenney, 2018-04-07:
Ah, you're indeed correct. In that case I suppose it doesn't matter much. Thanks!

commit b17d184b (2018-02-27): uefi: Allow overriding ESP mount point via conf file (Fixes: #421)
commit 5d82cc7e (2018-02-27): uefi: Allow overriding ESP mount point via conf file (Fixes: #421)
commit fd3b4f2d (2018-02-27): uefi: Allow overriding ESP mount point via conf file (Fixes: #421)
commit be059fbc (2018-02-27): uefi: Allow overriding ESP mount point via conf file (Fixes: #421)