fwupd/fwupd #430: fwupd does not specify correct as-glib dependency

DimStar77, 2018-03-06:
Just udpated fwupd, according to meson.build it still requires as-glib 0.6.13

Building with 0.7.4 though is not enough, as it then fails with:

```
[   69s] /home/abuild/rpmbuild/BUILD/fwupd-1.0.5/build/../src/fu-engine.c:893: undefined reference to `as_store_get_apps_by_provide'
[   69s] collect2: error: ld returned 1 exit status
```

superm1, 2018-03-06:
That's a bit odd because fwupd does explicitly check which version of appstream-glib is on your system to not use that symbol.
https://github.com/hughsie/fwupd/blob/34e48634af6c39b10e8d6dec989cb083da38c194/src/fu-engine.c#L892

DimStar77, 2018-03-06:
Ah - i see.. this is due to us having a git checkout of 'shortly after' 0.7.4 in our build system, as 0.7.4 caused bad regressions.. so we are at 0.7.4+3 commits, which in turn identifies itself to the build system already as 0.7.5.

That's all good then, and just our special case. Sorry for the noise

closed 2018-03-06 by DimStar77

superm1, 2018-03-06:
Thanks for confirming. If you are seeing some regression in 0.7.4 that aren't already fixed in master, please do report them to to http://github.com/hughsie/appstream-glib to get sorted.

DimStar77, 2018-03-06:
I'm a constant nagger on as-glib :P which is why we had to move a bit beyond the tag. And I'll have to do the same with 0.7.6 again (but that's just FYI)

superm1, 2018-03-06:
OK :) I guess in the future be more careful to just backport patches rather than move up all the commits to potentially a new version that's missing things to avoid this kind of stuff happening.  Like if you're moving past 0.7.6 make sure you definitely don't backport https://github.com/hughsie/appstream-glib/commit/d5bd692f5f285e0fa86e8caf146b790f11f45849