fwupd/fwupd #431: fwupd should delete .cap files on ESP when system is rebooted

hughsie, 2018-03-06:
Nothing actually deletes the .cap file when the firmware file has been deployed onto the hardware. If you do multiple firmware updates and only have a 200Mb `/boot/efi` (the Fedora default) then very quickly you are out of space.

We could handle this in two ways:

 * delete it in fwupdate.efi, i.e. after the firmware update has run and the .cap file is no longer required
 * delete $ESP/fwupdate-*.cap in fwupd on next reboot

...the latter requiring a new fwup_get_esp_mountpoint() method to get the correct mountpoint.

@vathpela @superm1 @GavinChengGao  -- any other ideas?

superm1, 2018-03-06:
The problem with using `fwup_get_esp_mountpoint` is that fwupdate now lets you override this with a command line option (or a fwupd configuration variable).

I'd argue the first proposal is the best solution.  Reason being fwupdate.efi is already deleting the EFI variable when it's done with it.  Once the payload is loaded into memory but before UpdateCapsule is called seems like a reasonable time to delete it.

The quicker (and dirtier) solution is to just change the fwupdate cleanup unit to run on the next boot (eg fwupd deletes `/var/lib/fwupdate/done`).  Assuming that works, that is probably quicker to implement then spinning a new fwupdate release.

superm1, 2018-03-06:
The other thing that should be noted is that fwupdate (at one point) made an effort to try to [re-use files](https://github.com/rhboot/fwupdate/blob/master/linux/libfwup.c#L1473
).
Actually looking at the [EFI code ](https://github.com/rhboot/fwupdate/blob/master/efi/fwupdate.c#L882)it's deleting the `Linux-Firmware-Updater` entry not the fwupdate-* variable.  So that re-use stuff should still be active.

hughsie, 2018-03-06:
> fwupd deletes /var/lib/fwupdate/done

I hadn't thought of that. What *exactly* does deleting this achieve? Is there any potential for regression?

superm1, 2018-03-06:
Makes this [cleanup job ](https://github.com/rhboot/fwupdate/blob/master/linux/cleanup.in) run assuming the systemd unit is enabled (it's a oneshot)

hughsie, 2018-03-06:
So it clears the efivars too -- which I think means we'd only want to do it on success as otherwise we'll get no detailed debugging.

superm1, 2018-03-06:
Deleting from that script is actually bound to fail come to think of it if you overrided ESP too though.
So either fwupdate.efi should delete the file and variable or fwupd should do it on success. fwupd can parse the variable to figure out where the ESP was that pointed to the file was.

hughsie, 2018-03-06:
@vathpela is travelling today I think, but when he gets back we can ask him what he prefers.

hughsie, 2018-03-06:
I talked to @vathpela today and he wanted fwupd to do the deletion. I'll work on a patch after bedtime.

closed 2018-03-09 by hughsie