fwupd/fwupd #436: Dell XPS 13 9370: Firmware update doesn’t work

paulmenzel, 2018-03-13:
Updating the firmware of the Dell XPS 13 9370 doesn’t work. The command below succeeds, but after the restart, the update is not applied and the GRUB menu is shown after some seconds like during normal boot.

```
$ sudo fwupdmgr update
Downloading 0.1.2.0 for XPS 13 9370 System Firmware...
Updating 0.1.2.0 on XPS 13 9370 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]:
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.5
daemon version:	1.0.5
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.7.0
  VersionBootloader:    11.7.0
  Icon:                 computer
  Created:              2018-03-12

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.0.2
  VersionLowest:        0.1.0.2
  Icon:                 computer
  Created:              2018-03-12
  UpdateState:          success

UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-03-12

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* ubuntu	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* debian	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\debian\grubx64.efi)
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(3,GPT,fe1219bf-0100-4eb6-a684-673418a4475c,0xb2c800,0x36fd2800)/File(boot\efi\EFI\debian\fwupx64.efi)
Boot0006* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-7ceaf7a8-0611-4480-9e30-64d8de420c7c-0
```

```shell
$ tree /boot/efi
/boot/efi
└── EFI
    └── debian
        ├── fw
        │   ├── fwupdate-54MsnF.cap
        │   ├── fwupdate-CKobTW.cap
        │   ├── fwupdate-hCkoNv.cap
        │   ├── fwupdate-M9ELss.cap
        │   ├── fwupdate-MIVNfe.cap
        │   ├── fwupdate-Mlz4k5.cap
        │   ├── fwupdate-QHPYiu.cap
        │   └── fwupdate-w4i8tn.cap
        └── fwupx64.efi

3 directories, 9 files
```

Please answer the following questions:

- Operating system and version: Debian Sid/unstable
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt install fwupd
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

superm1, 2018-03-13:
Can you please check your journal for `fwupd.service`?  There should be a cached `efibootmgr` output from "right" before the reboot.  Can you share that?  The one you shared looks like it's after the reboot.

superm1, 2018-03-13:
Also did you briefly see some text pop up before grub at all?

superm1, 2018-03-13:
One more thought - 
```
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(3,GPT,fe1219bf-0100-4eb6-a684-673418a4475c,0xb2c800,0x36fd2800)/File(boot\efi\EFI\debian\fwupx64.efi)
```

Your entry has the wrong path (it includes `boot\efi` at the beginning).  Did you manually change something to impact this? Eg did you manually create the entry or create it from a chroot or any other extraneous scenarios?

Or was this created automatically by `fwupdmgr update` (and the associated libraries it uses)?

paulmenzel, 2018-03-13:
On 03/13/2018 01:11 AM, Mario Limonciello wrote:
> Can you please check your journal for `fwupd.service`?  There should
> be a cached `efibootmgr` output from "right" before the reboot.  Can
> you share that?  The one you shared looks like it's after the
> reboot.

```
$ journalctl -u fwupd -b -1
Mar 13 16:17:48 xps139370 systemd[1]: Starting Firmware update daemon...
Mar 13 16:17:49 xps139370 fwupd[1474]: disabling plugin because: failed 
to coldplug raspberrypi: Raspberry PI firmware updating not supported, 
no /boot/start.elf
Mar 13 16:17:49 xps139370 fwupd[1474]: disabling plugin because: failed 
to coldplug synapticsmst: MST firmware updating not supported, missing 
kernel support.
Mar 13 16:17:49 xps139370 fwupd[1474]: disabling plugin because: failed 
to coldplug thunderbolt_power: missing kernel support for 
intel-wmi-thunderbolt
Mar 13 16:17:49 xps139370 fwupd[1474]: using plugins: dfu, udev, dell, 
colorhug, amt, nitrokey, upower, steelseries, unifying, csr, ebitdo, 
thunderbolt, altos, uefi
Mar 13 16:17:49 xps139370 fwupd[1474]: Daemon ready for requests
Mar 13 16:17:49 xps139370 systemd[1]: Started Firmware update daemon.
Mar 13 16:18:07 xps139370 fwupd[1474]: Boot Information:
                                        BootNext: 0002
                                        BootCurrent: 0001
                                        Timeout: 0 seconds
                                        BootOrder: 0001,0000,0002
                                        Boot0000* ubuntu 
HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
                                        Boot0001* debian 
HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\debian\grubx64.efi)
                                        Boot0002* Linux-Firmware-Updater 
\fwupx64.efi 
HD(3,GPT,fe1219bf-0100-4eb6-a684-673418a4475c,0xb2c800,0x36fd2800)/File(boot\efi\EFI\debian\fwupx64.efi)
                                        Boot0006* UEFI: KXG50ZNV512G 
NVMe TOSHIBA 512GB, Partition 1 
HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
Mar 13 16:18:16 xps139370 systemd[1]: Stopping Firmware update daemon...
Mar 13 16:18:16 xps139370 systemd[1]: Stopped Firmware update daemon.
```

paulmenzel, 2018-03-13:
On 03/13/2018 01:29 AM, Mario Limonciello wrote:
> Also did you briefly see some text pop up before grub at all?

No, there is no text shown as far as I can see.

paulmenzel, 2018-03-13:
On 03/13/2018 07:14 AM, Mario Limonciello wrote:
> One more thought -
> ```
> Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(3,GPT,fe1219bf-0100-4eb6-a684-673418a4475c,0xb2c800,0x36fd2800)/File(boot\efi\EFI\debian\fwupx64.efi)
> ```
>
> Your entry has the [...]

No, I didn’t put this file there.

> Or was this created automatically by `fwupdmgr update` (and the associated libraries it uses)?

Yes, it’s created by `fwupdmgr update`.

Looking more into it, the ESP doesn’t seem to be mounted. Maybe a check 
should be added for that?

superm1, 2018-03-14:
Alright this is starting to make a lot more sense now.  I think that this issue and your other issue #438 are both caused by the exact same issue.

Your EFI system partition wasn't mounted when `fwupdate` was installed. 
`d-i` should have configured the `/etc/fstab` to have an entry for the ESP. 

Normally when it's mounted and fwupdate is installed it puts `fwupx64.efi` into the ESP. Since yours wasn't mounted it was just put into an empty directory.

Then when there was an attempt at create a firmware update entry.  Since it looked like the file was in place it made the NVRAM entry but it was invalid since the ESP didn't actually contain the file.

So I think we can do better, we can probably check and confirm the ESP is mounted before scheduling an update, that would help prevent the situation that the update is scheduled while you're in an invalid configuration.

To get to the root of this though - 
Did you manually do something to change that or did your `/etc/fstab` not get configured to mount it automatically? Or is there a problem in d-i?

superm1, 2018-03-14:
Here's a fix to not let this happen in the future (run an update with an unmounted ESP).
https://github.com/hughsie/fwupd/pull/439

I still want to get to your root cause though, so please answer my previous question.

paulmenzel, 2018-03-14:
On 03/14/2018 02:15 AM, Mario Limonciello wrote:

[…]

Thank you for diagnosing the problem.

> To get to the root of this though - Did you manually do something to
> change that or did your `/etc/fstab` not get configured to mount it
> automatically? Or is there a problem in d-i?
I installed the system in with legacy mode enabled in the firmware – as 
my Grml device wasn’t recognized in UEFI mode. Then only later, I 
converted it to UEFI, and forgot to add the ESP entry in `/etc/fstab`.

superm1, 2018-03-14:
Gotcha OK, so this should be a pretty rare problem.  It is still worth catching so I think we should merge my PR for it for the future.

commit cac0ef2c (2018-03-14): uefi: Check that EFI system partition is mounted (Fixes: #436)

If the ESP isn't mounted it's possible that the update may look
succesful on the Linux side, but certain to fail later.
commit 937299d7 (2018-03-14): uefi: Check that EFI system partition is mounted (Fixes: #436)

If the ESP isn't mounted it's possible that the update may look
succesful on the Linux side, but certain to fail later.
closed 2018-03-14 by hughsie