fwupd/fwupd #437: Requesting to upload report after a successful update - expected?

maximbaz, 2018-03-13:
It's a rather small annoyance and I'm not actually sure if this is a bug or a feature, so I just wanted to let you know.

After I updated firmware and rebooted, every time I used `fwupdmgr get-updates` or `fwupdmgr get-devices` I got the following message:

```shell
________________________________________________

Devices that have been updated successfully:

 • Precision 5520 System Firmware (0.1.6.2 → 0.1.7.1)

Upload report now? (Requires internet connection) [Y|n]: 
```

I could not get rid of it until I agreed to upload the report, after I did so, this message disappeared.

I would assume you want to report only failures, but not successful updates? In this case, such a prompt should not have appeared. But if you do want successful reports too, then it's fine.

Feel free to close this ticket when you look at it.

<details>
<summary>Requested info</summary>

```shell
$ fwupdmgr --version
client version: 1.0.6
daemon version: 1.0.6
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34

$ fwupdmgr get-devices
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.50
  VersionBootloader:    11.8.50
  Icon:                 computer
  Created:              2018-03-13

Precision 5520 TPM 2.0
  DeviceId:             de98f10c9b146ba4ced2fc37cad31ee5b8808f6d
  Guid:                 8bc551f7-f7ca-5ea8-a4cb-38ae16ecfa17
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.1.0
  Icon:                 computer
  Created:              2018-03-13

Precision 5520 TPM 1.2
  DeviceId:             e0527307c0aa37127c77c5c1f777cbe348591d1a
  Guid:                 61c84b2d-1924-5867-91e2-63ae6a4373af
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-03-13

Precision 5520 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.7.1
  VersionLowest:        0.1.7.1
  Icon:                 computer
  Created:              2018-03-13

GM107GLM [Quadro M1200 Mobile]
  DeviceId:             3f0f51a4f03bccdad9f8ae1334bad4584b5ba875
  Guid:                 f201385b-abb8-522b-b393-a11c2bdb4ca8
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Icon:                 audio-card
  Created:              2018-03-13

Unknown Device
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 3ab5fe9e-bef7-5421-920a-ef7301884933
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-03-13

Precision 5520  Thunderbolt Controller
  DeviceId:             f747745114def3d21ba9db6841b81b8e8c05d4e2
  Guid:                 c5364422-2ec3-5092-b30e-03bdb1a53e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2018-03-13

$ efibootmgr -v
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002
Boot0000* Windows Boot Manager	HD(1,GPT,4ff8a9b2-1fe3-430e-b8dd-ef7ee5145358,0x800,0x1f4000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* arch	HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\grubx64.efi)
Boot0002* arch	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-B3-71-B2-1A-24)/HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\grubx64.efi)
Boot0003* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,2ac1f58a-1140-c242-8072-29779dc87cf9,0x800,0x32000)/File(\EFI\arch\fwupx64.efi)

$ efivar -l | grep fw
<no output>

$ tree /boot/efi
/boot/efi
└── EFI
    ├── arch
    │   ├── fw
    │   ├── fwupx64.efi
    │   └── grubx64.efi
    └── Dell
        ├── Bios
        │   └── Recovery
        │       ├── BIOS_CUR.RCV
        │       └── BIOS_PRE.rcv
        └── logs
            ├── diags_current.xml
            └── diags_previous.xml

7 directories, 6 files
```

Please answer the following questions:

- Operating system and version: Arch Linux 4.15.8-1-ARCH
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

</details>

ArchangeGabriel, 2018-03-13:
I think they want both, because it’s useful to know how many attempts for that much failures, but they will know better than me of course. ;)

hughsie, 2018-03-13:
It is expected, as we want to see the ratio between failures and success on the dashboard. Maybe we need to do a better job communicating that on the prompt. `Upload report now?` is somewhat terse.

commit edcd6bc7 (2018-04-10): Improve the update report message (Closes: #437)
commit 8c71a15a (2018-04-23): Improve the update report message (Closes: #437)
commit 5a256531 (2018-04-23): Improve the update report message (Closes: #437)
closed 2018-04-23 by superm1