fwupd/fwupd #438: No such file or directory	{error #1} libfwup.c:1143 set_up_boot_next(): could not find paths for shim and fwup: No such file or directory

paulmenzel, 2018-03-13:
Mounting the ESP to `/boot/efi`, updating the firmware shows the error below.

```
$ sudo fwupdmgr update
Downloading 0.1.2.0 for XPS 13 9370 System Firmware...
Updating 0.1.2.0 on XPS 13 9370 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} libfwup.c:1005 get_paths(): could not find shim or fwup on ESP: No such file or directory	{error #1} libfwup.c:1143 set_up_boot_next(): could not find paths for shim and fwup: No such file or directory
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.5
daemon version:	1.0.5
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.7.0
  VersionBootloader:    11.7.0
  Icon:                 computer
  Created:              2018-03-12

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.0.2
  VersionLowest:        0.1.0.2
  Icon:                 computer
  Created:              2018-03-12
  UpdateState:          success

UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-03-12

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* ubuntu	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* debian	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\debian\grubx64.efi)
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(3,GPT,fe1219bf-0100-4eb6-a684-673418a4475c,0xb2c800,0x36fd2800)/File(boot\efi\EFI\debian\fwupx64.efi)
Boot0006* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-7ceaf7a8-0611-4480-9e30-64d8de420c7c-0
```

```shell
$ tree /boot/efi
/boot/efi
├── EFI
│   ├── Boot
│   │   ├── en-us
│   │   │   └── bootx64.efi.mui
│   │   └── shimx64.efi
│   ├── debian
│   │   ├── fw
│   │   │   └── fwupdate-DiXb8K.cap
│   │   └── grubx64.efi
│   ├── dell
│   │   ├── bios
│   │   │   └── recovery
│   │   │       └── bios_cur.rcv
│   │   └── logs
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
│   ├── Microsoft
│   │   └── Boot
│   │       ├── BCD
│   │       ├── BCD.LOG
│   │       ├── en-us
│   │       ├── Fonts
│   │       │   ├── chs_boot.ttf
│   │       │   ├── cht_boot.ttf
│   │       │   ├── jpn_boot.ttf
│   │       │   ├── kor_boot.ttf
│   │       │   ├── malgun_boot.ttf
│   │       │   ├── meiryo_boot.ttf
│   │       │   ├── msjh_boot.ttf
│   │       │   ├── msyh_boot.ttf
│   │       │   ├── segmono_boot.ttf
│   │       │   ├── segoe_slboot.ttf
│   │       │   └── wgl4_boot.ttf
│   │       └── Resources
│   │           └── bootres.dll
│   └── ubuntu
│       ├── fw
│       ├── fwupx64.efi
│       ├── grub.cfg
│       ├── grubx64.efi
│       ├── mmx64.efi
│       └── shimx64.efi
└── en-us
    └── bootmgr.efi.mui

17 directories, 27 files
```

Please answer the following questions:

- Operating system and version: Debian Sid/unstable
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt install fwupd
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

superm1, 2018-03-14:
So this is of course related to the other issue. Your ESP wasn't mounted when you installed 'fwupdate'.  This is really a duplicate of https://github.com/rhboot/fwupdate/issues/97 where fwupdate won't install the file itself, it expects it to be there already.

You can solve this issue by mounting your ESP and running:
`dpkg-reconfigure fwupdate`

Since this is a duplicate of that issue however I'm going to close it.

closed 2018-03-14 by superm1

paulmenzel, 2018-03-14:
Running ` dpkg-reconfigure fwupdate` with ESP mounted, indeed fixed this.

```
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

$ findmnt /boot/efi 
TARGET    SOURCE         FSTYPE OPTIONS
/boot/efi /dev/nvme0n1p1 vfat   rw,relatime,fmask=0077,dmask=0077,codepage=437,i
$ sudo dpkg-reconfigure fwupdate
Installing fwupx64.efi to EFI system partition.
$ sudo fwupdmgr update
Downloading 0.1.2.0 for XPS 13 9370 System Firmware...
Updating 0.1.2.0 on XPS 13 9370 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: n
```