fwupd/fwupd #442: fwupd service takes 800 ms to start

paulmenzel, 2018-03-14:
With the Debian Sid/unstable package fwupd 1.0.6-2, `systemctl blame` shows that the service takes 800 ms to start.

```
$ systemd-analyze blame
          2.006s systemd-logind.service
           819ms fwupd.service
           242ms dev-nvme0n1p3.device
           184ms NetworkManager.service
           148ms systemd-timesyncd.service
           142ms systemd-journal-flush.service
           101ms upower.service
            96ms apparmor.service
            75ms systemd-journald.service
            75ms timidity.service
            72ms udisks2.service
            68ms ModemManager.service
            67ms user@114.service
            63ms libvirtd.service
            49ms binfmt-support.service
            49ms accounts-daemon.service
            45ms user@1000.service
            44ms ebtables.service
            44ms systemd-udev-trigger.service
            41ms packagekit.service
            34ms systemd-tmpfiles-setup.service
            34ms avahi-daemon.service
            32ms boot-efi.mount
            30ms speech-dispatcher.service
            29ms lvm2-monitor.service
            27ms systemd-udevd.service
            26ms networking.service
            26ms wpa_supplicant.service
            26ms systemd-rfkill.service
            25ms proc-sys-fs-binfmt_misc.mount
            24ms systemd-fsck@dev-disk-by\x2duuid-C484\x2d9038.service
            18ms gdm.service
            18ms ssh.service
            15ms systemd-tmpfiles-setup-dev.service
            14ms polkit.service
            13ms bluetooth.service
            13ms systemd-backlight@backlight:intel_backlight.service
            12ms colord.service
            12ms systemd-update-utmp.service
            11ms switcheroo-control.service
            11ms minissdpd.service
            11ms libvirt-guests.service
            10ms systemd-backlight@leds:dell::kbd_backlight.service
             9ms systemd-modules-load.service
             9ms blk-availability.service
             9ms sys-kernel-debug.mount
             9ms dev-hugepages.mount
             9ms pppd-dns.service
             8ms systemd-remount-fs.service
             8ms kmod-static-nodes.service
             7ms systemd-update-utmp-runlevel.service
             6ms dev-mqueue.mount
             6ms sys-fs-fuse-connections.mount
             6ms systemd-random-seed.service
             6ms systemd-sysctl.service
             5ms systemd-user-sessions.service
             2ms rtkit-daemon.service
             2ms openvpn.service
$ journalctl -o short-precise -b -u fwupd
-- Logs begin at Tue 2018-03-13 14:05:41 CET, end at Wed 2018-03-14 17:34:00 CET. --
Mar 14 17:32:54.050215 xps139370 systemd[1]: Starting Firmware update daemon...
Mar 14 17:32:54.251046 xps139370 fwupd[1445]: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Mar 14 17:32:54.254824 xps139370 fwupd[1445]: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Mar 14 17:32:54.257922 xps139370 fwupd[1445]: using plugins: dfu, udev, dell, colorhug, amt, nitrokey, upower, steelseries, unifying, csr, ebitdo, thunderbolt, altos, uefi
Mar 14 17:32:54.268432 xps139370 fwupd[1445]: Daemon ready for requests
Mar 14 17:32:54.870133 xps139370 systemd[1]: Started Firmware update daemon.
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.50
  VersionBootloader:    11.8.50
  Icon:                 computer
  Created:              2018-03-14

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.2.0
  VersionLowest:        0.1.2.0
  Icon:                 computer
  Created:              2018-03-14

UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-03-14

```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0003
Boot0000* ubuntu	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* debian	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\debian\grubx64.efi)
Boot0003* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(\EFI\debian\fwupx64.efi)
Boot0006* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,d99bc320-3294-4939-b69d-6c8498110580,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
```

Please answer the following questions:

- Operating system and version: Debian Sid/unstable
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt install fwupd
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

paulmenzel, 2018-03-14:
Looking at the SVG file created by systemd-bootchart it says 82 ms: fwupd [1445]82.0ms. Though the time stamps in the log messages verify the 800 ms. No idea, what is right.

hughsie, 2018-03-14:
If you run the daemon with `--verbose` you get some nice internal profiling, e.g.

	###################################################################################### @0244ms FuEngine:load 244ms
	 #                                                                                     @0005ms AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz} 1ms
	  #                                                                                    @0010ms AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz} 4ms
	   #                                                                                   @0010ms AsStore:store-from-root 1ms
					  #################################                                    @0146ms FuEngine:load-plugins 94ms
													   #                                   @0150ms FuEngine:setup 4ms
													   #                                   @0149ms FuEngine:setup{upower} 3ms
														 ############################      @0232ms FuEngine:coldplug 81ms
														 ###                               @0160ms FuEngine:coldplug{unifying} 9ms
															###################            @0217ms FuEngine:coldplug{amt} 56ms
																				###        @0227ms FuEngine:coldplug{udev} 9ms
																				###        @0227ms FuPluginUdev:coldplug{pci} 9ms
																					#      @0231ms FuEngine:coldplug{uefi} 4ms

What does this say for you?

paulmenzel, 2018-03-14:
On 03/14/2018 07:09 PM, Richard Hughes wrote:
> If you run the daemon with `--verbose` you get some nice internal profiling, e.g.

[…]

> What does this say for you?

I added the switch to the `StartExec=` line in the systemd service unit.

```
$ journalctl -o short-precise -b -u fwupd
-- Logs begin at Tue 2018-03-13 14:05:41 CET, end at Wed 2018-03-14 
23:50:21 CET. --
Mar 14 23:50:05.106001 xps139370 systemd[1]: Starting Firmware update 
daemon...
[…]
Mar 14 23:50:05.746284 xps139370 fwupd[1500]: 22:50:05:0746 Fu  ::added 
8de6c7959053fd5798006dcc63590d33fa5e51cb
Mar 14 23:50:05.919351 xps139370 fwupd[1500]: 22:50:05:0919 Fu 
registering D-Bus service
Mar 14 23:50:05.921207 xps139370 fwupd[1500]: 
###################################################################################### 
@0131ms FuEngine:load 131ms
Mar 14 23:50:05.921665 xps139370 fwupd[1500]:     # 
                                                             @0007ms 
FuEngine:load-quirks 1ms
Mar 14 23:50:05.922039 xps139370 fwupd[1500]:      ### 
                                                             @0013ms 
AsStore:store-from-file{/var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz} 5ms
Mar 14 23:50:05.922421 xps139370 fwupd[1500]:        # 
                                                             @0013ms 
AsStore:store-from-root 1ms
Mar 14 23:50:05.922823 xps139370 fwupd[1500]:                ########## 
                                                             @0039ms 
FuEngine:load-plugins 16ms
Mar 14 23:50:05.923149 xps139370 fwupd[1500]: 
#####                                                        @0049ms 
FuEngine:setup 9ms
Mar 14 23:50:05.923484 xps139370 fwupd[1500]: 
#                                                            @0043ms 
FuEngine:setup{upower} 3ms
Mar 14 23:50:05.923801 xps139370 fwupd[1500]: 
  ###                                                        @0049ms 
FuEngine:setup{uefi} 5ms
Mar 14 23:50:05.924130 xps139370 fwupd[1500]: 
         ##############################################      @0125ms 
FuEngine:coldplug 71ms
Mar 14 23:50:05.924447 xps139370 fwupd[1500]: 
         ###                                                 @0060ms 
FuEngine:coldplug{udev} 5ms
Mar 14 23:50:05.924765 xps139370 fwupd[1500]: 
         ###                                                 @0060ms 
FuPluginUdev:coldplug{pci} 5ms
Mar 14 23:50:05.925197 xps139370 fwupd[1500]: 
              ###################################            @0115ms 
FuEngine:coldplug{amt} 54ms
Mar 14 23:50:05.925522 xps139370 fwupd[1500]: 
                                                 #           @0117ms 
FuEngine:coldplug{unifying} 2ms
Mar 14 23:50:05.925847 xps139370 fwupd[1500]: 
                                                   #         @0120ms 
FuEngine:coldplug{uefi} 2ms
Mar 14 23:50:05.926449 xps139370 systemd[1]: Started Firmware update daemon.
```

-- Logs begin at Tue 2018-03-13 14:05:41 CET, end at Wed 2018-03-14 23:50:57 CET. --
Mar 14 23:50:05.106001 xps139370 systemd[1]: Starting Firmware update daemon...
Mar 14 23:50:05.185281 xps139370 fwupd[1500]: 22:50:05:0185 Fu  Verbose debugging enabled (on console 0)
Mar 14 23:50:05.185921 xps139370 fwupd[1500]: 22:50:05:0185 As  run FuEngine:load
Mar 14 23:50:05.185921 xps139370 fwupd[1500]: 22:50:05:0185 Fu  loading config values from /etc/fwupd/daemon.conf
[... 468 lines cut ...]
Mar 14 23:50:05.924447 xps139370 fwupd[1500]:                                    ###                                                 @0060ms FuEngine:coldplug{udev} 5ms
Mar 14 23:50:05.924765 xps139370 fwupd[1500]:                                    ###                                                 @0060ms FuPluginUdev:coldplug{pci} 5ms
Mar 14 23:50:05.925197 xps139370 fwupd[1500]:                                         ###################################            @0115ms FuEngine:coldplug{amt} 54ms
Mar 14 23:50:05.925522 xps139370 fwupd[1500]:                                                                            #           @0117ms FuEngine:coldplug{unifying} 2ms
Mar 14 23:50:05.925847 xps139370 fwupd[1500]:                                                                              #         @0120ms FuEngine:coldplug{uefi} 2ms
Mar 14 23:50:05.926449 xps139370 systemd[1]: Started Firmware update daemon.
Mar 14 23:50:05.926745 xps139370 fwupd[1500]: 22:50:05:0926 Fu  FuMain: acquired name: org.freedesktop.fwupd
Mar 14 23:50:05.929030 xps139370 fwupd[1500]: 22:50:05:0928 Fu  Called GetRemotes()
Mar 14 23:50:05.929348 xps139370 fwupd[1500]: 22:50:05:0929 Fu  Called GetResults(*)
Mar 14 23:50:05.929469 xps139370 fwupd[1500]: 22:50:05:0929 Fu  FuHistory: get devices
Mar 14 23:50:05.929661 xps139370 fwupd[1500]: 22:50:05:0929 Fu  FuHistory: got sql result 8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
Mar 14 23:50:06.336074 xps139370 fwupd[1500]: 22:50:06:0336 Fu  Called GetDevices()
Mar 14 23:50:06.336842 xps139370 fwupd[1500]: 22:50:06:0336 Fu  Called GetUpgrades(8a21cacfb0a8d2b30c5ee9290eb71db021619f8b)
Mar 14 23:50:06.336948 xps139370 fwupd[1500]: 22:50:06:0336 Fu  no requirement on id{org.freedesktop.fwupd}
Mar 14 23:50:06.336948 xps139370 fwupd[1500]: 22:50:06:0336 Fu  no requirement on firmware{(null)}
Mar 14 23:50:06.336948 xps139370 fwupd[1500]: 22:50:06:0336 Fu  no parameter given for firmware{bootloader}
Mar 14 23:50:06.336948 xps139370 fwupd[1500]: 22:50:06:0336 Fu  no parameter given for firmware{vendor-id}
[... cut ...]
Mar 14 23:50:11.772510 xps139370 fwupd[1500]: 22:50:11:0772 Fu  Called GetUpgrades(8a21cacfb0a8d2b30c5ee9290eb71db021619f8b)
Mar 14 23:50:11.772593 xps139370 fwupd[1500]: 22:50:11:0772 Fu  no requirement on id{org.freedesktop.fwupd}
Mar 14 23:50:11.772593 xps139370 fwupd[1500]: 22:50:11:0772 Fu  no requirement on firmware{(null)}
Mar 14 23:50:11.772593 xps139370 fwupd[1500]: 22:50:11:0772 Fu  no parameter given for firmware{bootloader}
Mar 14 23:50:11.772593 xps139370 fwupd[1500]: 22:50:11:0772 Fu  no parameter given for firmware{vendor-id}
Mar

hughsie, 2018-03-15:
Doesn't that say 131ms?

paulmenzel, 2018-03-23:
What does `systemctl blame | grep fwupd` say on your system?

hughsie, 2018-03-23:
Unknown operation blame.

paulmenzel, 2018-03-23:
On 03/23/2018 02:40 PM, Richard Hughes wrote:
>      Unknown operation blame.

Sorry, `systemd-analyze blame | grep fwupd`.

hughsie, 2018-03-23:
> 903ms fwupd.service

Most of that is probably waiting for the thunderbolt hardware to reply, in local tests it takes half a second or so. Being honest, I'm less worried about the real world duration of startup, and more worried about the CPU time -- fwupd does a lot of waiting for hardware which is going to be paralleled up nicely.

superm1, 2018-04-10:
Updates and queries around thunderbolt are filled with timeouts, I agree it's not going to change.
@paulmenzel you can confirm thunderbolt is the root cause of the longer time perceived by systemd by modifying `/etc/fwupd/fwupd.conf` and disabling the **thunderbolt** and **thunderbolt-power** plugins.

superm1, 2018-04-23:
I'm going to mark this one closed.  if you feel that's in error, you can re-open and discuss with more technical details provided.

closed 2018-04-23 by superm1