fwupd/fwupd #447: fwupd 1.0.6 fails to update 8bitdo gamepad

hguemar, 2018-03-23:
I failed to update the 8bitdo nes30 pro game pad on Fedora 27 using the base system package of fwupd.
I downgraded version of fwupd to make it succeed.

Current version in Fedora 27 (fails)
```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.6
	gusb:	0.2.11
	fwupdate:	10
	efivar:	32

```

Previous version in Fedora 27 (works)
```shell
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.7.2
	gusb:	0.2.11
```

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

Please answer the following questions:

- Operating system and version: Fedora 27
- How did you install fwupd: dnf
- Have you tried rebooting? YES

------

More details on what exactly happened.

If I connect the gamepad:
```shell
$ fwupdmgr get-devices # I cut unrelated parts of the output
8Bitdo  
  DeviceId:             5ec44b20be17233754d2a925e8fa8ee7ac91aef9
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 ba7f0ecd-1972-5bc0-b474-ffdc5fa17645
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              1.69
  Icon:                 input-gaming
  Created:              2018-03-23
$ fwupdmgr get-updates
8Bitdo NES30 Pro has firmware updates:
GUID:                    4fa94782-9ab0-531f-b038-a3e79ddf4b19
GUID:                    0b8f150f-21ca-5d84-94fb-0e728380eed2
GUID:                    043c7673-bd45-5e2e-b8b9-c22df2dfec2a
GUID:                    b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
ID:                      com.8bitdo.nes30pro.firmware
Update Version:          4.01
[... 95 lines cut ...]
Running as unit: run-r317bec2d1c8d48e196def6a94bc89a45.service
  Verifying        : fwupd-0.9.7-1.fc27.x86_64                                                                                                                                                                 1/7 
  Verifying        : fwupd-labels-0.9.7-1.fc27.x86_64                                                                                                                                                          2/7 
  Verifying        : gnome-software-3.26.1-3.fc27.x86_64                                                                                                                                                       3/7 
  Verifying        : libdfu-0.9.7-1.fc27.x86_64                                                                                                                                                                4/7 
  Verifying        : gnome-software-3.26.7-1.fc27.x86_64                                                                                                                                                       5/7 
  Verifying        : fwupd-1.0.6-1.fc27.x86_64                                                                                                                                                                 6/7 
  Verifying        : fwupd-labels-1.0.6-1.fc27.x86_64                                                                                                                                                          7/7 

Installed:
  libdfu.x86_64 0.9.7-1.fc27                                                                                                                                                                                       

Downgraded:
  fwupd.x86_64 0.9.7-1.fc27                                       fwupd-labels.x86_64 0.9.7-1.fc27                                       gnome-software.x86_64 3.26.1-3.fc27                                      

Complete!
$ fwupdmgr update
Downloading 4.01 for BOOTLOADER Gamepad...
Updating 4.01 on BOOTLOADER Gamepad...
Decompressing…         [-                                       ]
Restarting device…     [****************************************]
```

I was able to reproduce this issue with two nes30 pro (one with firmware 1.69, the other 1.70).
```

hguemar, 2018-03-23:
CC @hadess

superm1, 2018-09-28:
Many plugins have been reworked to use new hotplug stuff in the daemon.  It would be useful if you can retry this with 1.1.2 or master to see if it's improved.

With master the daemon will wait longer to let you replug and switch modes as well.

superm1, 2018-10-15:
I'm closing older bugs with `needsinfo` tags such as this one.  Can you please retry with a current release (1.1.3) and if the issues continue to exist, please re-open.

Even if the current release is not yet packaged for your distro, you can install it via snap like this:
```
# snap install --classic fwupd
```

closed 2018-10-15 by superm1