fwupd/fwupd #449: Missing instructions about updating Logitech keyboards

Ashark, 2018-03-25:
How to update firmware of Logitech wireless keyboards?
I have Logitech K850 wireless keyboard. Its stock firmware 04.01.0010 had a bug (first key was skipped when start typyng from sleep mode). Logitech fixed that problem by releasing [SecureDFU utilyty](https://community.logitech.com/s/question/0D531000058b3B7CAI/logitech-response-to-research-findings)  for Windows and macOS. It updates unifying recievers and these devices as well:

- K780 MULTI-DEVICE WIRELESS KEYBOARD
- K375s MULTI-DEVICE KEYBOARD
- WIRELESS TOUCH KEYBOARD K400 PLUS
- MK850 PERFORMANCE
- ILLUMINATED LIVING-ROOM KEYBOARD K830

I have launched that utilyty in macOS and updated my keyboard. Firmware version became 04.03.0015 and that bug has gone **.

Now I am interesting if updating firmware could be done from Linux. For some reason, Logitech did not uploaded its firmwares to [their lvfs account](https://fwupd.org/lvfs/search?value=logitech), but they are available in [Logitech/fw_updates](https://github.com/Logitech/fw_updates).

Exploring that repository, I have figured out that K850 keyboard is mentioned as MPK04. There is Makefile there, so I can make a .cab file to use it with fwupdmgr.
**But how can I actually start flashing firmware to keyboard?** When I run fwupdmgr get-devices, keyboard is not appeared there as specific device (unlike unifying recievers). But interesting fact is that Logitech actually released firmware for K780 keyboard in lvfs. Obviously, it's assumed that updating could be done from linux using fwupdmgr.

So how do I do that for K850 keyboard? Is there a way to update Logitech keyboard firmware from Linux? And if yes, then we should create a wiki page in this repository.

** However, keyboard started behave strange. Its modifiers keys were acting like they were swapped: start -> Alt_L, alt -> Super_L, alt_gr -> Super_R, right ctrl -> Alt_R. So it started behave like apple keyboard for all wireless slots (1, 2 and 3). To fix that I just pressed and hold `fn + P` for 6 seconds while being on needed slot and it became normal. To make it behave like an apple keyboard, switch to needed wireless slot, then press and hold `fn + O` for 6 seconds. I did not found this information in internet (got it from Logitech support), so I am sharing this for those who is searching.

---

```shell
$ fwupdmgr --version
client version: 1.0.6
daemon version: 1.0.6
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             ae38beffa8ec78c4220bbf35cedbf2c011de6653
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.08_B0030
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2018-03-25

Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              8.0.4
  VersionBootloader:    8.0.4
  Icon:                 computer
  Created:              2018-03-25

BCM20702A0
  DeviceId:             727aaff5feebfb2b80381c4e8d9e844629fca864
  Guid:                 1f4486b4-9d67-5f00-a52d-170fd8a4f0a4
  Guid:                 b04b321a-5b67-56bb-9a11-7c64d023a60d
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               Broadcom Corp
  VendorId:             USB:0x0A5C
  Version:              1.12
  Icon:                 drive-harddisk-usb
  Created:              2018-03-25

3rd Gen Core processor Graphics Controller
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 e91d9416-9ec2-5b2a-ac80-415605b96cb0
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-03-25

```
- Operating system and version: ArchLinux 4.15.11-1-ARCH
- How did you install fwupd:  pacman
- Have you tried rebooting: Yes
- Are you using an NVMe disk: No

Possibly related: #352
As I see, that person has his K780 keyboard displayed in get-devices list.

hughsie, 2018-03-25:
We're still working out all the edge cases for secure uploads, work in progress.

hkdhanwani, 2018-07-09:
Thanks Ashark - my modifier keys were swapped and I spent ages trying to figure it out, but pressing fn + p for 6 seconds worked.

Ashark, 2018-07-09:
@hkdhanwani I am glad that it was usefull

hughsie, 2018-09-12:
@Ashark -- Can you try with fwupd 1.1.2 -- it includes a much rewritten and improved unifying plugin and seems to work a lot better for more people. It can update my K870 just fine.

Ashark, 2018-09-12:
@hughsie Hi, thanks for info. Sorry, I could not try it right now, because my workplace is disassembled. I will try it when it will be possible.

Ashark, 2018-09-18:
> It can update my K870 just fine
@hughsie Maybe you mean K780?

superm1, 2018-09-18:
@Ashark if possible, can you please run the daemon in verbose mode `--verbose` and share output from it?

Ashark, 2018-09-18:
@superm1 could you please instruct me in more details. I am using [this](https://www.archlinux.org/packages/community/x86_64/fwupd/) package, I run `$ grep -r "ExecStart" .`  and got two files: `./usr/lib/systemd/system/fwupd.service:ExecStart=/usr/lib/fwupd/fwupd
./usr/lib/systemd/system/fwupd-offline-update.service:ExecStart=/usr/bin/fwupdmgr install-prepared`
Which of them should I edit?
You can write me a pm, so here I will write only relevant information.

superm1, 2018-09-18:
@Ashark `/usr/lib/systemd/system/fwupd.service`.  After you've added the verbose argument run:

```
systemctl daemon-reload
systemctl restart fwupd.service
```

Or reboot your machine.

Afterwards you will see in the journal for fwupd (`journalctl -u fwupd.service`) verbose logging turned on.

Ashark, 2018-09-18:
@superm1 thanks. Done.
<details><summary>Verbose log</summary><p>

```
-- Logs begin at Sun 2018-08-19 23:58:13 MSK, end at Tue 2018-09-18 23:47:45 MSK. --
Sep 18 20:34:37 unihost systemd[1]: Starting Firmware update daemon...
Sep 18 20:34:38 unihost fwupd[5137]: disabling plugin because: failed to startup dell_esrt: UEFI firmware already supported
Sep 18 20:34:38 unihost fwupd[5137]: disabling plugin because: failed to startup redfish: no SMBIOS table
Sep 18 20:34:38 unihost fwupd[5137]: disabling plugin because: failed to startup dell: Firmware updating not supported
Sep 18 20:34:38 unihost fwupd[5137]: disabling plugin because: failed to coldplug amt: ME refused connection
Sep 18 20:34:38 unihost fwupd[5137]: Daemon ready for requests
Sep 18 20:34:38 unihost systemd[1]: Started Firmware update daemon.
Sep 18 23:47:15 unihost systemd[1]: fwupd.service: Current command vanished from the unit file, execution of the command list won't be resumed.
Sep 18 23:47:25 unihost systemd[1]: Stopping Firmware update daemon...
Sep 18 23:47:25 unihost systemd[1]: Stopped Firmware update daemon.
Sep 18 23:47:25 unihost systemd[1]: Starting Firmware update daemon...
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0414 Fu  Verbose debugging enabled (on console 0)
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0414 As  run FuEngine:load
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0414 Fu  loading config values from /etc/fwupd/daemon.conf
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 Fwupd looking for os-release at /etc/os-release
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 As  run AsStore:load-installed{/usr/share/fwupd/metainfo}
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 As  Emitting ::changed() [load-installed]
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 Fu  using config path of /etc/fwupd
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0415 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  using config path of /usr/lib/fwupd
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  ordering lvfs=fwupd+1
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  ordering lvfs-testing=lvfs+1
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 As  run FuEngine:load-smbios
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 As  run FuEngine:load-hwids
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property Manufacturer=System manufacturer
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property EnclosureKind=3
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property Family=To be filled by O.E.M.
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property ProductName=System Product Name
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property ProductSku=SKU
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BiosVendor=American Megatrends Inc.
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BiosVersion=1002
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BiosMajorRelease=05
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BiosMinorRelease=0c
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BaseboardManufacturer=ASUSTeK COMPUTER INC.
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  smbios property BaseboardProduct=PRIME Z370-A
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {22172ae7-9b43-58f3-a8ec-9bf97befa80c}   <- System manufacturer&To be filled by O.E.M.&System Product Name&SKU&American Megatrends Inc.&1002&05&0c
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {e78e644e-6088-5280-8cf4-e2a7a5ee46e1}   <- System manufacturer&To be filled by O.E.M.&System Product Name&American Megatrends Inc.&1002&05&0c
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {1b2554f5-9e02-5f83-8b55-7d00e29979ff}   <- System manufacturer&System Product Name&American Megatrends Inc.&1002&05&0c
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {e70855fd-0c66-5379-936b-b20a84865c76}   <- System manufacturer&To be filled by O.E.M.&System Product Name&SKU&ASUSTeK COMPUTER INC.&PRIME Z370-A
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {9b3edbb4-9a86-5379-ad5e-4c6df298d9f8}   <- System manufacturer&To be filled by O.E.M.&System Product Name&SKU
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {ceb119e2-5411-5694-a8f1-57a5767fe85d}   <- System manufacturer&To be filled by O.E.M.&System Product Name
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {21af4f6b-87a3-5eec-a8a5-b0996e680fab}   <- System manufacturer&SKU&ASUSTeK COMPUTER INC.&PRIME Z370-A
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0416 Fu  {d2957e9e-702e-591d-a829-36238d29d986}   <- System manufacturer&SKU
[... cut ...]
Sep 18 23:47:25 unihost fwupd[13340]:                                                 #                                      @0288ms FuEngine:coldplug{thunderbolt_power} 2ms
Sep 18 23:47:25 unihost systemd[1]: Started Firmware update daemon.
Sep 18 23:47:25 unihost fwupd[13340]: 20:47:25:0937 Fu  FuMain: acquired name: org.freedesktop.fwupd
Sep 18 23:47:45 unihost fwupd[13340]: 20:47:45:0707 ThunderboltPower Bolt force power support: 0
Sep 18 23:47:45 unihost fwupd[13340]: 20:47:45:0707 ThunderboltPower Setting force power to 0 using kernel
```
</p></details>

hughsie, 2018-09-20:
@Ashark -- I think I know what's going on. The K850 is only available in the Logitech embargoed channel -- have you somehow managed to get a copy of the .cab file for that out of band? the unifying plugin only adds devices that have updates on the LVFS -- we had a few reports that just querying devices for "what firmware have you got installed" caused them to hang, and so we opted to only query the devices that we knew had updates -- see the code: https://github.com/hughsie/fwupd/blob/master/plugins/unifying/fu-plugin-unifying.c#L101

I could add a config file option (or a quirk?) to add all devices if you want.

Ashark, 2018-09-21:
Embargoed channel means that Logitech has uploaded firmware for K850 to lvfs, but has hidden it from end users, correct? Why do they do so? As I said in first note, there is a repository from which I could build a cab file [for my keyboard](https://github.com/Logitech/fw_updates/tree/master/MPK04/MPK04.01).
So having needed cab file is insufficient to make update. You could try to make K850 keyboard visible in get-devices output and then I will test flashing it from Linux.

hughsie, 2018-09-24:
@SonicFrog do you know why the firmware was left in embargo?

Ashark, 2019-09-23:
```
$ fwupdmgr --version
client version: 1.2.8
checkout info:  1.2.7-1-g622e054d
compile-time dependency versions
        gusb:   0.3.0
        efivar: 35
daemon version: 1.2.8
```
Still my K850 keyboard is not listed in `fwupdmgr get-devices`.

hughsie, 2019-09-23:
Some of this was fixed in https://github.com/fwupd/fwupd/commit/d9f52c75738939cc9a6edcf268cc139904523c98 which will be in the upcoming 1.3.2 release. It's still the case where Logitech have to push a stable firmware out before it will show. MPK04.01_B0010 was released now, and so the device should show up.

closed 2019-09-23 by hughsie