fwupd/fwupd #451: fwupdmgr handles LVFS being offline ungracefully

daenney, 2018-03-30:
```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

```

I tried to update my firmware today for a Dell XPS 9360 as GNOME Software informed me an update is available:
```
XPS 13 9360 System Firmware has firmware updates:
GUID:                    5ffdbc0d-f340-441c-a803-8439c8c0ae10
ID:                      com.dell.uefi5ffdbc0d.firmware
Update Version:          0.2.6.2
Update Name:             XPS 13 9360 System Update
Update Summary:          Firmware for the Dell XPS 13 9360
Update Remote ID:        lvfs
Update Checksum:         SHA1(959f78e66aead47f12954c3fba28344d7d4f7f2b)
Update Location:         https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab
Update Description:      Fixes: - Fixed incorrect Windows operating system logon message.
                         
                         Enhancements: - Enhanced SMM Security Mitigation feature. - Updated Embedded Controller Engine Firmware. - Updated Realtek USB LAN firmware</p>
```

However, when actually updating I get greeted with:
```
Downloading 0.2.6.2 for XPS 13 9360 System Firmware...
Fetching firmware https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab

Failed to download https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab: Bad Gateway
```

This error message is rather unhelpful. Doing some manual spellunking it turns out LVFS is temporarily offline. There's no way for me to know this aside from just opening the URL to the `.cab` in the browser and being greeted by a message that says so and similarly `fwupdmgr` doesn't get any context about the error either. The status page only contains vendor status information, not LVFS status itself.

It would be great if `fwupdmgr` could return a more helpful error message here, by for example extracting the 502 response code and returning a human friendly error instead.

superm1, 2018-03-30:
This actually appears to be a different problem with LVFS - it's online but that URL isn't downloading properly.

hughsie, 2018-04-03:
> for example extracting the 502 response code and returning a human friendly error instead

What would a human-friendly error look like in that case? I think the proper fix is to make the server more robust to this kind of thing, and also for the download to fall over to a "dumb" file-server if the main instance is down.

daenney, 2018-04-06:
For me, something simple like "LVFS is currently having issues serving firmware downloads" would've been enough, or the same message you do get if you open the .cab URL in a browser. Some kind of indication that this is not a problem on my side would be helpful.

I agree with the sentiment though, it would be nice if the server was more robust and if there were a fallback. But even in that situation, I don't think returning the HTTP status code's description when something does go wrong is helpful to the end user.

daenney, 2019-09-24:
That's not very nice

hughsie, 2019-09-24:
> That's not very nice

The thing is, we're a small group of people mostly working on this part time. The LVFS isn't some huge corporate thing with hundreds of employees. If you can, submitting a pull request implementing the desired changes is the best way to fix bugs like this. We're only human, and can only cope with so many open bugs before we just decide it's not fun anymore and go work on other things.

daenney, 2019-09-24:
I think it's fine if you can't work on everything. I don't believe anyone expects you to either. But it's a little depressing to submit a request for improvement to then get it closed as "ya sorry whatever" by some bot.

daenney, 2019-10-03:
What I'm objecting to here is not closing the bug, that's fine. But I would've liked to have gotten a reason by a human, instead of a bot deciding that since nobody interacted with my last comment for 30 days it wasn't worth keeping around.

closed 2019-10-03 by daenney

hughsie, 2019-10-03:
@daenney we just don't have the resources to do this kind of bug triage, sorry. For instance, this morning I woke up to three new issues and 6 different firmware-related emails from different people. I've still not worked though the issues reported yesterday.