fwupd/fwupd #454: Unable to Update XPS 13 9360

MichaelAquilina, 2018-04-09:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.3.0
	fwupdate:	9
	efivar:	31

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.5.1
  VersionBootloader:    11.5.1
  Icon:                 computer
  Created:              2018-04-09

XPS 13 9360 TPM 1.2
  DeviceId:             6f9095025a636a5185bfce08ed8e9fc845bc4b8a
  Guid:                 f9bdd338-b410-5e73-902d-7b6e4694bb56
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              5.81.2.1
  Icon:                 computer
  Created:              2018-04-09

XPS 13 9360 TPM 2.0
  DeviceId:             a60b665c769b019e30f31001e91ab31e24d7a035
  Guid:                 73c2051d-8688-56fb-a93f-d56a9b455e52
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-04-09

XPS 13 9360 System Firmware
  DeviceId:             2ed144f9aa3160a4b8bdf266ee7c84c0179f8a41
  Guid:                 48af7d21-3145-4f5e-b1b4-c8fc67b6e646
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.0.7
  VersionLowest:        0.1.0.7
  Icon:                 computer
  Created:              2018-04-09
  Modified:             2018-04-09
  UpdateState:          failed
  UpdateError:          UEFI firmware update failed: {error #0} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory	{error #1} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory	{error #2} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory	{error #3} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory	{error #4} libfwup.c:1206 get_fd_and_media_path(): mkostemps(/boot/efi/EFI/opensuse/fw/fwupdate-wYURRl.cap) failed: No such file or directory	{error #5} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory	{error #6} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory	{error #7} efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-48af7d21-3145-4f5e-b1b4-c8fc67b6e646-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory	{error #8} lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory	{error #9} libfwup.c:1206 get_fd_and_media_path(): mkostemps(/boot/efi/EFI/opensuse/fw/fwupdate-TobOrT.cap) failed: No such file or directory

HD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-04-09

XPS13 9360 Thunderbolt Controller
  DeviceId:             336606a4c6bfd0f07394494aaf692be97869e1e4
  Guid:                 4d86f168-e1cc-5995-afd3-ae9df6a14f5e
  Guid:                 dfd51125-338f-56ff-b721-fa3bea8e534e
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2018-04-09
  Modified:             2018-04-09
  UpdateState:          success
```

Please answer the following questions:

- Operating system and version: OpenSUSE Tumbleweed
- How did you install fwupd: preinstalled with system
- Have you tried rebooting?: Yes
- Are you using an NVMe disk?: I think so
- Is secure boot enabled (only for the UEFI plugin)?: not enabled

superm1, 2018-04-09:
What does your EFI System partition look like?  To me this stands out:
```
mkostemps(/boot/efi/EFI/opensuse/fw/fwupdate-TobOrT.cap) failed: No such file or directory
```

* Is your ESP mounted to /boot/efi?
* If so do you have that directory that's missing?  (Can you please give `tree` output?)

MichaelAquilina, 2018-04-09:
I'm not sure what ESP is :sweat_smile: 

but here is the tree of my boot:
```
/boot
├── boot.readme
├── config-4.15.10-1-default
├── config-4.15.13-1-default
├── config-4.16.0-1-default
├── do_purge_kernels
├── efi
│   └── EFI
│       ├── boot
│       │   ├── bootx64.efi
│       │   └── fallback.efi
│       └── opensuse
│           ├── boot.csv
│           ├── grub.cfg
│           ├── grub.efi
│           ├── grubx64.efi
│           ├── MokManager.efi
│           └── shim.efi
├── grub2
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── i386-pc
│   ├── locale
│   │   ├── ast.mo
│   │   ├── ca.mo
│   │   ├── da.mo
│   │   ├── de_CH.mo
│   │   ├── de.mo
│   │   ├── en@quot.mo
│   │   ├── eo.mo
│   │   ├── es.mo
│   │   ├── fi.mo
│   │   ├── fr.mo
│   │   ├── gl.mo
│   │   ├── hr.mo
│   │   ├── hu.mo
│   │   ├── id.mo
│   │   ├── it.mo
│   │   ├── ja.mo
│   │   ├── ko.mo
│   │   ├── lt.mo
│   │   ├── nb.mo
│   │   ├── nl.mo
│   │   ├── pa.mo
│   │   ├── pl.mo
│   │   ├── pt_BR.mo
│   │   ├── ru.mo
│   │   ├── sl.mo
│   │   ├── sr.mo
│   │   ├── sv.mo
│   │   ├── tr.mo
│   │   ├── uk.mo
│   │   ├── vi.mo
│   │   ├── zh_CN.mo
│   │   └── zh_TW.mo
│   ├── themes
│   │   └── openSUSE
│   │       ├── ascii.pf2
│   │       ├── COPYING.CC-BY-SA-3.0
│   │       ├── DejaVuSans10.pf2
│   │       ├── DejaVuSans12.pf2
│   │       ├── DejaVuSans-Bold14.pf2
│   │       ├── highlight_c.png
│   │       ├── logo.png
[... 274 lines cut ...]
├── initrd -> initrd-4.16.0-1-default
├── initrd-4.15.10-1-default
├── initrd-4.15.13-1-default
├── initrd-4.16.0-1-default
├── symvers-4.15.10-1-default.gz
├── symvers-4.15.13-1-default.gz
├── symvers-4.16.0-1-default.gz
├── sysctl.conf-4.15.10-1-default
├── sysctl.conf-4.15.13-1-default
├── sysctl.conf-4.16.0-1-default
├── System.map-4.15.10-1-default
├── System.map-4.15.13-1-default
├── System.map-4.16.0-1-default
├── vmlinux-4.15.10-1-default.gz
├── vmlinux-4.15.13-1-default.gz
├── vmlinux-4.16.0-1-default.gz
├── vmlinuz -> vmlinuz-4.16.0-1-default
├── vmlinuz-4.15.10-1-default
├── vmlinuz-4.15.13-1-default
└── vmlinuz-4.16.0-1-default

11 directories, 348 files
```

One weird thing I have (which was a mistake I want to fix) is that the boot partition is encrypted. I suspect this was a mistake in the setup process. Could this be affecting it?

superm1, 2018-04-09:
Looking at https://build.opensuse.org/package/view_file/Base:System/fwupdate/fwupdate.spec?expand=1 I suspect you're missing the package `fwupdate-efi` which should do those actions for you.

MichaelAquilina, 2018-04-09:
> looking at https://build.opensuse.org/package/view_file/Base:System/fwupdate/fwupdate.spec?expand=1 I suspect you're missing the package fwupdate-efi which should do those actions for you.

that worked! Got a prompt to restart now which I will try. It seems like `fwupdate-efi` is a missing dependency then? I would assume that `fwupdmgr` should depend on `fwupdate-efi`?

MichaelAquilina, 2018-04-09:
errr actually, accepting the prompt now gives me this
```
No upgrades for XPS 13 9360 TPM 1.2, current is 5.81.2.1: 5.81.2.1=same
Downloading 0.1.3.2 for XPS 13 9360 System Firmware...
Updating 0.1.3.2 on XPS 13 9360 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
No upgrades for XPS13 9360 Thunderbolt Controller, current is 21.00: 21.00=same

An update requires a reboot to complete. Restart now? [Y|n]: Y
GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.201" (uid=1000 pid=21038 comm="fwupdmgr update ") interface="org.freedesktop.systemd1.Manager" member="Reboot" error name="(unset)" requested_reply="0" destination="org.freedesktop.systemd1" (uid=0 pid=1 comm="/usr/lib/systemd/systemd --system --deserialize 18")

```

superm1, 2018-04-09:
Great, glad that worked.  We'll make a known issues for this problem.  You should file something with your distro to find out why fwupdate-efi wasn't installed by default but fwupd was (that's the bug there).

Regarding your second issue can you follow up in another issue to not convolute this one, let's treat it separately.
In that thread please add the relevant output and please answer: are you doing this somewhere without a GUI (eg SSH)?  I think PolicyKit should have popped something up for that.

MichaelAquilina, 2018-04-09:
Will do, thanks for the help :)

closed 2018-04-09 by MichaelAquilina

superm1, 2018-04-09:
Here's the known issue for this now:
https://github.com/hughsie/fwupd/wiki/LVFS-Triaged-Issue:-OpenSuSE:-Missing-fwupdate-efi-package

If you have some results from your notifying distro maintainers that it's fixed in some particular version please feel free to change that page or let us know here so we can update it.

MichaelAquilina, 2018-04-09:
Thanks! I've already posted to the mailing list to make them aware about this issue :)