fwupd/fwupd #467: Clang is failing to build

superm1, 2018-04-16:
After turning Ubuntu CI back onto clang noticed this:
```
FAILED: libfwupd/fwupd@sha/fwupd-client.c.o 
clang  -Ilibfwupd/fwupd@sha -Ilibfwupd -I../libfwupd -Ilibfwupd/.. -I../libfwupd/.. -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/json-glib-1.0 -Xclang -fcolor-diagnostics -pipe -D_FILE_OFFSET_BITS=64 -Werror -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wignored-qualifiers -Wimplicit-function-declaration -Wincompatible-pointer-types-discards-qualifiers -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wno-error=cpp -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-variable -Wwrite-strings -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -g -O2 -fdebug-prefix-map=/build/build=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="Fwupd"' '-DLOCALSTATEDIR="/var"' -MD -MQ 'libfwupd/fwupd@sha/fwupd-client.c.o' -MF 'libfwupd/fwupd@sha/fwupd-client.c.o.d' -o 'libfwupd/fwupd@sha/fwupd-client.c.o' -c ../libfwupd/fwupd-client.c
../libfwupd/fwupd-client.c:113:1: error: unused function 'glib_listautoptr_cleanup_FwupdClientHelper' [-Werror,-Wunused-function]
G_DEFINE_AUTOPTR_CLEANUP_FUNC(FwupdClientHelper, fwupd_client_helper_free)
^
/usr/include/glib-2.0/glib/gmacros.h:462:22: note: expanded from macro 'G_DEFINE_AUTOPTR_CLEANUP_FUNC'
  static inline void _GLIB_AUTOPTR_LIST_FUNC_NAME(TypeName) (GList **_l) { g_list_free_full (*_l, (GDestroyNotify) func); } \
                     ^
/usr/include/glib-2.0/glib/gmacros.h:443:48: note: expanded from macro '_GLIB_AUTOPTR_LIST_FUNC_NAME'
#define _GLIB_AUTOPTR_LIST_FUNC_NAME(TypeName) glib_listautoptr_cleanup_##TypeName
                                               ^
<scratch space>:162:1: note: expanded from here
glib_listautoptr_cleanup_FwupdClientHelper
^
../libfwupd/fwupd-client.c:113:1: error: unused function 'glib_slistautoptr_cleanup_FwupdClientHelper' [-Werror,-Wunused-function]
/usr/include/glib-2.0/glib/gmacros.h:463:22: note: expanded from macro 'G_DEFINE_AUTOPTR_CLEANUP_FUNC'
  static inline void _GLIB_AUTOPTR_SLIST_FUNC_NAME(TypeName) (GSList **_l) { g_slist_free_full (*_l, (GDestroyNotify) func); } \
                     ^
/usr/include/glib-2.0/glib/gmacros.h:445:49: note: expanded from macro '_GLIB_AUTOPTR_SLIST_FUNC_NAME'
#define _GLIB_AUTOPTR_SLIST_FUNC_NAME(TypeName) glib_slistautoptr_cleanup_##TypeName
                                                ^
<scratch space>:163:1: note: expanded from here
glib_slistautoptr_cleanup_FwupdClientHelper
^
2 errors generated.
```

hughsie, 2018-04-16:
I think clang is clearly wrong here, right? Should we turn off Wunused-function for clang at the moment?

superm1, 2018-04-16:
Actually I think clang might be right about what's going on.
Both of:
```
_GLIB_AUTOPTR_LIST_FUNC_NAME
_GLIB_AUTOPTR_SLIST_FUNC_NAME
```
are declared when you do `G_DEFINE_AUTOPTR_CLEANUP_FUNC`
see: https://github.com/GNOME/glib/blob/master/glib/gmacros.h#L477

So that generates some other functions that just happen to not be used anywhere in the code since the `g_autolist` and `g_autoslist` aren't used with this type.

superm1, 2018-04-16:
(That said, I think `-Wunused-function` for clang is probably a fine solution for this problem)

superm1, 2018-04-16:
It looks like same error in altos, amt, unifying csr, maybe others.  Yeah probably better to just turn it off for all clang builds rather than per module.

commit 3be987b7 (2018-04-16): Disable -Wunused-function on clang builds (Closes #467)

GLib creates two static inline functions for paramaters that may
not be used that set off warnings in clang but not gcc.

Ignore these on clang builds everywhere that
G_DEFINE_AUTOPTR_CLEANUP_FUNC is used.
commit 602b8057 (2018-04-16): Disable -Wunused-function on clang builds (Closes #467)

GLib creates two static inline functions for paramaters that may
not be used that set off warnings in clang but not gcc.

Ignore these on clang builds everywhere that
G_DEFINE_AUTOPTR_CLEANUP_FUNC is used.
commit 467ef7a1 (2018-04-16): Disable -Wunused-function on clang builds (Closes #467)

GLib creates two static inline functions for paramaters that may
not be used that set off warnings in clang but not gcc.

Ignore these on clang builds everywhere that
G_DEFINE_AUTOPTR_CLEANUP_FUNC is used.
commit 3ac48411 (2018-04-16): Disable -Wunused-function on clang builds (Closes #467)

GLib creates two static inline functions for paramaters that may
not be used that set off warnings in clang but not gcc.

Ignore these on clang builds everywhere that
G_DEFINE_AUTOPTR_CLEANUP_FUNC is used.
closed 2018-04-17 by hughsie