fwupd/fwupd #470: update logitech firmware to RQR12.08_B0030

arekm, 2018-04-17:
Logitech windows software upgrades unifying receiver firmware to RQR12.08_B0030
while fwupd knows only about RQR12.07_B0029.

Please consider providing new version, too.

arekm, 2018-04-17:
https://github.com/Logitech/fw_updates/tree/master/RQR12/RQR12.08

arekm, 2018-06-10:
And 12.09 now:
https://github.com/Logitech/fw_updates/tree/master/RQR12/RQR12.09

ktecho, 2018-06-15:
I'm interested in this as well, as I updated from RQR12.03_B0025 to RQR12.07_B0029 and now my mouse is left-clicking everything as fast as I click the button. I can't even select a text from a web page because it clicks a lot. Maybe latest firmwares fixes this...

ktecho, 2018-06-25:
It seems like it was a hardware problem, so please forgot my last comment.

hughsie, 2018-09-12:
>Can fwupd handle signed DFU

Yes! Can you try with 1.0.9 or preferably 1.1.2 -- the latter includes a much rewritten and improved unifying plugin and seems to work a lot better for more people.

closed 2018-09-12 by hughsie