fwupd/fwupd #472: CI is failing the debian builds

superm1, 2018-04-18:
This started happening a few days ago.

Lintian fails with this:
```
+ lintian ../fwupd_1.0.6+r41+gc60e7cd_amd64.changes -IE --pedantic --no-tag-display-limit --suppress-tags bad-distribution-in-changes-file --suppress-tags source-contains-unsafe-symlink --suppress-tags changelog-should-mention-nmu --suppress-tags debian-watch-file-in-native-package --suppress-tags source-nmu-has-incorrect-version-number --suppress-tags no-symbols-control-file --allow-root
P: fwupd source: package-uses-old-debhelper-compat-version 10
E: fwupd-tests: library-not-linked-against-libc usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_test.so
I: fwupd: hardening-no-fortify-functions usr/bin/fwupdmgr
I: fwupd: hardening-no-fortify-functions usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_altos.so
I: fwupd: hardening-no-fortify-functions usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_synapticsmst.so
I: fwupd: hardening-no-fortify-functions usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_unifying.so
E: fwupd: library-not-linked-against-libc usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_upower.so
N: 3 tags overridden (3 info)
```

The error level tags cause the failures.  I looked at the files built locally and they do dynamically link to libc.

```
$ ldd build/obj-x86_64-linux-gnu/plugins/test/libfu_plugin_test.so
        linux-vdso.so.1 =>  (0x00007ffee57fb000)
        libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f0f3561c000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f0f3530b000)
        libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f0f35103000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0f34d39000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f0f34ac9000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0f348ac000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f0f35a72000)
```

objdump however doesn't show libc.so.6 as needed:
```
$ objdump -x build/obj-x86_64-linux-gnu/plugins/test/libfu_plugin_test.so | grep needed -i
  NEEDED               libgobject-2.0.so.0
  NEEDED               libglib-2.0.so.0
```

Now what's really peculiar is that building with Ubuntu (bionic) this doesn't happen.
```
$ objdump -x build/obj-x86_64-linux-gnu/plugins/test/libfu_plugin_test.so | grep needed -i
  NEEDED               libgobject-2.0.so.0
  NEEDED               libglib-2.0.so.0
  NEEDED               libpthread.so.0
  NEEDED               libc.so.6
```

So something in Debian testing started to introduce an optimization causing this problem.

@ximion any ideas?

superm1, 2018-04-18:
Actually forcing Ubuntu's build to use `CC=gcc` instead of `clang` causes it to fail too.
```
$ sudo docker run -e CC=gcc -i -t -v `pwd`:/build fwupd-ubuntu
E: fwupd: library-not-linked-against-libc usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_upower.so
E: fwupd-tests: library-not-linked-against-libc usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_test.so
$ objdump -x build/obj-x86_64-linux-gnu/plugins/test/libfu_plugin_test.so | grep needed -i
  NEEDED               libgobject-2.0.so.0
  NEEDED               libglib-2.0.so.0
```

hughsie, 2018-04-18:
Really odd. I don't think fwupd should be have to add libc manually to `plugin_deps`...

superm1, 2018-04-18:
I agree, it shouldn't.
I looked at the two closest build logs(https://travis-ci.org/hughsie/fwupd/jobs/367313640 and https://travis-ci.org/hughsie/fwupd/jobs/367844771) to find the differences.  It's not CFLAGS or anything like that.
The ones that stand out are:
* https://tracker.debian.org/news/948916/accepted-lintian-2582-source-all-into-unstable/
* https://tracker.debian.org/news/948615/accepted-gcc-7-730-16-source-into-unstable/

(There was a few other changes like cups and sqlite3 too)

I can reproduce the problem with older lintian, so it's probably caused by the new GCC.

hughsie, 2018-04-18:
> so it's probably caused by the new GCC

That would be my guess too. Maybe we just ignore those lintian warnings?

superm1, 2018-04-18:
I just downgraded to 7.3.0-15ubuntu2 in Ubuntu (Ubuntu had it cached fortunately here: https://launchpad.net/ubuntu/+source/gcc-7/7.3.0-15ubuntu2/+build/14540379)

The problem goes away.

I'd say that's probably enough to just ignore them, but then I did one more check.
Running the build with default flags for the project (eg just `meson build; ninja -C build`) it comes out OK too even with the newer compiler.
```
# objdump -x build/plugins/test/libfu_plugin_test.so | grep needed -i
  NEEDED               libgobject-2.0.so.0
  NEEDED               libglib-2.0.so.0
  NEEDED               libc.so.6
```

So it's a combination of some flag being set and the new compiler.
**Link time**
These were the flags of a good (manual build from `meson build;ninja -C build`)
```
[140/220] cc  -o plugins/test/libfu_plugin_test.so 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -Wl,--as-needed -shared -fPIC -Wl,--start-group -Wl,-soname,libfu_plugin_test.so -Wl,-z,relro -Wl,-z,now -lappstream-glib -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0 -lgusb -lgobject-2.0 -lglib-2.0 -lusb-1.0 -lsoup-2.4 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -larchive -Wl,--end-group
```

Here's the bad build:
```
[135/220] cc  -o plugins/test/libfu_plugin_test.so 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -Wl,--as-needed -shared -fPIC -Wl,--start-group -Wl,-soname,libfu_plugin_test.so -Wl,-z,relro -Wl,-z,now -g -O2 -fdebug-prefix-map=/build/build=. -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -Wl,-z,now -lappstream-glib -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0 -lgusb -lgobject-2.0 -lglib-2.0 -lusb-1.0 -lsoup-2.4 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -larchive -Wl,--end-group  
```

Diff is:
```
-g -O2 -fdebug-prefix-map=/build/build=. -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -Wl,-z,now
```

**Compile time**
Good:
```
[139/220] cc  -Iplugins/test/fu_plugin_test@sha -Iplugins/test -I../plugins/test -Iplugins/test/../.. -I../plugins/test/../.. -Iplugins/test/../../src -I../plugins/test/../../src -Iplugins/test/../../libfwupd -I../plugins/test/../../libfwupd -I/usr/include/libappstream-glib -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/gio-unix-2.0/ -I/usr/include/uuid -I/usr/include/libgcab-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -O0 -g -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-error=cpp -Wno-unknown-pragmas -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_GNU_SOURCE -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginTest"' -MD -MQ 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -MF 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o.d' -o 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -c ../plugins/test/fu-plugin-test.c
```

Bad:
```
[134/220] cc  -Iplugins/test/fu_plugin_test@sha -Iplugins/test -I../plugins/test -Iplugins/test/../.. -I../plugins/test/../.. -Iplugins/test/../../src -I../plugins/test/../../src -Iplugins/test/../../libfwupd -I../plugins/test/../../libfwupd -I/usr/include/libappstream-glib -I/usr/include/uuid -I/usr/include/gio-unix-2.0/ -I/usr/include/libgcab-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Werror -std=c99 -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-error=cpp -Wno-unknown-pragmas -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prot [... cut ...] ix-map=/build/build=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginTest"' -MD -MQ 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -MF 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o.d' -o 'plugins/test/fu_plugin_test@sha/fu-plugin-test.c.o' -c ../plugins/test/fu-plugin-test.c
```

The differences are bigger there, but it looks like making a single optimization change causes it:
Going from `-O0` to `-O2`

superm1, 2018-04-18:
OK so i've got a temporary PR in place (that we should eventually revert).
It's filed in Ubuntu here: https://bugs.launchpad.net/ubuntu/+source/gcc-7/+bug/1765134
and in Debian here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=896012

commit 6b0eb078 (2018-04-18): trivial: Compile upower and test plugins with -O0 (Closes: #472)

This is a workaround for what looks like a compiler bug introduced
in Debian and Ubuntu gcc-7.

After the compiler bug is sorted in Debian testing it should be
reverted.
closed 2018-04-18 by hughsie