fwupd/fwupd #476: Error message lacks information about device

hadess, 2018-04-19:
Using current git master:
```
(fwupd:17385): FuPluginDfu-WARNING **: 17:35:07.153: failed to parse interface data: interface found, but not the correct length for functional data: 33 bytes: 09 21 01 c8 00 00 40 01 01 10 40 09 02 24 06 04 00 45 6c 6c 61 44 6f 63 6b 08 0b 02 03 01 00 20 07
```

I don't know which device this is for, even debug mode doesn't ever print which device it's handling.

Same for:
```
(fwupd:17451): Fu-WARNING **: 17:37:26.783: failed to add USB device: unexpected device response
```

hughsie, 2018-04-19:
> print which device it's handling

What were you expecting? Something like "Bus 002 Device 005" or a string vendor/model number?

hadess, 2018-04-19:
Anything that could allow to identify the device, such as vid/pid would be enough. I don't think that multiple similar devices is a big problem for those cases, but if you have a good way to differentiate them...

closed 2018-04-19 by hughsie