fwupd/fwupd #480: Fails to start using the systemd unit

Keruspe, 2018-04-25:
When launching fwupdmgr or `systemctl start fwupd`, the systemd service fails with: `/usr/x86_64-pc-linux-gnu/libexec/fwupd/fwupd: error while loading shared libraries: libcrypto.so.1.0.0: cannot enable executable stack as shared object requires: Operation not permitted`

When launching `/usr/x86_64-pc-linux-gnu/libexec/fwupd/fwupd` manually, everything works fine. I guess some permission is missing from the systemd service?

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.0.6
```

Please answer the following questions:

- Operating system and version: Exherbo
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): from source using package manager
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

Keruspe, 2018-04-25:
Commenting the `MemoryDenyWriteExecute=yes` line fixes this

hughsie, 2018-04-25:
MemoryDenyWriteExecute is probably doing what it's supposed to be doing -- it's a security protection. You probably need to work out what (and why) a library is trying to execute writable memory. I suspect it's Exherbo-specific, as we don't see any failures on any distro in the CI matrix: https://travis-ci.org/hughsie/fwupd

superm1, 2018-04-25:
Agree, probably a compile flag you're passing during build to either fwupd or a dependency.

`readelf --all` on the fwupd binary and `execstack -q` on all binaries can probably help figure out the executable stack issue.