fwupd/fwupd #482: Needs to deal with composite devices

hughsie, 2018-04-27:
At the moment fwupd has a flat list of devices that mostly correlate to `FuDevice`:item-of-hardware. This is starting to become not true. For instance,

* A drawing tablet is one physical device, but internally it has a bluetooth module and a main system firmware, either of which that can be updated by the vendor. There's no relationship between the two firmwares and updates are not released together.

* A Logitech keyboard (upgradable) is only available when the Logitech unifying receiver (also upgradable) is plugged in. In some instances you need to update the keyboard firmware before the unifying firmware, or vice-versa.

* A modem device has two firmwares, a chipset firmware and a carrier-specific (e.g. Sprint, AT&T) firmware on the same physical device. Some carrier firmwares are incompatible with some chipset firmwares, and this is further complicated as you can have up to 8 chipset firmwares installed at any one time, and up to 64 different carrier firmwares available. Both are user selectable, depending on the SIM used.

I think we need to do two things:

* Add a concept of a "parent" device (possibly setting the `FuDevice` server side, and exposing the `device_id_parent` string on the client device over D-Bus).

* Add nesting in the `fwupdmgr get-devices` output to make this mapping more obvious. I was thinking something like this:

     ○ Unifying Receiver
       ├─ Created:              2018-04-27
       ├─ DeviceId:             00e5a560d608250caaa89e428d5a975362473104
       ├─ Flags:                updatable|supported|registered
       ├─ Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1,cc4cbfa9-bf9d-540b-b92b-172ce31013c1
       ├─ Icon:                 preferences-desktop-keyboard
       ├─ Plugin:               unifying
       ├─ Summary:              A miniaturised USB wireless receiver
       ├─ Vendor:               Logitech
       ├─ VendorId:             USB:0x046D
       ├─ Version:              RQR24.05_B0029
       └─ VersionBootloader:    BOT03.01_B0008
             └─ ○ Unifying Keyboard K750
               ├─ Created:              2018-04-27
               ├─ DeviceId:             5ad59b26d556e226f9e5e1ca13e4b717fe19802c
               ├─ Flags:                updatable|supported|registered
               ├─ Guid:                 9414852e-fa88-51e7-96f5-45268b29b86c
               ├─ Icon:                 preferences-desktop-keyboard
               ├─ Plugin:               unifying
               ├─ Summary:              A USB wireless keyboard
               ├─ Vendor:               Logitech
               ├─ VendorId:             USB:0x046D
               └─ Version:              MQM24.05_B0029

Ideas and comments welcome.

superm1, 2018-05-15:
I think this can be closed now right?  Parent devices are in and you can now define whether parent needs to be installed first or order doesn't matter.

hughsie, 2018-05-15:
Agreed.

closed 2018-05-15 by hughsie